velvet-os/imagebuilder #90: It is possible to install different init systems.

aneeshlingala, 2022-10-20:
I have used Devuan Linux repos to make this possible.

What is Devuan?
As said on their DistroWatch: Devuan GNU+Linux is a Linux distribution forked from Debian in 2015. The project's primary goal is to provide a variant of Debian without the complexities and dependencies of systemd, an init system and services manager originally developed by Red Hat and later adopted by most other Linux distributions.

I have install two init systems, sysvinit, and runit, since I like runit, I am currently running it.
I have noticed that it is a bit faster and less bloated than systemd. 
Should I make a guide on how to do this?

Thanks, 
Aneesh

hexdump0815, 2022-10-21:
@aneeshlingala - wirting down information which might be useful for others is a good idea most of the time i think ... did you install it based on the existing images (i.e. adjusted them) or did you modify the images for it or built your own ones?

aneeshlingala, 2022-10-22:
I have installed it based on existing images. Everything is working out-of-the box.

aneeshlingala, 2022-10-22:
Well, I have made a simple guide for this topic, it is [issue #91](https://github.com/hexdump0815/imagebuilder/issues/91)

hexdump0815, 2022-10-23:
@aneeshlingala  - maybe we can close this issue here then?

aneeshlingala, 2022-10-23:
Issue Closed.

closed 2022-10-23 by aneeshlingala