velvet-os/imagebuilder #91: A guide on different init systems. (working: runit, sysvinit)

aneeshlingala, 2022-10-22:
**Well, you might be bored of systemd, or you hate it because its too bloated, this is your special guide on how to install different init systems**

1. _**Info about init**_

__You may ask, "what is init?"._

Well, init is a very crucial process that has to be run as PID 1 or it will halt boot. It is the master process that starts all of the child processes and sets up the system for normal use, this usually happens with no user input. Usually init hands over this process to another init system, runit, sysvinit, systemd, s6, etc. But sometimes init can also run on its own, pretty cool right?

2. _**Enough info, lets start installing**_

**NOTE: IF YOU DON'T FOLLOW INSTRUCTIONS CORRECTLY, IT COULD RUIN YOUR INSTALL.**

Working display managers
SDDM, installed by running the command `sudo apt install sddm`
SLIM, installed by running the command `sudo apt install slim`

Not working display managers
LightDM, (Somebody can help me on this? Seems to work on devuan out-of-the-box but still not working?)

Installing git, nano, ucf and wget by running the command `sudo apt install git nano wget ucf`
Make sure you cd to your home directory, `cd ~`
Now, run the command `git clone http://ftp.us.debian.org/debian/pool/main/o/orphan-sysvinit-scripts/orphan-sysvinit-scripts_0.07_all.deb`
Now, clone the repository, `git clone https://git.devuan.org/farmatito/migration`
Now, run `cd migration`
Now run `sudo bash migration.sh`
Now it is going to output "Warning use at your own risk. BACKUP YOUR DATA FIRST!!!", on this prompt type "y" to continue.
Now, it is going to ask "Do you want contrib repos to be included in your apt sources? [y|n]", on this question type "y."
Now, it is going to ask "Do you want non-free repos to be included in your apt sources? [y|n]", on this question type "y."
After some time, it is going to ask "Do you want to edit /etc/network/interfaces now? [y|n]", on this question type "n".
Now reboot by running the command `sudo reboot`
Now, when your back in the system, login, and run the command `sudo nano /etc/apt/sources.list`
Now, remove any other entries that are not pointing to "devuan.org".
Then, run `sudo apt update`
Now, you are not going to have network, don't worry, run the command "cd ~ && sudo dpkg -i orphan-sysvinit-scripts_0.07_all.deb"
Now run, `cd /etc/init.d`
Then, run `sudo bash network-manager start && sudo bash networking start`
Once your connected to the network again, run `sudo apt reinstall base-files && sudo apt upgrade`
Now, run `sudo apt clean`
You can stop here if you want sysvinit, but if you want runit, then continue.
For runit, run `sudo apt install runit-init init`, if it tells you to input "Yes, do as I day", then do it.
Then, run `sudo reboot`
Your done!

3. Cleanup
First run `cd ~`
Then run `sudo rm -rf migration`
After that, run `sudo rm -rf orphan-sysvinit-scripts_0.07_all.deb`

_**Well, this is the end of the guide, I hope you have a good journey on your new init system.**_

hexdump0815, 2022-10-23:
thanks a lot for writing this down - might be useful for others wanting to try it

aneeshlingala, 2022-10-23:
> thanks a lot for writing this down - might be useful for others wanting to try it

No problem :)