velvet-os/imagebuilder #92: odroid u2/u3 CIFS module not loaded?

gociii, 2022-10-26:
I'm having a bit of trouble invoking CIFS kernel module, getting the following output off commands:

_when mounting_ 
localhost:~# mount.cifs -vvvvv -o user=user,pass=pass,vers=3.0 //nas/backup /mnt/backup
mount error: cifs filesystem not supported by the system
mount error(19): No such device
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)

localhost:~# modprobe cifs
modprobe: ERROR: could not insert 'cifs': Exec format error

localhost:~# find /lib/modules/ -name cifs.ko
/lib/modules/5.19.1-stb-exy+/kernel/fs/cifs/cifs.ko

localhost:~# find /lib/modules/ -name cifs.ko
/lib/modules/5.19.1-stb-exy+/kernel/fs/cifs/cifs.ko

Any ideas please?

hexdump0815, 2022-10-26:
@gociii - i'm surprised that there is an cifs.ko at all:
```
$ grep CIFS config.exy
# CONFIG_CIFS is not set

gociii, 2022-10-28:
Thanks for your reply @hexdump0815 

I guess "CONFIG_CIFS=y" needs to be set in .config file prior to building the whole thing...

How much free space is typically required for the build?

hexdump0815, 2022-10-29:
@gociii - fyi: i have added cifs as a module for future kernel builds: https://github.com/hexdump0815/kernel-config-options/commit/0fdca8b1cf12171abf61d852bd8a6a0f30452303

gociii, 2024-03-27:
Issue resolved in bookworm Spet 2023 release. Tested on odroid-u3+

closed 2024-03-27 by gociii