velvet-os/imagebuilder #93: Help! U3 kernel panic with mmc works fine with sdcard (or mmc on sdcard adapter)

McNova, 2022-10-26:
U-Boot 2021.01+dfsg-3ubuntu0~20.04.4 (Sep 21 2021 - 15:55:38 +0000)

CPU:   Exynos4412 @ 1 GHz
Model: Odroid based on Exynos4412
Type:  u3
DRAM:  2 GiB
LDO20@VDDQ_EMMC_1.8V: set 1800000 uV; enabling
LDO22@VDDQ_EMMC_2.8V: set 2800000 uV; enabling
LDO21@TFLASH_2.8V: set 2800000 uV; enabling
MMC:   sdhci@12530000 - probe failed: -19

Loading Environment from MMC... *** Warning - bad CRC, using default environment

Boot device: MMC(0)
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
no mmc device at slot 2
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf                                                                                                              
1156 bytes read in 4 ms (282.2 KiB/s)                                                                                                                 
1:      linux                                                                                                                                         
Retrieving file: /extlinux/../initrd.img-5.18.1-stb-exy+                                                                                              
14439296 bytes read in 504 ms (27.3 MiB/s)                                                                                                            
Retrieving file: /extlinux/../zImage-5.18.1-stb-exy+                                                                                                  
8737264 bytes read in 307 ms (27.1 MiB/s)                                                                                                             
append: earlycon console=ttySAC1,115200n8 console=tty1 mem=2047M smsc95xx.macaddr=ba:5d:6d:41:68:6f root=LABEL=rootpart ro loglevel=8 rootwait net.ife
Retrieving file: /extlinux/../dtb-5.18.1-stb-exy+/exynos4412-odroidu3.dtb                                                                             
54856 bytes read in 6 ms (8.7 MiB/s)                                                                                                                  
Kernel image @ 0x41000000 [ 0x000000 - 0x8551f0 ]                                                                                                     
## Flattened Device Tree blob at 40800000                                                                                                             
   Booting using the fdt blob at 0x40800000                                                                                                           
   Loading Ramdisk to ba0b3000, end bae78380 ... OK                                                                                                   
   Loading Device Tree to ba0a2000, end ba0b2647 ... OK                                                                                               
                                                                                                                                                      
Starting kernel ...                                                                                                                                   
                                                                                                                                                      
[    0.000000] Booting Linux on physical CPU 0xa00                                                                                                    
[    0.000000] Linux version 5.18.1-stb-exy+ (root@armv7l) (gcc (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP PR2
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d                                                                        
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache                                                               
[    0.000000] OF: fdt: Machine model: Hardkernel ODROID-U3 board based on Exynos4412                                                                 
[    0.000000] earlycon: exynos4210 at MMIO 0x13810000 (options '')                                                                                   
[    0.000000] printk: bootconsole [exynos4210] enabled                                                                                               
[    0.000000] Memory policy: Data cache writealloc                                                                                                   
[    0.000000] Reserved memory: created DMA memory pool at 0xbf700000, size 8 MiB                                                                     
[    0.000000] OF: reserved mem: initialized node region-mfc-right, compatible id shared-dma-pool                                                     
[    0.000000] Reserved memory: created DMA memory pool at 0xbd300000, size 36 MiB                                                                    
[    0.000000] OF: reserved mem: initialized node region-mfc-left, compatible id shared-dma-pool                                                      
[    0.000000] cma: Reserved 96 MiB at 0xb4000000                                                                                                     
[    0.000000] Samsung CPU ID: 0xe4412220                                                                                                             
[    0.000000] Zone ranges:                                                                                                                           
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000006fffffff]                                                                                 
[    0.000000]   HighMem  [mem 0x0000000070000000-0x00000000bfefffff]                                                                                 
[    0.000000] Movable zone start for each node                                                                                                       
[    0.000000] Early memory node ranges                                                                                                               
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000bd2fffff]                                                                                
[    0.000000]   node   0: [mem 0x00000000bd300000-0x00000000bfefffff]                                                                                
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000bfefffff]                                                                       
[    0.000000] Running under secure firmware.                                                                                                         
[    0.000000] 8<--- cut here ---                                                                                                                     
[    0.000000] Unable to handle kernel paging request at virtual address 3ae78374                                                                     
[    0.000000] [3ae78374] *pgd=00000000                                                                                                               
[    0.000000] Internal error: Oops: 5 [#1] PREEMPT SMP ARM                                                                                           
[    0.000000] Modules linked in:                                                                                                                     
[    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 5.18.1-stb-exy+ #1                                                                             
[    0.000000] Hardware name: Samsung Exynos (Flattened Device Tree)                                                                                  
[    0.000000] PC is at memcmp+0x58/0x74                                                                                                              
[... 9 lines cut ...]
[    0.000000] Register r1 information: non-slab/vmalloc memory                                                                                       
[    0.000000] Register r2 information: non-paged memory                                                                                              
[... cut ...]
[    0.000000] Code: e2844004 e2811004 e3520003 9affffed (e5940000)                                                                                   
[    0.000000] ---[ end trace 0000000000000000 ]---                                                                                                   
[    0.000000] Kernel panic - not syncing: Attempted to kill the idle task!                                                                           
[    0.000000] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---   

This dump is from [210803-01] but more recent does the same.

hexdump0815, 2022-10-26:
@McNova - to me it looks like there is still some other u-boot on the emmc - see https://github.com/hexdump0815/u-boot-misc/blob/efb094f6f9981407d67255450607279a41c6e351/readme.exy#L38-L52 for how to write u-boot to emmc - the u-boot.dd to write can be found following https://github.com/hexdump0815/imagebuilder/blob/main/systems/odroid_u3/get-files.sh

good luck and best wishes - hexdump

christian-vallieres, 2022-10-26:
@hexdump0815 you we're right! I didn't have the right u-boot, it was for another image. Easy fix :

`dd iflag=dsync oflag=dsync if=/dev/mmcblk0 bs=512 count=1300 skip=1 of=/dev/mmcblk1boot0`

hexdump0815, 2022-10-28:
i guess we can close this issue then? - please reopen if required.

closed 2022-10-28 by hexdump0815