velvet-os/imagebuilder #95: Dualbooting?

cdd100, 2022-10-30:
I have a lenovo C330 with a 64gb EMMC. Is it possible to dualboot chrome os and ubuntu on the emmc and get a selection menu on boot? Thanks.

aneeshlingala, 2022-10-31:
@cdd100 You can do this by flashing Ubuntu (not chromeos) onto a SD Card or usb (SD Card preferred), but you won't get a selection menu, instead you have to press CTRL U to boot into Ubuntu, or press CTRL D to boot into chromeos, but make sure that you need to have developer mode enabled.

cdd100, 2022-10-31:
I have ubuntu on a usb already and it works fine, I was wondering if I could dualboot on to the emmc and not need to carry a usb with me, since I need to keep chrome os.

hexdump0815, 2022-10-31:
i think to have both on emmc will not work as chromeos is not allowing something like this as far as i know ...

cdd100, 2022-11-01:
Ok, thanks!

aneeshlingala, 2022-11-05:
I'd recommend a sd card, since it is hard to notice without trying to find it, and also don't try to format it from ChromeOS, or it will not work anymore.

closed 2022-11-17 by cdd100