velvet-os/imagebuilder #96: Question about dd

cdd100, 2022-10-31:
I might install Ubuntu to my emmc, but I was wondering If I could use dd if my usb drive is bigger than my emmc.

hexdump0815, 2022-10-31:
@cdd100 - not if you already extended the rootfs ... also in general it is not the best idea to dd the image to emmc as the btrfs used might get confused in case you have two times a filesystem with the same uuid ... it is better to do the installation basically following the doc/install-to-emmc... doc (which sadly is a bit chaotic and needs arewrite)

good luck and best wishes - hexdump

hexdump0815, 2022-11-21:
@cdd100  - i plan to maybe write this down somewhere as well, but it might take some time ... in short: create the partitions like in this document https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L34-L43 create filesystems on them and then mount them and rsync you usb/sd booted system to them like https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L156-L157 ... then adjust the /etc/fstab on emmc and dd the vmlinux kpart to the first partition of the emmc ... this might work, but i did not test it this way and make sure you understand the steps before trying it

good luck in case you are going to try it

cdd100, 2022-11-23:
I managed to write the image to the disk by copying the extracted .img to my usb and restoring it with gnome disks.

cdd100, 2022-11-23:
Not a huge issue, but the battery is always at 100% for some reason. Is it possible to fix this?

hexdump0815, 2022-11-23:
@cdd100 - battery percentage should just work - at least it does for my test oak systems ... also dumping the image to emmc is a bit dangerous as it might confuse the btrfs filesystem to have two of them around with the same filesystem uuid - for that its best to boot an older image and write a newer one to emmc, this way the roofs of both images have different filesystem uuids as they were generated seperately

please do not change issue topics as it might be complicated for others to follow what an issue is about, its better to open a new issue for new topics

closed 2022-11-23 by cdd100