velvet-os/imagebuilder #97: Wifi dongle not working

cdd100, 2022-11-14:
I have tried Ubuntu 22.04 on my lenovo c330, and it works fine except for the fact that my wifi dongle wont work. Somehow the internal wifi of my chromebook is broken, so I ordered a wifi dongle (panda wireless PAU03 - Yes I understand it is uncommon) and it worked perfectly fine with chrome os. When I go to the settings in Ubuntu, there is no wireless options at all except for VPN, and the only way I can get internet is by attaching my phone and enabling usb tethering.

hexdump0815, 2022-11-14:
@cdd100 - what chipset is it using? maybe the driver for it is not yet compiled into the kernel?

cdd100, 2022-11-15:
It uses a ralink RT2870 chipset.

cdd100, 2022-11-21:
where would I find the drivers?

hexdump0815, 2022-11-21:
@cdd100 - the kernel needs the CONFIG_RT2X00 config option set which is not the case for most of the kernel builds used here and among them is the one for the oak chromebooks ... i'll put it onto my todo list to add it to the default kernel config for future kernels (i plan to build new ones a bit after v6.1 is out) ... in case it is needed earlier then a new kernel has to be built

cdd100, 2022-11-23:
Are there any wifi adapters that would work with the current kernel build?

hexdump0815, 2022-11-23:
@cdd100 - most should work - i tried to add them some time ago - https://github.com/hexdump0815/kernel-config-options/blob/main/additional-options-generic.cfg#L60-L101 ... i just seemed to overlook the ralink ones back then

cdd100, 2022-11-29:
Bought a TP-link TL-WN725N V1 and it now connects to wifi, but repeatedly disconnects and asks for the password on some wifi networks. Is linux itself causing this, the adapter, or the wifi network itself? Seems the only workaround is by running speedtest.net in the background and it makes it work for some reason. It uses a RTL8188 chipset.

hexdump0815, 2022-11-29:
@cdd100 - i think some of the linux network drivers are not of the best quality and for that reason also live in the staging directory of the linux kernel sources - not sure if this is the case here, but in general finding a perfectly working usb wifi dongle for linux seems to not be that easy ... maybe pinging some website is enough to keep it alive? maybe there others had the problem too and you can find some info by searching on the net?

cdd100, 2022-11-29:
I could try to make a script to ping Google constantly Hopefully it will work.