velvet-os/imagebuilder #98: Battery always at 100 percent

cdd100, 2022-11-23:
On my lenovo chromebook c330 the battery is always at 100 percent. Not a huge issue but the only way I can judge the battery is by looking at the light next to the charging port, which is not precise. What would be causing this?

hexdump0815, 2022-11-24:
battery percentage should just work - at least it does for my test oak systems (just my old response from the other issue for completeness)

cdd100, 2022-11-24:
Huh... I wonder what is causing it? In gnome, it sits at estimating and xfce says it has 10,000 hours remaining.

hexdump0815, 2022-11-26:
i have no real idea actually - maybe looking into /sys/class/power_supply and the kernel log via dmesg gives some hints ...

cdd100, 2022-11-26:
[image]

This error about sbs-battery 6-00b shows multiple times.
Seems like it is using a fallback.

cdd100, 2022-11-26:
This error too:
[image]

hexdump0815, 2022-11-26:
some of such errors seem to be normal  - what does "find /sys/class/power_supply/sbs-*/ -mindepth 1 -maxdepth 1 -type f -exec cat {} \;" result in? maybe some errors in it?

cdd100, 2022-11-27:
All I can get is "find: missing argument to `-exec'"

cdd100, 2022-11-27:
It can at least tell when being plugged in.

hexdump0815, 2022-11-27:
sorry, i should have quoted it properly and as a result the backslash got lost - i fixed it now above

cdd100, 2022-11-27:
Here is the output:
`3932100
65535000
ffff
cat: /sys/class/power_supply/sbs-6-000b/technology: Invalid argument
-1000
65535000
1
65535000
cat: /sys/class/power_supply/sbs-6-000b/manufacturer: Invalid argument
65535000
655350000
Battery
65535000
2107
100
31
Calibration required
3932100
65535
65535000
Discharging
cat: /sys/class/power_supply/sbs-6-000b/model_name: Invalid argument
-1000
65535000
62804
65535000
0
Full
655350000
3932100
15
655350000
`

hexdump0815, 2022-11-27:
i do not have my oak systems at hand right now - will check on them during the next days ... just an idea: maybe let the battery run down completely once and afterwards charge it completely to recalibrate the battery logic - maybe that makes things better - inspired by the "Calibration required" ...

harmw, 2023-07-26:
Running into the exact same issue on both postmarketos latest (kernel 6.something) and this debian project. Running an older kernel 5.10.50 since the latest produced a boot disk oops. Boots fine but the battery readings are garbage and exactly as described in this issue.

Fwiw this battery is a replaced part.

I'd like to debug this, just not a clue where to start 😂

hexdump0815, 2023-07-26:
sadly i have no real idea what to do about this - for me it works fine on arm ... as far as i understand it there is a small controller in the battery which is taking care of everything around charging and status and which is asked about the battery percentage for instance, so i guess either this controller does not work well anymore or the communication with it is not fully ok anymore ... one possible test would be to check it from chromeos side, if you see the same problem there as well or not

harmw, 2023-07-27:
just did a clean recovery to chromeos (kernel `4.19.281` / `aarch64`), and battery is fine there. From the `power_supply_info` cmd I can now see fields like voltage, energy, current and charge showing proper values (instead of bogus values that seem to indicate some int maxing out at 65.535)

This time, I also see the technology (Li-ion), vendor (LGC) and model name (L15L3PB1) fields - those were all unavailable in regular linux.

Path is `/sys/class/power_supply/sbs-6-000b`, which I suspect is the exact same on ordinary Linux (probably because it's from a kernel interface?).

When I execute `power_supply_info` (without `sudo`) It returns the regular stuff from `sysfs`, _but_ it also prints a little error:

```
[cros_ec_prefs_source.cc(23)] Failed to open /dev/cros_ec: Permission denied (13)
```

No idea if this is relevant though 😅

hexdump0815, 2023-07-29:
thanks a lot for checking - so this way a hardware problem is ruled out, so most probably somehow your battery is not properly supported by mainline - is it a default battery or maybe a replaced one?

harmw, 2023-07-30:
Battery is still original but the touchpad was replaced due to a malfunction.

hexdump0815, 2023-07-31:
sadly i have no real idea how to deal with that  - i'm currently working on testing the v6.4 kernel, so maybe it might be worth a try to test that once i have it tested working ...

tuccimon, 2025-04-27:
@hexdump0815 just adding that I have the exact same problem with my C330. Since it's EOL in June 2025, I figured I'd look for alternatives, and I stumbled upon your project. I have tried your latest image (Sept 2023), Postmarket OS, and Arch Linux ARM. The latter kept looping and displaying the same errors over 2 minutes until I killed the boot process (same error that user @cdd100 posted earlier).

Also, wanted to say thanks for all the hard work you are doing with this project.