velvet-os/imagebuilder #102: is it possible to change the linux distro?

cdd100, 2022-12-15:
Would it be possible to replace the debian rootfs with an arm64 fedora rootfs and keep the other partitions? Just interested.

hexdump0815, 2022-12-15:
@cdd100 - you just need /boot /lib/modules and maybe some files or tweaks for your system from the imagebuilder/systems/yoursystem dir and it should work with any common rootfs i would say ... so if you are familiar with doing such things it should not be a bit problem

cdd100, 2022-12-16:
Ok, I am going to try to install a fedora silverblue rootfs. Hopefully it works. If not, I will try fedora workstation.

cdd100, 2022-12-16:
It just gives me a kernel panic saying "no working init system found".

cdd100, 2022-12-17:
I have done some more experimenting and it seems that any tampering with the system files (such as /usr or even /home) I get a kernel panic about no working init found. Does it have to do with permissions?

cdd100, 2022-12-17:
I was able to covert to bedrock linux although.

hexdump0815, 2022-12-17:
@cdd100 - sorry, can't help with this due to no time and no real fedora experience - i remember silverblue was something with a readonly rootfs maybe, so that would be way more complex, better try with some more normal version ... good luck

cdd100, 2022-12-17:
I am just gonna stick to bedrock linux as it works extremely well.

closed 2022-12-17 by cdd100