velvet-os/imagebuilder #103: use ext4 for chromebook disk images

bsammon, 2022-12-27:
I notice that ChromeOS (on my Lenovo Duet 5) cannot read the root partition on the thumbdrive I made from your image.  I'm guessing/assuming that this problem is not specific to the Duet.
If the images were configured to be ext4 instead, then they could be read from ChromeOS.
I think that would be an improvement.
I'm not too familiar with btrfs, and the advantages it has over ext4.

hexdump0815, 2022-12-28:
@bsammon - the rootfs type can be defined at image build time - for trogdor for instance here: https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/partition-mapping.txt#L2 ... i do not plan to go away from btrfs any more as it simply has so many advantages over ext4 like:
- it allows automatic compression of the data written to the filesystem - this is enabled by default in the images and allows to put close to twice as much data onto the disks as without compression (depending of the data of course) which makes a big difference for systems with only a 16gb or 32gb emmc or sd card - a nice side effect of this is: if the data is written compressed to storage, less writes are required which is good for the life time of sd, emmc, ssd etc
- btrfs is automatically checksumming its data and checks those checksums on read, so one actually has a chance to notice when the storage starts to die while this will most probably happen silently with ext4
- btrfs has lots of very nice features beyond ext4 (snapshots, volume management(, but not all of them are really stable - this is why i only use the basic btrfs ufunctionaly in the images.

if you want to exchange data with chromeos maybe add an extra ext4 partition or use an sd card / usb drive with ext4 for exchanging files

good luck and best wishes - hexdump

bsammon, 2022-12-28:
> if you want to exchange data with chromeos maybe add an extra ext4 partition

But be very careful using parted to make the partition...

hexdump0815, 2022-12-28:
@bsammon - i'll for sure test things carefully before changing anything :)