velvet-os/imagebuilder #104: kernel image debs

bsammon, 2022-12-28:
It would be handy if there were kernel image packages available as part of your releases.  Either already-installed packages in the releases, or deb files I could download separately.

I expect/assume there's some process that I can learn to just build the packages myself -- it's on my todo list -- so this isn't high priority, just a "would be nice" request.

My use case is that I would like to experiment with kexec.

Or maybe it's possible to pull/create kexec-usable images from the kernel partitions in the disk images.

hexdump0815, 2022-12-28:
@bsammon - there are no plans to provide kernel packages right now as this would simply be too much effort and time (which i sadly do not have) to provide this and even more so keep it up to date - instead i prefer to encourage people to build their own kernels which is not that complicated once one gets used to it ... i tried to write some documents to at least get started with that topic:

https://github.com/hexdump0815/imagebuilder/blob/main/doc/building-own-kernels.md
https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md
https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-kernel-headers.md

please let me know or even better create pull request with fixes in case you find any errors or mistakes in those documents

good luck and best wishes - hexdump

bsammon, 2022-12-28:
I definitely understand the not-wanting-to-add-extra-work thing.

Here's a thought -- since you're already building kernels, howabout just putting the kernel vmlinux file you built somewhere we could download, in addition to making it part of the image?  Without the extra work of making it a .deb file, that seems like it might not be much work.

As for building my own kernel, the main obstacle/question I have is which kernel source tree (or patch-set) to use.  I haven't read all your documentation yet, but I haven't found (or maybe just haven't understood) the answer to this.  Those three documents you link to here match my recollection from building 2.0 kernels (from kernel.org tarballs) back in the day, but doesn't cover the specifics.

hexdump0815, 2022-12-28:
@bsammon - in the system subdir for each system in the repo there is a readme which points to a readme for building the kernel, sometimes for u-boot and for mesa too if applicable - this readme is in another repo which has all the patches required etc. as well and in the releases section of that repo you can find some kernels i built and mostly tested in a tar file

hexdump0815, 2022-12-28:
@bsammon - here are the stops to checkout the kernel sources for trogdor:
```
cd /compile/source
git clone https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git
mv linux-stable linux-stable-cbq
cd linux-stable-cbq
git checkout v6.1.1
```
for trogdor no extra patches are required if v6.1.1 is used (should work, but not 100% sure) - for v5.19.1 just use the readme from https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/tree/5.19.1-stb-cbq+ (i.e. the 5.19.1-stb-cbq+ tag of that repo)

good luck and best wishes - hexdump