velvet-os/imagebuilder #105: No soundcards detected in latest image

JackBinary, 2022-12-31:
Purchased a Lenovo Flex 3 for a fun side project: an ARM-powered linux laptop. Almost everything works perfectly, save for audio. cat /proc/asound/cards returns "--- no soundcards ---"

output of cat /proc/device-tree/model is "Google fennel14 sku0 board"

what might I be missing?

hexdump0815, 2023-01-01:
@JackBinary - can you please put a dmesg somewhere onto a pastebin and link it here to see if there is anything in it pointing towards potential problems? also please check, if /etc/rc.local in the image you used is handling the audio setup for you system (i.e. that it gets detected there properly) - maybe run "bash -x /etc/rc.local" as root to see if it does what it is supposed to do for your system. the latest images should have it in already, but it might miss some last minute changes or still have bugs. this issue might also be an interesting read about debugging such sound problems maybe: https://github.com/hexdump0815/imagebuilder/issues/54

good luck and best wishes - hexdump

JackBinary, 2023-01-02:
[https://pastebin.com/H7ZCGh79](url)
One notable line for me was the following:
`[    8.822911] mt8183_da7219 mt8183-sound: ASoC: failed to instantiate card -6`

running `bash -x /etc/rc.local` gave me the following:
[https://pastebin.com/5A2VCDw7](url)
which also states no sound cards found.

In the meantime, I'll take a look at #54 as suggested.

hexdump0815, 2023-01-03:
@JackBinary - oh - that error i have not seen yet - could you maybe try my latest mt8183 kernel from https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.1.0-stb-mt8%2B maybe? some info about how to install this newer kernel can be found here: https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md ... please let me know if anything around that document is not really clear - documentation is not really my strength :)

best wishes and good luck - hexdump

JackBinary, 2023-01-04:
thanks for the help so far, and fully understandable about the documentation, I'm no better when it comes to it. Unfortunately, following the instructions, while I was able to extract and update the initramfs, I was unable to actually get the new kernel to boot. I keep booting to 5.19

hexdump0815, 2023-01-04:
@JackBinary - i just reread my kernel update instructions and noticed that one step (the dd of the kpart image to the first partition) was not explicitely mentioned and updated them now, so maybe have a look at the instructions again.

JackBinary, 2023-01-08:
Sorry for the late response, I came down with an illness. Unfortunately, while I am now able to boot into the new kernel, it has not changed the status of the audio problem.

hexdump0815, 2023-01-08:
@JackBinary - best wishes for you to get better ... there is no time pressure here :)

i have no real idea yet what to do with this - i'll try to see if i can get access to this hardware somehow to look deeper into it ... in case you find out anything more, please let me know