velvet-os/imagebuilder #108: Failed to create filesystem.

Demon-Gyoza, 2023-01-14:
First, I ran:
`sudo ./get-files.sh aarch64 chromebook oak jammy`
Then:
`sudo ./create-fs.sh chromebook_oak aarch64 jammy`
But it throws this at me:
```
mknod: /compile/local/imagebuilder-aarch64-jammy-cache/test-dev-null: Operation not permitted
E: Cannot install into target '/compile/local/imagebuilder-aarch64-jammy-cache' mounted with noexec or nodev

error while running debootstrap - giving up
```
Know why this is going on?

hexdump0815, 2023-01-15:
this looks like you are running it on chromeos, which has some special security to for instance not allow to execute any command on external media or maybe even local media - can this be the case?

robots, 2023-02-05:
same problem here. Tried to remount / to -o exec,dev, but permission denied.

hexdump0815, 2023-02-06:
@robots - i wouln't be surprised if it simply would not work on chromeos due to the security features ... maybe it might work in one of those native chromeos linux vms?

robots, 2023-02-07:
it didnt work in virtual linux on chromeos either. I ended up using older image for trogdor and updating it manually.

hexdump0815, 2023-02-07:
@robots - the best option for using the framework is to use it on one of its images - this way i always build images and this should be tested to work quite well - usually having a 32gb sd card / usb stick should be enough. so simply downloading the latest image, getting it running and then building newer images on it should work most of the time.

hexdump0815, 2023-02-07:
@robots - to make things a bit easier i sat down and started to write at least some initial information about how to use the imagebuilder framework at https://github.com/hexdump0815/imagebuilder/blob/main/doc/using-the-imagebuilder-framework.md ... feedback, fixes and improvements are welcome.

closed 2023-03-29 by Demon-Gyoza