velvet-os/imagebuilder #110: Unable to boot image on Veyron_Mighty

unixabg, 2023-01-17:
Greetings,
I am new to your project and I want to say thank you for the project. I have interest in getting several of the edugear chromebook m series, Google_Veyron_Mighty booting with your project. More details about device below:
 
Machine Model: mighty
  Write Protect: HW=1 SW=1
  Last Boot Version: RO=Google_Veyron_Mighty.6588.160.0 ACT/A=Google_Veyron_Mighty.6588.197.0
  Firmware Updater:  RO=null    RW=null

I have downloaded and dd both the ubuntu and debian images from https://github.com/hexdump0815/imagebuilder/releases/tag/220820-01 .

Then I gunzip the download. Next I write to usb drive with:

dd if=imageFile.img of=/devsdX

With the chromebook in developer mode, I hit ctrl+u to boot the usb key and nothing happens except the screen changes to backlit and the units do not boot. I might be missing something simple, but is there anyway to debug why they are not booting?

Thank you.

hexdump0815, 2023-01-17:
@unixabg - mighty should work as a mighty was the machine i tested the images with (medion s2015) ... do you get the u-boot prompt? if not, maybe try to dd the different u-boot images in the extra dir of the 4rd partition onto the first - maybe one of them works better?

good luck and best wishes - hexdump

hexdump0815, 2023-01-17:
@unixabg - another idea: please try different sd card types - sometimes older and smaller ones work better than newer and larger ones on those older machines ...

unixabg, 2023-01-18:
Greetings,

Thank you for quick response. I will reply inline.

> @unixabg - mighty should work as a mighty was the machine i tested the images with (medion s2015) ... do you get the u-boot prompt?

No.

> if not, maybe try to dd the different u-boot images in the extra dir of the 4rd partition onto the first - maybe one of them works better?

I might be missing something simple here but after writing the image to the usb key, on my intel Linux dev machine with amd64 architecture, I see:

db                          8:16   1  7.5G  0 disk  
├─sdb1                       8:17   1   32M  0 part  
├─sdb2                       8:18   1   32M  0 part  
├─sdb3                       8:19   1  512M  0 part  
└─sdb4                       8:20   1  2.9G  0 part 

But I am unable to mount the partition 4, hence can not attempt the dd of extra u-boot images.

> 
> good luck and best wishes - hexdump

I need it. I am sure I am missing something simple. And please know I appreciate your assistance.

unixabg, 2023-01-18:
Greetings,

Update, I switched to sd card and found on partition 3 the extra folder you described. Then I went one at a time on the uboot and dd the images and test. And viola! The magic one was minnie:

dd if=uboot.kpart.cbr-minnie of=/dev/mmcblk0p1

hexdump0815, 2023-01-19:
@unixabg - one more question: which dtb are you using now in the end: the mighty or the minnie one?

unixabg, 2023-01-21:
@hexdump0815 - I am booting the extlinux entry of mighty bu using the u-boot image of minnie

hexdump0815, 2023-01-21:
@unixabg - ok thanks a lot for this info - so its similar to my medion mighty: jerry u-boot and mighty dtb ... just wanted to make sure that yours is not a full minie in the end also using the minnie dtb besides the minnie u-boot