velvet-os/imagebuilder #112: Is there an irc channel for the project?

unixabg, 2023-01-18:
Greetings,
I am curious if there is an irc channel for the project? If not, I could set one up at oftc and just want to confirm a channel name.
Thank you for the project and your consideration.

hexdump0815, 2023-01-18:
@unixabg - there is none and i'm already struggling to keep up with all the irc channels i'm already following :) ... i prefer to keep it with the github issues for now in case someone has problems with or ideas for the images ... as your focus seems to be on chromebooks then maybe this one could come close: https://oftc.irclog.whitequark.org/linux-cros-arm/ - it is in general about running mainline linux on chromebooks, so also postmarketos, archlinux etc.

if you want to setup a channel for this project here specifically feel free to do so as long as you maintain it and if needed moderate it as i have not enough time for that right now

best wishes - hexdump

unixabg, 2023-01-18:
@hexdump0815 - thank you for the reply. I will checkout the channel. On special channels and keeping up I understand. Hence I think I will try the suggested for now. Marking issue closed.

closed 2023-01-18 by unixabg