velvet-os/imagebuilder #116: How to install to the chromebook internal emmc drive?

unixabg, 2023-01-20:
Greetings,

Is there a simple way to install to the chromebook internal emmc after getting booted from sd card?

hexdump0815, 2023-01-20:
@unixabg - as this seems to come up again and again, i think i'll have to write some notes about this, but it still might take some time ... there is this: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt which i'm always using for setting up new chromebooks for myself, but it is very hard to read as it covers way more than just a simple chromebook install ...

unixabg, 2023-01-30:
@hexdump0815 - I think I understand what to do. I will try out on my chromebook_veyron eduGear m4 and report back. If I can make work I will attempt to help write something up using the doc you linked as a template.

hexdump0815, 2023-01-31:
@unixabg - oh thats nice to hear that my notes are someone useful still in the end :)

hexdump0815, 2023-01-31:
@unixabg ... and congratulations to your new linux notebook :) - running linux from emmc on chromebooks makes them surprisingly capable devices in my opinion

unixabg, 2023-02-10:
@hexdump0815 - so I am able to install manually. I have also tried just to dd the imagebuilder-prebuilt.img file to the emmc, but I can't seem to make that work. Is there a reason why one can not just dd the imagebuilder-prebuilt.img to the emmc?

hexdump0815, 2023-02-10:
@unixabg - i think at least one problem is that the images use btrfs and btrfs has a uuid per filesystem and gets confused whenever the same uuid appears twice ... a possible workaround might be to dd the latest image from a booted older image (i.e. two different btrfs fs uuids in use)

mrwhistles, 2023-02-15:
For what it's worth I was able to "dd" the "kukui" image directly to the emmc of my hp chromebook 11a and everything is working perfectly. I booted from a newly imaged USB drive, downloaded the image to the live system, imaged the emmc and rebooted.

unixabg, 2023-02-15:
@mrwhistles - first thank you for the response! It brings me hope that we can just do this instead of an install script I have been trying to get working on chromebook_veyron eduGear m4. I think with the model chromebook_veyron eduGear m4, the issue is related to needing to seek beyond the read only area per https://github.com/hexdump0815/imagebuilder/blob/85d97b2e9293e616ee839ff276971a588a5a72a5/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L41  .  It might be trivial, but could you post the command you used to dd for others who may not know exactly what to do? Again thank you for responding to this thread and just a simple dd is the best route I think if we can get it documented well (includeing special cases like chromebook_veyron eduGear m4).

mrwhistles, 2023-02-16:
Hey, of course. 

Snag your preferred release (I ultimately picked Jammy). I ended up downloading chromebook_kukui-aarch64-jammy.img.gz

Uncompress the archive:
# gunzip  chromebook_kukui-aarch64-jammy.img.gz

Use something like fdisk to discover the device name if your internal storage:
# sudo fdisk -l
Here's mine for example:
"Disk /dev/mmcblk0: 29.12 GiB"

Use dd to write the image:
# sudo dd if=chromebook_kukui-aarch64-jammy.img of=/dev/mmcblk0 bs=64k

Reboot and at the developer mode screen press ctrl+d like normal to boot Ubuntu.

unixabg, 2023-02-17:
Greetings,
@mrwhistles - well in the end, it looks like the bs=64k must have been the magic and no need to seek on the dd command for the chromebook_veyron eduGear m4. Again thank you for posting, this makes things very easy. I am not sure whether to leave this open or close. Mabye @hexdump0815 has a preference.

hexdump0815, 2023-02-17:
@unixabg @mrwhistles - i think on veyron the dd option might not work as on emmc the beginning is not properly writable on this device (you can write, but cannot read it back properly - as a result the primary gpt partition table will be broken), but i might give it a try to create the veyron image even for the sd card keeping this in mind so that the image can be dd'ed easily to emmc as well - i'll try to play around with this over the weekend maybe ... i'm close to building new images anyway, so if it works i can adjust the new image then already accordingly

for dd'ing i would still recommend to use two different images for booting and for the one to dd, so that both have different btrfs fs uuids - it might still work with the same uuid as long as the emmc root fs is not mounted directly afterwards, but if it is mounted directly afterwards i guess it will result in trouble as btrfs cannot properly handle two filesystems with the same uuid and will get confused as far as i know ... but as said: simply using two different images should work around this problem as for each image a new btrfs fs is being created resulting in a different fs uuid

@unixabg - one thing i would definitely recommend for your plan to put linux to emmc on those chromebooks: remove the write protect screw in the chromebooks and set the gbb flags to 0x19 as described in the corresponding section here: https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks - especially those older chromebooks tend to forget all their dev mode, allow boot from usb, no beep etc. settings when the battery drains completely and the only way to get out of this is to restore chromeos via restore disk and set everything there again, which will overwrite all data on your emmc, so no chance to access the data beforehand anymore in such a situation - setting the gbbflags will set the chromebook to dev mode, allow boot from usb, no beep etc. as default in hardware - so even if it forgets everything, those settings will still be intact and you can boot from usb/sd at any time and you will not loose dev mode by accident this way

i'll update after my experiments with the new images during the next days - best wishes and good luck - hexdump

hexdump0815, 2023-02-18:
@unixabg - i did think a bit more about this and it looks like it is not possible to resolve this problem in the sd card image alone and some extra steps to fix up the partitions will be required after the dd for veyron devices ... i plan to finally write down some readme about how to install the images to emmc on chromebooks in different ways this weekend and will add a sections for those veyron specials in it as well - i'll let you know here when the document is committed to the tree ...

hexdump0815, 2023-02-18:
@unixabg - quick update, i finally managed to build a new image for veyron - sd-card/usb only still for now, emmc install doc will follow tomorrow ... feedback welcome if it works for you from sd/usb (did not have the time to test it myself yet) - https://github.com/hexdump0815/imagebuilder/releases/tag/230218-05

unixabg, 2023-02-19:
@hexdump0815 - just downloaded, dd to sdcard, and boots with uboot.kpart.cbr-minnie on sdcard p1. All appears in general good order thus far. Thank you for your help on extending the life of these devices. Let me know if you need any further testing on this model.

hexdump0815, 2023-02-19:
@unixabg - thanks a lot for the testing and the feedback ... as promised here is a first draft of a document describing some ways to install to emmc on arm chromebooks - feedback and testing (if possible of both described ways) would be very welcome ... also there might be still typos in it :) ... https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

best wishes and good luck - hexdump

hexdump0815, 2023-02-19:
@unixabg - i just found one little missing piece in that document and added some more little details - latest version is commited now ... just fyi

unixabg, 2023-03-03:
@hexdump0815 - I believe this issue has been sufficiently documented and I am marking closed.

closed 2023-03-03 by unixabg