velvet-os/imagebuilder #118: How to change the (desktop) environment?

IOOI-SqAR, 2023-02-04:
Hi there,

at first I want to thank you for your work! On my own I wouldn't be able to create a modern image of Linux for my Odroid U2! I am using the [odroid_u3-armv7l-jammy.img.gz](https://github.com/hexdump0815/imagebuilder/releases/download/220824-01/odroid_u3-armv7l-jammy.img.gz) image of August 24 of last year.

Everything worked fine so far besides replacing the desktop environment. I am just not a fan of XFCE and wanted to replace it with something else using `sudo tasksel`. I could install another environment but it didn't work out as it didn't replace XFCE. Then I tried uninstalling XFCE using `sudo apt-get purge xfce4` but it seems that XFCE wasn't installed that way but is somehow "hardwired" into the image. Long story short: I was unable to get rid of XFCE!

How about releasing an "server" image and letting the user install the environment of their choice using `tasksel`? Just an idea …

Thanks a lot,

Lars

hexdump0815, 2023-02-04:
@IOOI-SqAR - hi lars, there is nothing special done regading xfce - in theory it should be possible to delete it and install something else ... maybe first install something else and delete it afterwards as as packages like lightdm might depend on having a de around maybe?

here is how xfce gets installed: https://github.com/hexdump0815/imagebuilder/blob/main/scripts/create-chroot-stage-01.sh#L24 - maybe removing it via xubuntu-desktop meta package works better?

i might think about a smaller server only image at some point, but it will require some extra effort to make it work smooth within the framework and i'll need a bit of time for that which is limited right now ... but quite a few people asked for that already

happy to hear that another u2 is running close to mainline again now :)

best wishes - hexdump

IOOI-SqAR, 2023-02-04:
@hexdump0815 somehow I seem unable to switch to the "normal" Ubuntu desktop nor to uninstall xubuntu-desktop properly. Here is what I did:

before it all started and before I opened this issue I did:

```
sudo apt install ubuntu-desktop^
sudo reboot
```

after reading your answer I did:

```
sudo apt-get remove --auto-remove xubuntu-desktop
sudo reboot
```

which removed xubuntu-desktop and all its dependencies (at least it said it did)

but after the reboot the old desktop was back again, albeit with some changes (like now there was suddenly a dock and some other appearance did also change, like the start menu — wherein it still said "About Xfce").

A subsequent 
```
sudo apt install ubuntu-desktop^
sudo reboot
```
didn't help the cause.

So I am looking forward for your (possible) future server release which will hopefully work this out … :)

Kind regards,

Lars