velvet-os/imagebuilder #120: starfive_visionfive2-riscv64-sidriscv.img works. big thank you.

omac777, 2023-02-04:
starfive_visionfive2-riscv64-sidriscv.img.gz 
as available here: https://github.com/hexdump0815/imagebuilder/releases/tag/230115-01

Installed and worked like a charm.  Network is up on it.  Thank you.

omac777, 2023-02-04:
Here's what I did:

```
sudo umount /dev/sdb4
sudo wipefs --all /dev/sdb
sudo dd if=starfive_visionfive2-riscv64-sidriscv.img of=/dev/sdb bs=4M status=progress conv=fdatasync
sudo gparted /dev/sdb
```

Then plugged in the sdcard into the Starfive VisionFive 2 board and powered it up and was able to ssh into it right away no issues.

hexdump0815, 2023-02-04:
@omac777 - thanks a lot for the feedback ... i plan to have a closer look at the gpu stuff next whenever i find a bit of time for it ... this looks like a good starting point for that: https://github.com/riscv/meta-riscv/pull/382

omac777, 2023-02-07:
Have I told you that YOU ROCK today?

omac777, 2023-02-07:
What was the command you used in imagebuilder to launch the create image for starfive_visionfive2-riscv64-sidriscv.img.gz?  Thank you in advance.

hexdump0815, 2023-02-07:
@omac777 - i did not know vmdb2 before you mentioned it ... the image builder framework is something which i created as sideproject form something else and it grew over time to something which allows me to easily build images for many very different devices and it is still quite small and simple ... one thing which is different from other image build frameworks which usually crossbuild everything on x86 hardware is that this one is assumed to run on the same hardware (at least same architecture) and that it uses a prebuilt kernel, but kernel build notes for that kernel are linked in the readme.md for the corresponding system, so that it should not be a problem to build an own kernel (in the doc dir are some more infos about that) ... all this is just a spare time project of mine and thus documentation is not the best in some areas :)

best wishes and good luck - hexdump

hexdump0815, 2023-02-07:
@omac777 - to make things a bit easier i sat down and started to write at least some initial information about how to use the imagebuilder framework at https://github.com/hexdump0815/imagebuilder/blob/main/doc/using-the-imagebuilder-framework.md ... feedback, fixes and improvements are welcome.

omac777, 2023-02-16:
I'll I just tried the latest image from starfive
https://github.com/starfive-tech/VisionFive2/releases/tag/VF2_v2.8.0

I currently prefer your image over the above buildroot based image because buildroot doesn't feel right. It doesn't have apt-get/apt.

omac777, 2023-02-16:
Any possibility of having an apt repo for your stuff that could change for your image?  For example your image has kernel v2.5.0 and visionfive has a kernel 2.8.0 out.  I would prefer to apt-get upgrade a package from your repo rather than having to rebuild the kernel myself.  I'm surprised starfive isn't doing anything like your effort.

The other reason, I prefer this approach is you are also probably getting tired of rewriting images to sdcards as well.  package updating/upgrading makes much more sense.

hexdump0815, 2023-02-18:
@omac777 - i do not plan to provide kernel packages as i simply do not have the time for it - i provide for all systems provided by the framework notes about how to build a kernel (in the readme.md of the corresponding system dir) and recommend that as a way to keep the kernel up to date - the notes for the vf2 have already been update to v2.8.0 some time ago - https://github.com/hexdump0815/linux-starfive-visionfive2-kernel/blob/main/readme.vf2

the kernel build can be done easily done on the vf2 itself, takes about an hour and some more info about building and installing an own kernel can be found in the doc dir

best wishes and good luck - hexdump