velvet-os/imagebuilder #122: Help setting up imagebuilder build environment

unixabg, 2023-02-05:
Greetings,
I have been attempting to setup the build environment. What I have so far is:
1. Made a virtualbox debian install amd64
2. On the booted virtualbox vm I make an arm chroot with (command run as root):
First bootstrap with:
qemu-debootstrap --arch=armhf --include=git,ca-certificates,rsync,parted,gdisk,btrfs-progs bullseye /chroots/bullseye-armhf http://ftp.debian.org/debian/
3. Then I use the below script to mount the chroot  (command run as root):
#!/bin/bash

echo "Changin direcroty to /chroots/bullseye-armhf/"
cd /chroots/bullseye-armhf/

echo "Mounting bindings for dev, proc, sys, pts sid chroot."
mount -t proc /proc proc/
mount -t sysfs /sys sys/
mount --bind /dev dev/
mount --bind /run run/

echo "Starting chroot ..."
chroot ./ /bin/bash

echo "... exited the chroot so time to clean up ..."
umount -l /chroots/bullseye-armhf/dev
umount -l /chroots/bullseye-armhf/proc
umount -l /chroots/bullseye-armhf/sys
umount -l /chroots/bullseye-armhf/run

4. Once in the chroot I clone and cd to imagebuilder/scripts and I cat the readme.md. 
#no errors on prepare.sh
./prepare.sh

#no errors on get-files.sh
./get-files.sh chromebook_veyron armv7l bookworm

#got several errors on this one, listing the ones that may be problematic
./create-fs.sh chromebook_veyron armv7l bookworm
/create-chroot-stage-01.sh: line 32: tasksel: command not found
umount: /compile/local/imagebuilder-armv7l-bookworm-cache/dev/pts: target is busy.
umount: /compile/local/imagebuilder-armv7l-bookworm-cache/dev: target is busy.

<snip>

cp: cannot stat 'usr/share/initramfs-tools/hooks/fsck': No such file or directory
sed: can't read usr/share/initramfs-tools/hooks/fsck: No such file or directory
chroot: failed to run command 'update-initramfs': No such file or directory
mv: cannot stat 'tmp/fsck.org': No such file or directory
umount: /compile/local/imagebuilder-root/dev/pts: target is busy.
umount: /compile/local/imagebuilder-root/dev: target is busy.

#got a couple of errors on this command but image was produced
./create-image.sh chromebook_veyron armv7l bookworm

Create subvolume '/tmp/imagebuilder-mnt/swap'
ERROR: object is not a btrfs object: /tmp/imagebuilder-mnt/swap
ERROR: object is not a btrfs object: /tmp/imagebuilder-mnt/swap/file.0
Setting up swapspace version 1, size = 512 MiB (536866816 bytes)

Them I dd image to sdcard and dd the correct uboot for my hardware
Then attempt to boot sdcard, uboot works and images starts and then stops at:
run-init: can't execute '/sbin/init' : No such file or directory
run-init: can't execute '/etc/init' : No such file  or directory
run-init: can't execute '/bin/init' : No such file  or directory

Ideas welcome.

hexdump0815, 2023-02-06:
@unixabg - i'm a bit short on time tonight, but at least a short question: is there any deeper reason why you try to run it in a qemu vm on x86? in principle the framework is designed to be run natively, i.e. on an armv7l system to build an armv7l image ... i think for you the easiest case would be to dd my image onto a large enough sd card (32g+ i would say) or mount one at /compile in case you run from emmc and do all the steps there - it should be way faster this way than on a qemu emulated arm system on x86 ... i'm building all the armv7l images on an armv7l amlogic s805 mxq android tv box running mainline linux with 512mb ram and using a usb-stick as disk for example :) ... a good side effect of running on one of my images is that everything needed for the framework to work should be around - my mxq tv box is of course also running one of my images as its os ...

good luck and best wishes - hexdump

unixabg, 2023-02-07:
@hexdump0815 - I only attempted as described for convenience on my regular development machine. I will try on velvet os install and report back.

hexdump0815, 2023-02-07:
@unixabg - let me know if anything is unclear - might be a good base for writing some better documentation about how to use the framework. btw. i usually clone this repo to /compile/local/imagebuilder and i think right now /compile/local is still hardcoded as base for some dirs, so it might be best to start there - making it more flexible might be some possible improvement for later.

i think the problems you ran into with your vm setup is that you end up with a chrooted env within a chrooted env and things might maybe not work asexpected enymore then - also for the image build a loop device is used and this might also lead to problems if there are too many os stacks inside of each other maybe.

hexdump0815, 2023-02-07:
@unixabg - to make things a bit easier i sat down and started to write at least some initial information about how to use the imagebuilder framework at https://github.com/hexdump0815/imagebuilder/blob/main/doc/using-the-imagebuilder-framework.md ... feedback, fixes and improvements are welcome.

closed 2023-03-09 by unixabg