velvet-os/imagebuilder #123: No LTE - Modem not found!

llcool, 2023-02-09:
Hi,

Great work, this is the only image I found that work (boots) on my Acer Spin 513
The original image has an unstable  wifi but as a couple of updates the wifi is fully working. 
I found that the keyboard backlight was off so I add a line to /etc/rc.local
I replaced xubuntu  with kutuntu (using it now running from a 64Gb USB 3 disk).
I installed brave-browser (this is faster that Firefox).
I changed the keyboard model to 'Google Chromebook' (now the Chrome function keys work).

On Linux
`mmcli -L
No modems were found`

On Chrome
`mmcli -L
Qualcamm Sapdragon 7c` (something similar)

[image]

hexdump0815, 2023-02-10:
@llcool - happy to hear that it works for you - for lte maybe try to use the firmware files from your chromeos install - i think devices with lte use a different firmware with modem support maybe ... good luck and let me know here in case you find out anything - hexdump

bruceloco, 2023-04-23:
Hi
Which image is this? I am faffing about with cadmium for several days now, and can't boot because of the blank screen on my acer spin 513 - lazor-r9-lte.

I want to use image builder to build void linux.
thanks

llcool, 2023-04-23:
I had a blank screen on other images but both these images work https://github.com/hexdump0815/imagebuilder/releases/tag/230218-03
bookworm is Debian and jammy is Ubuntu

I have not tried building an image but there are some notes on this github.

bruceloco, 2023-04-23:
> I had a blank screen on other images but both these images work https://github.com/hexdump0815/imagebuilder/releases/tag/230218-03 bookworm is Debian and jammy is Ubuntu
> 
> I have not tried building an image but there are some notes on this github.

Thanks, i did just that, sleep has some issues, but at least i have a working installation

hexdump0815, 2023-04-23:
@bruceloco - i think the sleep issues are only happening when running from usb and not when running from emmc and iirc there is maybe even a fix for the usb issue in a later kernel version (not sure if it was v6.2 or v6.3)

bruceloco, 2023-04-24:
> @bruceloco - i think the sleep issues are only happening when running from usb and not when running from emmc and iirc there is maybe even a fix for the usb issue in a later kernel version (not sure if it was v6.2 or v6.3)

Thanks, I spent a week fighting cadmium builds and failed miserably 😄 
I will try Linux 6.3
From what I gathered, I can just get your config from
https://raw.githubusercontent.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/master/config.cbq-6.1.11-stb-cbq

And build the kernel with it

hexdump0815, 2023-04-24:
interesting - when i tested to daily drive it in january, it was working quite well for me and i did not notice the problems you described - that was with my v6.1.1 kernel

bruceloco, 2023-04-25:
How did you extract the build options from chrome?
I cannot boot, mount the partitions with ChromeOS Kernel to see if there is something in boot.

hexdump0815, 2023-04-26:
what do you mean with build options?

bruceloco, 2023-04-27:
The kernel configuration you mention 
https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/misc.cbq/options/additional-options-special.cfg
You got this from a chromebook somehow? or how did you get to these flags?

The kernel config options required for the let's say the Acer Spin 513 which is the sc7180.

bruceloco, 2023-04-27:
this is really good, my config only has the sc7180 config stuff in, no extra chip stuff in

bruceloco, 2023-05-13:
Question:
I build a void-musl for my trogdor lazor - Acer 513 LTE
Pretty much the same config used by you.
You have 4 partitions, 2 chrome os kernel partitions, one calledboot (3rd partition) - the contents are initrd and dtb and other stuff and the rootfs partition.
I followed the cadmium, but there it only discusses 3 partitions
same first 2, and the last one with rootfs
Does the trogdor lazor require 4 partitions? is this a coreboot thing?

hexdump0815, 2023-05-14:
no, you can go with two partitions even if you want - only the first kernel partition is required (the second one is just for testing alternative kernels and thus quite useful to have) - for 3 and 4 i use a separation of /boot and / but you can have /boot as part of the / partition as well ... i use this separation as it makes luks encrypted root fs handling easier in case one wants to use it

bruceloco, 2023-05-15:
Awesome, I got it running using your config in void-musl - kernel 6.3
However it is failing loading the firmware files, says it cannot find then
they are in /usr/lib/firmware/qcom and it prints that it failed to load err -2 - which is no such file or directory
So no wlan
I am testing the adreno gpu one:
"Direct firmware load for qcom/a630_sqe.fw failed with error -2"
[adreno_request_fw](https://elixir.bootlin.com/linux/v6.3/C/ident/adreno_request_fw)

So don't know what is going on there

bruceloco, 2023-05-15:
and i think i know what it is
void uses runit, i see systemd in your dmesg initializing several services
Is this something you configure in the kernel build in the image? an initrd thing? or just the etc config for the Acer 513 trogor-lazor?

hexdump0815, 2023-05-15:
@bruceloco - maybe have a look at: https://github.com/hexdump0815/qcom-tools-build and https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/postinstall.sh#L10 and https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/postinstall-chroot.sh#L3-L4 ... maybe something similar like this on your side might help

bruceloco, 2023-05-15:
That might be the issue, the rmtfs is the one doing the comms and remote fs for the firmware right?
will give that a try

bruceloco, 2023-05-16:
Well, got the services running, wlan running - but a shutdown means segfault or kernel oops, possibly a firmware issue.
Will try the newer firmware to see if it works better.
The graphics one is still not loaded, so ploughing through but almost there

hexdump0815, 2023-05-16:
the 64bit chromebook images are not using an initrd by default, so it should not be related to that

bruceloco, 2023-05-17:
I know what it is:
In your system you have xde installed, i have a barebones, no window manager yet
later calls are made and the FW is loaded
So I installed Wayland, mesa and sway - this then loaded the firmware, so the error isn't actually an error.
So if anyone complains that the FW isn't loaded, that's is because no DRM client is loaded

hexdump0815, 2023-05-17:
@bruceloco - oh that is an interesting information ... i'm happy to see that you got nearly everything running as well now ... please update here in case you make any progress beyond what is already working in the images so far

bruceloco, 2023-05-30:
Another thing to be done is to disable the radio before shutdown(or in the shutdown script).
Something like
nmcli radio wifi off(or something like this)

If this is not done, it takes ages to shutdown, creates kernel oops or similar things, tries to restart the service, etc.
the above command makes shutdown nice and clean

hexdump0815, 2023-05-31:
@bruceloco - thanks a lot - i noted it down and will try to add it the next time i'm working on trogdor

bruceloco, 2023-05-31:
> @bruceloco - thanks a lot - i noted it down and will try to add it the next time i'm working on trogdor

Forgot to add, it needs to be enabled at boot, if disabled at shutdown ;-)