velvet-os/imagebuilder #126: Trogdor lazor kernel - Missing support for vfat, dos, nfs, nfs4

llcool, 2023-02-13:
Unable to mount fat filesystem.
An error occurred while accessing 'Home', the system responded: The requested operation has failed: Error mounting /dev/sdb1 at /media/linux/16GB: unknown filesystem type 'vfat'

Kernel: 5.19.1-stb-cbq+
`linux@changeme:~$ cat /proc/filesystems `
nodev   sysfs
nodev   tmpfs
nodev   bdev
nodev   proc
nodev   cgroup
nodev   cgroup2
nodev   cpuset
nodev   devtmpfs
nodev   binfmt_misc
nodev   configfs
nodev   debugfs
nodev   tracefs
nodev   securityfs
nodev   sockfs
nodev   bpf
nodev   pipefs
nodev   ramfs
nodev   devpts
        ext3
        ext2
        ext4
        squashfs
nodev   ecryptfs
nodev   mqueue
nodev   selinuxfs
        btrfs
nodev   pstore
        fuseblk
nodev   fuse
nodev   fusectl

hexdump0815, 2023-02-13:
@llcool - this is most probably due to the fact that the trogdor kernel config was at that time still based on the chromeos kernel config - meanwhile i have changed it to be defconfig based and also added a lot of generic kernel config options (among them some filesystems) which i add to all the kernels ... i have just built a v6.1.11 kernel which should be much better: https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.1.11-stb-cbq ... i did not have time to test it myself yet, but will do so soon and will build a new image based on that soon as well - in case you want to try it out beforehand yourself, there are some notes about installing such a different kernel in the doc dir

good luch and best wishes - hexdump

hexdump0815, 2023-02-14:
@llcool - very nice to hear and see this, looks like kde is working well too :) ... the more people are actually using it the better we can make sure those devices stay well supported in mainline by finding upcoming problems in newer kernel versions early on

best wishes - hexdump

hexdump0815, 2023-02-18:
@llcool - just fyi: new image with v6.1.11 kernel - https://github.com/hexdump0815/imagebuilder/releases/tag/230218-03 ... i did not have the time yet to test it myself, but it should work