velvet-os/imagebuilder #127: Kernel panic on boot when building images for lazor

sowa705, 2023-02-13:
Hi, I've tried building images using the tutorial in the docs but the images crash on boot.
Error that I'm getting seems to be an issue with mounting the root volume.
[image]

hexdump0815, 2023-02-14:
@sowa705 - it looks like it still did mount the root fs ("... mounted root (btrfs filesystem) ..." but there seems to be no /sbin/init etc. on it - so one possible check would be to see if you can mount the 4th partition on another system supporting btrfs and check if it looks like a normal rootfs

one thing i came across yesterday was this: https://github.com/hexdump0815/imagebuilder/blob/main/doc/using-imagebuilder-on-btrfs.txt - i completely forgot about that as i have my system setup which i always use for building the images some time ago, but i remember that i had problems when building images initially on btrfs and that was the reason why i wrote that down (and i plan to merge this into the image build instructions so that it cannot be overlooked) - so in case you are building the image on your trogdor with its btrfs rootfs it might be worth to adjust the setup as mentioned in this document and to retry building the image ... in case this does not help then a full log of the image build would be good to see if maybe something fails silently ... somehow we will get this sorted out, as for me the image building is running very well and i'm running it on one of my images myself :)

good luck and best wishes - hexdump

sowa705, 2023-02-14:
Heh, i've read that file but ignored it as it just says it avoids useless compression. I'll try building the image again

hexdump0815, 2023-02-14:
@sowa705 - i'm not 100% sure if that it the reason, but i remember that i had some problems i think - good luck

hexdump0815, 2023-02-14:
@sowa705 - just fyi: i have now added that extra step for chromebooks to extract the bootable kernel image and the extra info for using the imagebuilder framework on an btrfs to the usage doc: https://github.com/hexdump0815/imagebuilder/blob/main/doc/using-the-imagebuilder-framework.md

hexdump0815, 2023-02-18:
@sowa705 - just fyi: new image with v6.1.11 kernel - https://github.com/hexdump0815/imagebuilder/releases/tag/230218-03 ... i did not have the time yet to test it myself, but it should work