velvet-os/imagebuilder #128: With fresh vf2 sid image systemd-container install gives Failed to adjust quota for subvolume  /var/lib/portables Bad file descriptor failed to create hard link /boot/initrd.img-5.15.0-vf2-260+.dpkg-bak /boot/initrd.img-5.15.0-vf2-260+ Operation not permitted

omac777, 2023-02-16:
Rather than calling the hostname changeme, please consider calling it "vf2-hexdump0815-sid" and also adding it to /etc/hosts.

I figured I would share a few odd errors while installing systemd-container especially after doing a fresh image install just before receiving these errors.  The sdcard was wipefs'ed just before.

- Failed to adjust quota for subvolume "/var/lib/portables": Bad file descriptor
- Failed to adjust quota for subvolume "/var/lib/machines": Bad file descriptor
- ln: failed to create hard link '/boot/initrd.img-5.15.0-vf2-260+.dpkg-bak' => '/boot/initrd.img-5.15.0-vf2-260+': Operation not permitted
 
```
linux@vf2-hexdump0815-sid 2023-02-16_12:50:36_EST : ~
 $ sudo apt-get install systemd-container
[sudo] password for linux: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libmpdec3
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  libnss-mymachines libnss-systemd libpam-systemd libsystemd-shared libsystemd0 libudev1 systemd systemd-sysv systemd-timesyncd udev
Suggested packages:
  systemd-homed systemd-userdbd systemd-boot systemd-resolved libtss2-esys-3.0.2-0 libtss2-mu0 libtss2-rc0
The following NEW packages will be installed:
  libnss-mymachines systemd-container
The following packages will be upgraded:
  libnss-systemd libpam-systemd libsystemd-shared libsystemd0 libudev1 systemd systemd-sysv systemd-timesyncd udev
9 upgraded, 2 newly installed, 0 to remove and 329 not upgraded.
Need to get 7148 kB of archives.
After this operation, 1460 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://ftp.ports.debian.org/debian-ports sid/main riscv64 systemd-timesyncd riscv64 252.5-2 [59.8 kB]
Get:2 http://ftp.ports.debian.org/debian-ports sid/main riscv64 libnss-systemd riscv64 252.5-2 [147 kB]
Get:3 http://ftp.ports.debian.org/debian-ports sid/main riscv64 libpam-systemd riscv64 252.5-2 [200 kB]
Get:4 http://ftp.ports.debian.org/debian-ports sid/main riscv64 systemd riscv64 252.5-2 [2784 kB]
Get:5 http://ftp.ports.debian.org/debian-ports sid/main riscv64 libsystemd-shared riscv64 252.5-2 [1467 kB]
Get:6 http://ftp.ports.debian.org/debian-ports sid/main riscv64 systemd-sysv riscv64 252.5-2 [41.3 kB]
Get:7 http://ftp.ports.debian.org/debian-ports sid/main riscv64 libsystemd0 riscv64 252.5-2 [299 kB]
Get:8 http://ftp.ports.debian.org/debian-ports sid/main riscv64 udev riscv64 252.5-2 [1525 kB]
Get:9 http://ftp.ports.debian.org/debian-ports sid/main riscv64 libudev1 riscv64 252.5-2 [100 kB]
Get:10 http://ftp.ports.debian.org/debian-ports sid/main riscv64 systemd-container riscv64 252.5-2 [372 kB]
Get:11 http://ftp.ports.debian.org/debian-ports sid/main riscv64 libnss-mymachines riscv64 252.5-2 [151 kB]
Fetched 7148 kB in 2s (3178 kB/s)            
Reading changelogs... Done
(Reading database ... 100429 files and directories currently installed.)
Preparing to unpack .../0-systemd-timesyncd_252.5-2_riscv64.deb ...
Unpacking systemd-timesyncd (252.5-2) over (252.4-1) ...
Preparing to unpack .../1-libnss-systemd_252.5-2_riscv64.deb ...
Unpacking libnss-systemd:riscv64 (252.5-2) over (252.4-1) ...
Preparing to unpack .../2-libpam-systemd_252.5-2_riscv64.deb ...
Unpacking libpam-systemd:riscv64 (252.5-2) over (252.4-1) ...
Preparing to unpack .../3-systemd_252.5-2_riscv64.deb ...
Unpacking systemd (252.5-2) over (252.4-1) ...
Preparing to unpack .../4-libsystemd-shared_252.5-2_riscv64.deb ...
Unpacking libsystemd-shared:riscv64 (252.5-2) over (252.4-1) ...
Preparing to unpack .../5-libsystemd0_252.5-2_riscv64.deb ...
Unpacking libsystemd0:riscv64 (252.5-2) over (252.4-1) ...
Setting up libsystemd0:riscv64 (252.5-2) ...
Setting up libsystemd-shared:riscv64 (252.5-2) ...
Setting up systemd (252.5-2) ...
Installing new version of config file /etc/systemd/system.conf ...
Installing new version of config file /etc/systemd/user.conf ...
(Reading database ... 100429 files and directories currently installed.)
Preparing to unpack .../systemd-sysv_252.5-2_riscv64.deb ...
Unpacking systemd-sysv (252.5-2) over (252.4-1) ...
Preparing to unpack .../udev_252.5-2_riscv64.deb ...
Unpacking udev (252.5-2) over (252.4-1) ...
Preparing to unpack .../libudev1_252.5-2_riscv64.deb ...
Unpacking libudev1:riscv64 (252.5-2) over (252.4-1) ...
Setting up libudev1:riscv64 (252.5-2) ...
Selecting previously unselected package systemd-container.
[... 69 lines cut ...]
W: Possible missing firmware /lib/firmware/iwlwifi-ma-a0-gf4-a0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-ma-a0-gf-a0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-ma-a0-hr-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-SoSnj-a0-jf-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-SoSnj-a0-hr-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-SoSnj-a0-gf-a0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-SoSnj-a0-gf4-a0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-ty-a0-gf-a0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-so-a0-gf-a0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-so-a0-hr-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-so-a0-jf-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-cc-a0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-QuQnj-b0-jf-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-QuZ-a0-jf-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-QuZ-a0-hr-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-Qu-b0-jf-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-Qu-c0-hr-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-QuQnj-b0-hr-b0-66.ucode for built-in driver iwlwifi
W: Possible missing firmware /lib/firmware/iwlwifi-Qu-b0-hr-b0-66.ucode for built-in driver iwlwifi
Processing triggers for libc-bin (2.36-8) ...
Processing triggers for man-db (2.11.2-1) ...
Processing triggers for dbus (1.14.4-1) ...
linux@vf2-hexdump0815-sid 2023-02-16_12:53:41_EST : ~
 $ 
```

llcool, 2023-02-17:
I found the image [220814-01 - debian bullseye and ubuntu jammy images for the snapdragon 7c sc7180 based acer sp513-1h, hp x2 and lenovo duet 5 chromebooks - trogdor lazor, coachz, homestar and others](https://github.com/hexdump0815/imagebuilder/releases/tag/220814-01) unstable before doing a couple of updates.

I installed systemd-container without any errors.
```
linux@changeme:~$ sudo apt install systemd-container
[sudo] password for linux: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libnss-mymachines
The following NEW packages will be installed:
  libnss-mymachines systemd-container
0 upgraded, 2 newly installed, 0 to remove and 12 not upgraded.
Need to get 513 kB of archives.
After this operation, 1,831 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 systemd-container arm64 249.11-0ubuntu3.6 [370 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libnss-mymachines arm64 249.11-0ubuntu3.6 [143 kB]
Fetched 513 kB in 0s (2,049 kB/s)          
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package systemd-container.
(Reading database ... 194755 files and directories currently installed.)
Preparing to unpack .../systemd-container_249.11-0ubuntu3.6_arm64.deb ...
Unpacking systemd-container (249.11-0ubuntu3.6) ...
Selecting previously unselected package libnss-mymachines:arm64.
Preparing to unpack .../libnss-mymachines_249.11-0ubuntu3.6_arm64.deb ...
Unpacking libnss-mymachines:arm64 (249.11-0ubuntu3.6) ...
Setting up systemd-container (249.11-0ubuntu3.6) ...
Created symlink /etc/systemd/system/multi-user.target.wants/machines.target → /lib/systemd/system/machines.target.
Setting up libnss-mymachines:arm64 (249.11-0ubuntu3.6) ...
First installation detected...
Checking NSS setup...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for dbus (1.12.20-2ubuntu4.1) ...
linux@changeme:~$ 
```

hexdump0815, 2023-02-18:
@omac777 @llcool - i think the problem is related to the use of btrfs as rootfs and most probably some special setup might be required for it to work properly - maybe the btrfs related part of https://github.com/hexdump0815/imagebuilder/blob/main/doc/using-the-imagebuilder-framework.md can help or some searching for information about running using systemd-containerd on btrfs is required ... it works on the snapdragon_7c image as this is still using ext4 as rootfs as it is in a very early and experimental state

good luck and best wishes - hexdump