velvet-os/imagebuilder #129: misc: ubuntu jammy image requests

hexdump0815, 2023-02-18:
i'll build debian bookworm images by default first now to limit the creation and testing effort a bit - this issue is here in case anyone is interested in an ubuntu jammy image as well for a certain system - in this case just add a comment for which system it would be required and i'll try to create and upload one during the following days ...

llcool, 2023-03-02:
Yes, Ubuntu image please. Nice work. :)

hexdump0815, 2023-03-03:
@llcool - for which image? i can only build them one by one on request and there are quite a few supported systems meanwhile :)

llcool, 2023-03-03:
Ubuntu Jammy for the chromebook trogdor lazor , please.

hexdump0815, 2023-03-04:
@llcool - ok queued, but might take a few days or a week :)

hexdump0815, 2023-03-09:
@llcool - i have uploaded a jammy image now to https://github.com/hexdump0815/imagebuilder/releases/tag/230218-03 ... i did not have time yet to test it myself, but plase give it a try and let me know if it works for you

gentoolinux, 2023-05-29:
Ubuntu Jammy for chromebook kukui would be great :-)

hexdump0815, 2023-06-03:
i'll put this onto my todo list, but it might take a bit

gentoolinux, 2023-06-13:
> @gentoolinux - i finally got to it - now there is also an ubuntu jammy image for kukui at https://github.com/hexdump0815/imagebuilder/releases/tag/230218-01

Thank you very much! I will try it this weekend.
By the way, the Debian bookworm image is working pretty good. The only issue I experienced so far is that the kernel crashes if I connect a USB-C dock. The dock (HDMI and Ethernet) is working fine with ChromeOS.

hexdump0815, 2023-06-13:
i think its most probably the hdmi port on the dock which makes trouble - hdmi over usb-c does not work proerly yet on kukui

aneeshlingala, 2023-06-17:
hexudump0815,

when kappa connects to usb-c display or hdmi, it detects the display but does not output anything. Any fixes?

Thanks,
Aneesh

hexdump0815, 2023-06-18:
@aneeshlingala - this is a known problem and i'm not aware of a working solution to fix this yet

gentoolinux, 2023-06-18:
@hexdump0815 - I have tested the jammy build on kukui. Almost everything works (Wifi, Bluetooth, Audio, Webcam, Touchscreen) :-) Only the USB-C port causes trouble. When I connect my dock it seems like all USB ports are disconnected. After connecting the dock I receive Input/Output errors for device sda (system was booted from USB drive). If the system is booted with the USB dock connected the last messages I see are
```
hub 1-1:1.0: hub_ext_port_status failed (err = -71)
usb 1-1: Failed to suspend device, error -71
```
Do you have any tips how to further trace down the issue?

hexdump0815, 2023-06-18:
@gentoolinux - i have seen similar problems when running with usb root - it looks like usb is not really clean yet in the code - running from emmc for me is perfectly fine and i guess the usb case is not being tested too well by the mediatek maintainers as its not a real usecase for chromeos, which always only runs from emmc

sea768, 2023-09-17:
Could you compile an ubuntu jammy image for the kukui baseboard? the older image actually mounted the root drive as read only. Thanks in advance!

hexdump0815, 2023-09-19:
@sea768 - i'll try to build one during the next days

sea768, 2023-09-20:
thanks :3

Mo-Xcoder, 2023-09-24:
Could you attempt to compile a jammy image for the aarch64 7c based systems?

hexdump0815, 2023-09-24:
@Mo-Xcoder - noted, but it might take a few days until i get to it and please keep in mind that quite a lot of things are not working on the galaxy book go ...

Mo-Xcoder, 2023-09-25:
> @Mo-Xcoder - noted, but it might take a few days until i get to it and please keep in mind that quite a lot of things are not working on the galaxy book go ...

Take your time, and thank you so much for these images

raremotas, 2023-09-28:
Hello would like to request compatibility for the Duet 5 (Homestar) Trogdor build.

hexdump0815, 2023-09-28:
@raremotas - you mean a jammy image for trogdor? if yes, then its noted now - it will also take a few days until i get to it ...

hexdump0815, 2023-09-29:
@sea768 - kukui jammy image is ready now at https://github.com/hexdump0815/imagebuilder/releases/tag/230917-01 - please let me know if it works for you - i only did a quick boot and login test so far with it ...

hexdump0815, 2023-09-29:
@Mo-Xcoder - snapdragon 7c woa jammy image is ready now at https://github.com/hexdump0815/imagebuilder/releases/tag/230308-01 - please let me know if it works for you - i only did a quick boot and login test so far with it ...

raremotas, 2023-10-01:
> @raremotas - you mean a jammy image for trogdor? if yes, then its noted now - it will also take a few days until i get to it ...

Yes I meant the trogdor image.

Thank you hexdump, no rush and much appreciated.

2xrick, 2023-11-01:
I hope there gonna be ubuntu update of peach-pit.
Btw thx for your work.

hexdump0815, 2023-11-01:
@2xrick - ok noted down - might take a few days ... @raremotas - sorry, i forgot about the trogdor one - it will hopefully come soon too ...

hexdump0815, 2023-11-11:
@raremotas  - it took a while, but the trogdor jammy image is now at https://github.com/hexdump0815/imagebuilder/releases/tag/230922-01 and @2xrick - the peach jammy image is now at https://github.com/hexdump0815/imagebuilder/releases/tag/230924-02 ... i did not find the time yet to test them, so please let me know if they work for you

edsub, 2024-01-17:
A Ubuntu Jammy image for Chromebook Oak (Acer Chromebook R13) please?

hexdump0815, 2024-01-18:
@edsub - request is noted down - as usual it might take a few days ...

li-clement, 2024-03-13:
Thx for your work, you make my little poor lenovo 100e(Oak) into a fully functional computer. And a Ubuntu Jammy image request, too...

hexdump0815, 2024-03-15:
@edsub @li-clement - sadly i'm currently having very little time, so most probably the next ubuntu images i'll build will be ubuntu 24.04 noble after it got released and after i fully adjusted the imagebuilder framework for it, so earliest around may ... i hope that is still ok

legendaryfishwastaken, 2024-07-04:
can you make ubuntu jammy for nyan_big?

hexdump0815, 2024-07-05:
@legendaryfishwastaken - i'll go though this github issue when i'll find the time to move on with the ubuntu images (currently there is not enough time for that) - maybe for some first try and older image: https://github.com/hexdump0815/imagebuilder/releases/tag/220611-01 maybe?

LukIsHere, 2024-07-05:
i mean this repo is all about building images urself
if ppl want images right now so badly why not just go ahead and point them towards that, so they can try? otherwise if unable, they have to wait ig

rangethepogi, 2024-07-22:
could you please make an ubuntu image for snapdragon_7c_woa?

usernameswagi, 2024-07-27:
could you please make a ubuntu 24 image for kukui? (IdeaPad Duet)

xenicle, 2024-11-04:
Thanks for your work it really appreciated ! I'm another asking for Ubuntu image for Chromebook Oak...

edsub, 2024-11-07:
No worries mate! Your work is appreciated!

selistones, 2025-01-15:
thank you. i have successfully runing but stuck now what is login and password? i stuck at prompt login

thenameisluk, 2025-01-15:
@selistones pls look at
https://github.com/hexdump0815/imagebuilder

kondoyo, 2025-02-24:
need jammy img for gru please . I test bookworm , but fail to build for dependency structure is differ from ubuntu (around freetype or so,,,)

hexdump0815, 2025-02-24:
@kondoyo - could you please try the debian trixie testimage from https://github.com/hexdump0815/imagebuilder-testimages/releases/tag/spring-release-2025-testimages-round-01 ? - its about two years newer than bookworm, so maybe things are better there ... we are currently focussing to get the trixie images finalized the next weeks and will look at ubuntu noble afterwards only

kondoyo, 2025-02-25:
yeah I will test building my project with trixie testimage later. thanks for early response !

CodingGenius0001, 2025-03-17:
Hi! I need Ubuntu Jammy image for oak (chromebook c330). Thanks!

hexdump0815, 2025-03-17:
@CodingGenius0001 - it is ubuntu ready to run, but with xfce as gui instaad of the default gnome (due to the lower ressource requirements for the usually a bit underpowered chromebooks), so maybe this is what confused you? you can simply install gnome (its documented in the docs i think) and it should then maybe look more like what you expected. those images here are built as ready to run images, so there is no installer

CodingGenius0001, 2025-03-17:
Yeah, I think xfce is what confused me lol :)
Can you post a link to the docs? I can't find it.

Or if you can make an image that has GNOME + the installer, that would be great!

Thanks!

thenameisluk, 2025-03-17:
@CodingGenius0001 https://github.com/hexdump0815/imagebuilder/blob/main/doc/postinst/switch-de.md

NVIDIAQuadroFX, 2025-11-18:
@hexdump0815 Could you make a Jammy image for magneton (IdeaPad Slim 3, Mt8186) with the updated speaker fix (build 251015)? I've tried the debian trixie version with the updated sound-fix but I would prefer Ubuntu. Also, is it possible to install LXQt on Ubuntu? I would like to try Lubuntu but it doesn't support ARM64.

NVIDIAQuadroFX, 2025-11-18:
@hexdump0815 Would it break my speakers and/or break anything else if I used a live install usb to try and install Ubuntu server for ARM?