velvet-os/imagebuilder #130: misc: report a not properly booting image

hexdump0815, 2023-02-18:
in case an image does not properly boot on your device, please add a comment here. beforehand plase make sure you read the readme.md for that system - some images need extra customization for one of the supported devices by the image in order to boot properly.

RagnarLodbrok79, 2023-03-02:
I can't make this boot on the chromebook minnie, i't only beeps (another bootable usb works).
I made a a bootable USB with dd on linux.
I read the specific instructions and I should copy u-boot from partition 3 to partition 1?
From gparted i have 4 partitions on the USB-drive.
I can find a U-boot on /dev/sdb3, should I put that on /dev/sdb1 ?
I can't mount sdb1, how to put it there?
[image]

unixabg, 2023-03-02:
Greetings,
the extra uboot options are on partition 3 in a folder called extra. In my case I used something like the below to copy the uboot that worked for my to the partition 1 of my sdcard:

mkdir /tmp/p3
mount /dev/mmcblk1p3 /tmp/p3

Now copy the image you want to try to the partition 1

dd if=/tmp/p3/extra/uboot.kpart.cbr-minnie of=/dev/mmcblk1p1

I have veyron_mighty chromebooks but I used the minnie uboot to make mine boot.
And I could only get sdcard to work on my chromebooks, not usb sticks.
Hope this helps.

RagnarLodbrok79, 2023-03-02:
It worked! Thanks alot!
Boots from usb stick as well!

hexdump0815, 2023-03-02:
@RagnarLodbrok79 - i'm happy to hear that it works in the end for you now - thanks to @unixabg for the quick help.

mattschwantes, 2023-03-03:
Thank you for this project!

I do have a boot issue on Asus Chromebit RK3288 (i.e. veyron-mickey) using the Deb8an release from 220820-01.

After dd'ing the mickey kpart from /boot/extra to /dev/sdc1 and adjusting extlinux.conf, there is console output on the attached display, but it seems to be unable to read extlinux.conf

I mentioned SDC above as I'm using a USB stick because the chromebit has no SD card slot (perhaps only SD is configured for mickey)?

I tried using kpart for the other devices including speedy, minnie, and jerry, but none of those produce video output (perhaps they expect a built-in display which chromebit doesn't have?)

I'll add another post with the console output:

mattschwantes, 2023-03-03:
Model: Google Mickey
Net: Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot: 0
Switch to partitions #0, OK
mmc0(part 0) is current device
MMC Device 1 not found
No mmc device at slot 1
Starting USB...
Bus usb@ff580000: USB DWC2
Scanning bus usb@ff580000 for devices... 1 USB device(s) found
Scanning usb for storage devices...0 storage device(s) found

Device (0): unknown device
No ethernet found.

mattschwantes, 2023-03-03:
The chromebit usb2.0 port has an unpowered hub attached. The hub has the USB flash drive, a USB ethernet dongle, and a USB keyboard-mouse dongle.

I tried using different USB drives and different USB ethernet dongles, but the result was unchanged.

Thank you again for creating this project!

hexdump0815, 2023-03-03:
@mattschwantes - thanks for the positive feedback ... it could be that usb support is broken in some of the u-boot binaries (i had to compile different older versions as later ones were broken for some devices) ... maybe the postmarketos guys have fixed this (they fixed some u-boot issues for veyron) - i'll have to try to build a u-boot from their tree and see if it helps ... i think i even have a mickey around somewhere, but it might take some time util i get to it ... you could maybe try to dd the vmlinux.kpart (instead of the u-boot.kpart) from the boot partition (3rd) to the first partition to boot the kernel directly - maybe this works in the meantime as it works without u-boot inbetween

mattschwantes, 2023-03-04:
Thanks for the suggestion @unixabg. I tried the uboot for the other veyron devices, but it didn't have video output.

Thank you @hexdump0815 for your suggestion!  I used dd if=vmlinux.kpart-5.19.1-stb-cbr+ of=/dev/sdc1 and the Chromebit booted directly to XFCE without issue.

Other than non-functional graphics acceleration, it seems to run well out of the box on veyron-mickey. I will test a bit more and may install this to the internal emmc in place of arch.

hexdump0815, 2023-03-04:
@mattschwantes - thanks for testing, i have added a note to the veyron readme: https://github.com/hexdump0815/imagebuilder/commit/e29da873e4bca9b1791d31ac89ed215d7de8c3b9 ... this should be useable with the only limitation that initrd loading is not possible this way ... long term we should have a look at u-boot and in case there is no way to get this working then the pmos guys have hacked some kind of initrd support to the regular boot sequence

gpu accel is disabled by default as it is not 100% stable in all scenarios - should be easy to test by commenting out the lines in https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_veyron/extra-files/etc/modprobe.d/blacklist-panfrost.conf i.e. in /etc/modprobe.d/blacklist-panfrost.conf

good luck and best wishes - hexdump

starfoxdot64, 2023-03-15:
I'm working with veyron-jerry at the moment, looks like it's hanging - I've also seen this problem with Arch Linux ARM, but it managed to initialize the screen and show a console during boot.

I guess this is confirmation that the CTL Chromebook J4 doesn't work? I've given it a couple tries, checked my configs, and I'm not seeing a change (even over HDMI). Just in case I'm missing something, I'm including pictures of my partition layout, and my extlinux.conf.

[image]
[image]

unixabg, 2023-03-16:
@starfoxdot64 - it looks like you are trying to get usb to boot and not sdcard. If you might read https://github.com/hexdump0815/imagebuilder/issues/130#issuecomment-1454021880 and in the next entry @mattschwantes confirms that it works. I can also confirm that that suggestion works on booting from usb on left side of eduGear M4 chromebook veyron_mighty.

starfoxdot64, 2023-03-16:
> @starfoxdot64 - it looks like you are trying to get usb to boot and not sdcard. If you might read https://github.com/hexdump0815/imagebuilder/issues/130#issuecomment-1454021880 and in the next entry [...]

Alright, I'll have to give that a shot later. I'll admit, I didn't look over the comments in this issue thread since it's not specific to Veyron boards, but that's definitely a good pointer for testing.

I can't say i'm the most optimistic though, since the CTL Chromebook J4+ is in the untested portion of veyron-jerry models.

xAbdaziz, 2023-03-16:
Hello,
I have X96Q2 TV box (H313 SoC) I tried to flash last feb build, and it's just stuck on red led, after connecting UART this shows in console:
`U-Boot SPL 2021.04-18166-g647b392bf2-dirty (Jan 27 2023 - 08:09:08 +0100)
DRAM:This DRAM setup is currently not supported.`

Here is my stock UART console log in case it's useful:
https://pastebin.com/9w0EiapP

hexdump0815, 2023-03-16:
@xAbdaziz - maybe your box is one of the newer ones with lpddr3 (h313 is just a low quality h616)? - the only way then is to build your own u-boot based on https://oftc.irclog.whitequark.org/linux-sunxi/2023-02-27#31929559 (if that even works at all) ... for that please keep in mind that gcc should not be 11 or 12 - see: https://oftc.irclog.whitequark.org/linux-sunxi/2023-03-02#31938041 ... otherwise using the images on tv boxes is "if it works it works - otherwise its bad luck" - too little consistency across devices to put too much effort into them

good luck and best wishes - hexdump

ps: there is also some lpddr4 code which might be worth a try: https://github.com/iuncuim/u-boot/commit/2a59bd30a82396f78279a972e79a9b04b56af2c7 via: https://oftc.irclog.whitequark.org/linux-sunxi/2022-12-09#31699422

xAbdaziz, 2023-03-16:
> @xAbdaziz - maybe your box is one of the newer ones with lpddr3 (h313 is just a low quality h616)? - the only way then is to build your own u-boot based on https://oftc.irclog.whitequark.org/linux-sunxi/2023-02-27#31929559 (if that even works at all) ... for that please keep in mind that gcc should not be 11 or 12 - see: https://oftc.irclog.whitequark.org/linux-sunxi/2023-03-02#31938041 ... otherwise using the images on tv boxes is "if it works it works - otherwise its bad luck" - too little consistency across devices to put too much effort into them
> 
> good luck and best wishes - hexdump
> 
> ps: there is also some lpddr4 code which might be worth a try: [iuncuim/u-boot@2a59bd3](https://github.com/iuncuim/u-boot/commit/2a59bd30a82396f78279a972e79a9b04b56af2c7) via: https://oftc.irclog.whitequark.org/linux-sunxi/2022-12-09#31699422

Thanks, yes it seems that i have LPDDR3 RAM
I will try compiling U-Boot from that repo and report back 👍🏻

xAbdaziz, 2023-03-16:
> @xAbdaziz - maybe your box is one of the newer ones with lpddr3 (h313 is just a low quality h616)? - the only way then is to build your own u-boot based on [...]

Looks like warpme already [tried](https://github.com/warpme/minimyth2/commit/49ffa0a746d507e4dfb0fdffc43bca053bfe73c0) that repo and it didn't work for him :(

hexdump0815, 2023-03-16:
@xAbdaziz - as you do not seem to fear to compile your own u-boot etc., maybe just ask at the #linux--sunxi irc channel - people there are usually quite helpful and might help to debug it together if someone is willing to do the compiling and testing ...

starfoxdot64, 2023-03-17:
> @starfoxdot64 - i think the chances should be good to get jerry working, maybe try the different u-boot options (even the non jerry ones) until you get the u-boot menu and if you are there try all options given there ...

@hexdump0815  I got it working with the veyron-speedy uboot build, but it's not booting the bookworm image. I have uboot's shell now though, which is progress. ^^

hexdump0815, 2023-03-17:
@starfoxdot64 - do you get any error messages when it tries to boot? you boot from sd-card right?

starfoxdot64, 2023-03-17:
> @starfoxdot64 - do you get any error messages when it tries to boot? you boot from sd-card right?

I found that it just wasn't detecting my USB on the right port. It works on the left, but Debian doesn't boot when using Jerry or Speedy's kernel. Mighty works though - I'm still figuring this out, but so far the J4+ isn't responding well to the Jerry presets/kernel.

hexdump0815, 2023-03-18:
@starfoxdot64 - nice to see the progress ... maybe it is not really a jerry in the mainline code? also the mighty of @unixabg in https://github.com/hexdump0815/imagebuilder/issues/110 seems to be some mix of minnie and mighty on the mainline side at the end :)

lets maybe move any further discussion over to https://github.com/hexdump0815/imagebuilder/issues/81 where it fits a bit better now

gociii, 2023-04-21:
is this expected to work on odroid-c1? ( as it uses the same SoC )

hexdump0815, 2023-04-21:
@gociii - it might be worth a try, but one thing missing for sure is some boot block - you may try to steal one from some other image (via dd: ```reading: dd if=/dev/otherimagedevice of=boot.dd bs=512 count=32768 status=progress - writing to your written image: dd if=boot.dd of=/dev/yourwrittenimagedevice bs=512 seek=1 skip=1 count=32767```) and depending on how the boot block works some adjustments of the boot scripts in /boot are most probably required as well - a dtb file for the c1 should be included i think. and there is no hdmi support.

best wishes and good luck - hexdump

hexdump0815, 2023-05-26:
sorry for the deleyed response, i'm mostly offline right now - i did not add or test direct nvme boot you, but looks like you got it somehow working: http://forum.rvspace.org/t/nvme-boot-using-visionfive2-software-v2-11-5/2464/20

danboid, 2023-05-27:
Kind of, but only by manually entering a few u-boot commands. This proves to me that is not a problem with your kernel, its just me that doesn't know how to use u-boot properly.

hexdump0815, 2023-08-02:
@isaygo - on which system?

slfdstrctn, 2023-08-03:
Hello! Thx you for your job! Device: Acer CB311-11H, don't work touchpad. Tested in Deban and Ubuntu. Other things seems work fine (sound, wifi-bt, touchscreen, etc).

isaygo, 2023-08-06:
> @isaygo - on which system?

cb5-311 2GB And i've also cant boot all images not 6.** kernel from your list

hexdump0815, 2023-08-06:
did you write the correct u-boot lpart image for the 2gb model as mentioned in the readme at https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_nyan ? ... a cb5-311 2g is the system i'm usually test boot the nyan images with :)

isaygo, 2023-08-09:
> did you write the correct u-boot kpart image for the 2gb model as mentioned in the readme at https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_nyan ? ... a cb5-311 2g is the system i'm usually test boot the nyan images with :)

like with latest image i've do

mkdir /tmp/u-boot
mount /dev/sda /tmp/u-boot
dd if=/tmp/u-boot/extra/*2g-noflicker* of=/dev/sda1

hexdump0815, 2023-08-09:
strange - that should work - maybe try the other u-boot options as well and please keep in mind the screen stays black for a while until something appears on it (kind of 30-60 seconds i think)

isaygo, 2023-08-14:
> strange - that should work - maybe try the other u-boot options as well and please keep in mind the screen stays black for a while until something appears on it (kind of 30-60 seconds i think)

so, my step by step:
1. download https://github.com/hexdump0815/imagebuilder/releases/tag/220611-01
2. unzip
3. win32diskimager
4/
mkdir /tmp/u-boot
mount /dev/sda /tmp/u-boot
dd if=/tmp/u-boot/extra/(tab)-2g  of=/dev/sda1

for debian and ubuntu - the same, fliking and then - black screen

hexdump0815, 2023-08-14:
@isaygo - maybe the hardware differes then, so that u-boot is not able to handle it properly - mine is cb5-311-t0b2 ... can you maybe connect some usb ethernet and see if it is maybe still alive over network and just the screen missing? this would make debugging this much easier.

isaygo, 2023-08-15:
> @isaygo - maybe the hardware differes then, so that u-boot is not able to handle it properly - mine is cb5-311-t0b2 ... can you maybe connect some usb ethernet and see if it is maybe still alive over network and just the screen missing? this would make debugging this much easier.

Ive such idea, but no usb adapter ( seeing the led of usb flash i think its alive, but no screen

hexdump0815, 2023-08-15:
@isaygo - what you may try is to try first the u-boot you have installed with setting a different default in extlinux/extlinux.conf in the third partition (https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_nyan/extra-files/boot/extlinux/extlinux.conf#L3-L5) and if that does not bring any change, you may combine this with experimenting with the different u-boot versions in the extra folder of the third partition instead of the 2g noflicker version you used so far.

isaygo, 2023-08-18:
> @isaygo - what you may try is to try first the u-boot you have installed with setting a different default in extlinux/extlinux.conf in the third partition [...]

Thanks a lot, and sorry for my stupidity, just remove "-fhd" Works better, yet consumes less memory. It's a pity there is no hardware support for video. I will now figure out how to transfer to eMMC. Thank you for your work

hexdump0815, 2023-08-18:
@isaygo - i'm happy to hear that you got some steps forward - for installing to emmc maybe this helps: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

isaygo, 2023-08-22:
> @isaygo - i'm happy to hear that you got some steps forward - for installing to emmc maybe this helps: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

I think about dd. it is safe?  I just can't find where the bootloader is, on emmc?  Can I return the chrome os if I want?

hexdump0815, 2023-08-22:
yes dd is safe as described in the readme (i.e. using different image versions to boot and to be written to emmc - otherwise the btrfs might get confused about two devices with the same btrfs fs uuid)

there is no need to worry about the bootloader on chromebooks as that is located in some separate spi flash, so one can erase the emmc without any risc and chrmoeos can also always be restored via recovery image as long as the original bootloader remains in the spi flash ... this is what makes chromebooks so nicely and rather risk-free useable ... maybe aside from the fact that one needs to get a bit used to the bootloader and partitioning :)

TheNathan0, 2023-09-20:
@vfonov, you may want to unblacklist Panfrost GPU driver for much better performance.  HexDump originally blacklisted it back when Panfrost was unstable but from my testing and experience Panfrost has become a solid and stable GPU driver.

jmsvigno, 2023-09-25:
i used the last image and says:
U-Boot SPL 2021.04-18166-g647b392bf2-dirty (Jan 27 2023 - 08:09:08 +0100)
DRAM:This DRAM setup is currently not supported.
my tvbox uses ik316/h313 axp is 305 and the hardware is EMCP 1gb of ram and 8
below are the dts files  and images that extract from the tvbox 
[front1ik316](https://github.com/hexdump0815/imagebuilder/assets/143033468/e0ff7525-66b5-4861-ae54-6b3bfdd48dff)
[ik313.dts.txt](https://github.com/hexdump0815/imagebuilder/files/12717763/ik313.dts.txt)
[ik3131.dts.txt](https://github.com/hexdump0815/imagebuilder/files/12717764/ik3131.dts.txt)

vfonov, 2023-09-25:
> @vfonov, you may want to unblacklist Panfrost GPU driver and enable desktop composting for much better performance. 
So, I did that, but didn't notice big improvement. Maybe there is something else that needs to be enabled?

hexdump0815, 2023-09-25:
@jmsvigno - it is written in the allwinner_h616 readme that some newer h313/h616 boxes will not work, besides that there is no support for tv boxes: either they work or they don't - sorry

jmsvigno, 2023-09-26:
ok, but how could I recompile the legacy version, trying to set the u-boot
parameters, I saw that it uses the linux kernel 4.9.170.com Android 10

andre2b, 2023-10-03:
Hello. I read in the readme file that newer LPDDR3 TV boxes are not supported. I have one of those. I've tried to boot from numerous Allwinner images without sucess, the only image that boots on my device is a Mini Arch image (
MiniArch-20230212-6.3.5-board-h313.x96_q_lpddr3-SD-Image.img.xz  from [https://github.com/warpme/miniarch/releases]) but it gives me only a command line without network.

My question is, would it be possible to extract the U-boot or device tree (I don't know the reason why all the images won't boot :(  from that image that boots and "transport" it to one of hexdump0815's images to make it boot?

Sorry for the noob question, but I would really like to try a Linux desktop one my TV Box.

PS: Congratulations for the awesome work.

jmsvigno, 2023-10-07:
@hexdump0815  I managed to boot the ik316 model with emcp, the problem now is that I can't use the network card, there is no hdmi output or audio, I tried creating a spl.bin with defconfig and dts h616-t95-2gb but it gives error in compilation
error: arch/arm/dts/sun50i-h616-t95-2gb.dtb: ERROR (path_references): /aliases: Reference to non-existent node or label "mmc0"
From what I noticed I was using the orangepizero2 defconfigs, dtb/dts and they didn't even have HDMI, network and audio parameters in the dtb/dts so when it started it didn't recognize the devices. Could you help me how to fix this dtb error

TheNathan0, 2023-10-11:
Is the Google ChromeOS EC used space module is needed to boot on Veyron?  I've tested two Veyron models and I've noticed that sometimes certain tasks (e.g. Firefox being in use) while on the battery shuts the machine down.  I checked the battery on my Veyron machines and they're in good health.  This isn't the case when plugged into the charger.  From the limited information I've found online, it could be the EC user space module being erratic, causing false-shutdowns.

vfonov, 2023-10-12:
> Is the Google ChromeOS EC kernel module is needed to boot on Veyron? I've tested two Veyron units and I've noticed that sometimes certain tasks (e.g. Firefox being in use) while on the battery shuts the machine down. I checked the battery on my Veyron machines and they're in good health. This isn't the case when plugged into the charger. From the limited information I've found online, it could be the EC kernel module being erratic, causing false-shutdowns.

Could this explain erratic behaviour of uboot bootloader on my C100P?

TheNathan0, 2023-10-12:
Potentially.  I've noticed some odd stuff happening with the u-boot here and there.  It seems like the 2GB and 4GB Veyron Minnie models are entirely different with some of the boot stuff.  The u-boots that boot both models function slightly differently I notice that in very subtle ways.

It seems like the ChromeOS EC kernel modules (there's two of them I've discovered) are required for functions such as the keyboard and battery indicator to work, which is bizarre.  I'm no kernel module expert, but I'll take a look at the module's code and see if I can find where it triggers a shutoff and see if I can just simply get rid of it.  I really hope it's just kernel space stuff and not the hardware EC being funny because there's absolutely no way to fix this (in fact, we still don't even have Libreboot on this device).  This erratic behavior is why I don't usually trust the laptop to run on its own, most especially when it's building packages.

TheNathan0, 2023-10-13:
About the EC, I recalled that it can be flashed.  I stand corrected in my previous remark.  The source code is available for it.  I don't believe the kernel module handles the shutting down mechanism at all after I took a look at it.  I'll look into it more and see if there's a way for me to be able to modify the EC.  I'll probably need to dump the EC's contents as a backup.  I think that Chrome OS is so light to the point of where the device rarely (if ever) would have encountered this kind of an issue.  I think the EC doesn't like it doing resource intensive tasks even though it doesn't realistically harm the unit at all.  Maybe so it doesn't ever "heat up".

TheNathan0, 2023-10-13:
I stand corrected on my last remark.  I forgot that the EC can be easily flashed.

I'm taking a look at this: [https://chromium.googlesource.com/chromiumos/platform/ec/+/refs/heads/factory-veyron-7505.B/driver/charger/isl9237.c](https://chromium.googlesource.com/chromiumos/platform/ec/+/refs/heads/factory-veyron-7505.B/driver/charger/isl9237.c)

Looking at where the voltages are referenced, I was reminded that sometimes I got some odd flickering before the machine would randomly shutoff and it would lessen depending on what was being loaded.  When something that caused the odd flickering was closed, the flickering would stop; have it flicker for awhile, the unit would just shutoff.  Another thing, plugging in the charger would also resolve this issue as well.  This leads me to believe that it's simply a voltage related issue that we're hitting that's causing these issues mainly.  This could be easily fixed with a voltage modification in the device-tree.  You could overclock this ChromeBook if you really wanted to this way (I believe that's how Tinkerboard users have done it). 

For a custom, heavily modified lightweight Gentoo fork + Chromium web browser, it makes sense that Google would simply make the device run at a lower voltage when on battery, though for users like us who have a little more than that, we will probably want something higher.

xAbdaziz, 2023-10-15:
> @hexdump0815 i used the last image and says: U-Boot SPL 2021.04-18166-g647b392bf2-dirty (Jan 27 2023 - 08:09:08 +0100) DRAM:This DRAM setup is currently not supported. my tvbox uses ik316/h313 axp is 305 and the hardware is EMCP 1gb of ram and 8 below are the dts files and images that extract from the tvbox [front1ik316](https://github.com/hexdump0815/imagebuilder/assets/143033468/e0ff7525-66b5-4861-ae54-6b3bfdd48dff) [ik313.dts.txt](https://github.com/hexdump0815/imagebuilder/files/12717763/ik313.dts.txt) [ik3131.dts.txt](https://github.com/hexdump0815/imagebuilder/files/12717764/ik3131.dts.txt)

Try using the x96_q.lpddr3 image from [Here](https://github.com/warpme/miniarch/releases/tag/20230531-6.3.5-g66a40f5ba)
It should work.

amirulshukry, 2023-10-18:
tv box: Vontar T9
Does not boot
https://github.com/hexdump0815/imagebuilder/releases/tag/230930-02

However if I use 
https://github.com/ophub/amlogic-s9xxx-armbian/releases/download/Armbian_bookworm_save_2023.10/Armbian_23.11.0_rockchip_beikeyun_bookworm_6.1.57_server_2023.10.13.img.gz
& use rk3328-t9.dtb from here, I am able to boot.

Unfortunately after a while there is a kernel panic & crash.

In any case, thank you for your effort.

gociii, 2023-11-03:
Release 230915-01 debian-bookworm. Didn't boot on odroid u3. LED red and blue kept stable. Ethernet and power connected. No HDMI connected.

michelberger, 2023-11-27:
Similar issue to #16 on my Asus Chromebook C201P Veyron Speedy. Builds 231001-02 and 230218-05 bookworm. Installed to microsd, .

It works! It actually runs quite fine... but only boots _sometimes_. Maybe 1 in every 6 boot attempts is successful. U-boot always loads and begins loading the kernel, showing the same expected output each time, but only sometimes the bookworm kernel succeeds in loading.

When it loads, u-boot says "Starting kernel...", screen goes black for a half-second (no backlight), and then comes back with the usual boot text. When it fails to load, the screen goes black (no backlight) and then never comes back on. Only indication it's still powered is the power light. I've waited a full ten minutes, and nothing to be done except a full power cycle and try again. It doesn't write anything to the syslogs, so I'm assuming the kernel just fails to load altogether.

Things I've tried:
1) Standard install: DD to microsd, set extlinux.conf and flash correct u-boot file to the first partition
2) Play around with the u-boot options (jaq, mighty, minnie, jerry, mickey)
3) Recreate the third partition as ext4, rsync root over and update the fstab in case btrfs was the issue (it wasn't, still booted at will)
4) Different SD cards (8GB and 64GB class 10 cards)
5) Set the GBB flags
6) Flash to EMMC following your rsync method (with veyron options and the suggested cgpt version)

The only thing that made any difference is flashing to emmc. With the SD card, succeeded to boot maybe 1 in 10 times; the emmc feels like it boots maybe 1 in 6. And it's much faster (go figure) once it does boot.

For what it's worth, I've had a similar/same issue with the official Debian image (though I moved onto another project after it failed to boot twice so I'm not sure if it's the same thing). I did try the official Arch image back in June, which booted the kernel every time, but Arch's image failed to fully load systemd and trying to configure networking manually caused kernel panics. I've had by far the most success keeping it out of the landfill with this project 🙂

hexdump0815, 2023-11-27:
@michelberger - sadly i do not really have any idea how to improve the situation in your case ... to me it looks a bit like some hardware is not initialized reliable on your chromebook - one thing worth trying might be adding "init=/bin/sh" to the kernel cmdline and see if this results in more reliable booting (this will essentially boot into kind of simple single user root shell mode without starting much) ... but as you do not see any output in the failing caes i'm not sure if this will make any difference ... another thing worth trying might be to add "earlycon" and "earlyprintk" to the kernel cmdline - not sure if this would give any extra output on this system.

vfonov, 2023-11-27:
@michelberger  for me, using `vmlinux.kpart-6.1.11-stb-cbr+` instead of `u-boot` helped to get a stable system.

hexdump0815, 2023-11-28:
@michelberger - first lets move the ongoing discussion better to #16 which is closer to the topic i think ... i think the reason the direct kernel boot is not working with emmc is that the "gpt" kernel cmdline is missing in the kernel builds as they are supposed to be booted via u-boot mostly where the kernel cmdline comes from in that case via extlinux.conf and that one contains the "gpt" option - one option would be to reassemble the vmlinux.kpart file with the "gpt" option added to the kernel cmdline basically following those steps https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/28c2b9340dd30b81ae1af734de1167b17fae224b/readme.cbr#L66C1-L74C25 out of the files you already have in /boot ... i'll add this option to the future kernel builds in a few minutes there

quick background: the emmc setup on veyron seems to be somewhat special by design in that it always breaks the primary gpt partition table - the "gpt" option for the kernel tells is to use the secondary gpt partition table at the end of the disk as well if the first one is corrupted

DvaMishkiLapa, 2024-01-05:
Good afternoon, encountered problems running [Ubuntu image 230917-01](https://github.com/hexdump0815/imagebuilder/releases/tag/230917-01) on Lenovo Ideapad duet 10.1 Chromebook - [google-krane](https://wiki.postmarketos.org/wiki/Lenovo_IdeaPad_Duet_Chromebook_(google-krane)).
I burned the image to a USB device, the kernel loads and displays some logs, but the boot does not go any further.
Exactly the same problem with the Debian image, under the same conditions.

<details>
  <summary>Photo Log</summary>

[image]

</details>

[97 more replies]