velvet-os/imagebuilder #131: odroid-u3 - missing kernel module for ufw/iptables-legacy ?

gociii, 2023-02-23:
SysInfo:
odroid-u3 & odroid-u3+
Linux 6.1.11-stb-exy+
iptables v1.8.9 (legacy)
ufw 0.36.1

Hi @hexdump0815 -don't think this is an actual issue with the kernel build.

When running ufw command I get the following:

Rules updated
Default incoming policy changed to 'deny'
(be sure to update your rules accordingly)
ERROR: problem running ufw-init
Warning: Extension limit revision 0 not supported, missing kernel module?
iptables-restore: line 75 failed
Warning: Extension LOG revision 0 not supported, missing kernel module?
Warning: Extension limit revision 0 not supported, missing kernel module?
iptables-restore: line 149 failed

Problem loading ipv6 (skipping)
Problem running '/etc/ufw/before.rules'
Problem running '/etc/ufw/user.rules'

History:
After installing ufw, I ran command ufw status. This results in:
ERROR: Couldn't determine iptables version

So, I ran : update-alternatives --set iptables /usr/sbin/iptables-legacy which outputs:
update-alternatives: using /usr/sbin/iptables-legacy to provide /usr/sbin/iptables (iptables) in manual mode

From a bit of googling, I understand that ufw is meant to work with the full iptables/netfilter build in the kernel. 
CONFIG_NETFILTER=y seems to be baked in already at

https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/config.exy-6.1.11-stb-exy%252B (line 954)

I'm guessing there's a better approach to the update-alternatives... Any ideas please?

A big thanks (as always) :)

hexdump0815, 2023-02-23:
@gociii - do you need this for docker? i have decided to switch from docker.io to podman and podman-docker which seems to be much easier to handle with its requirements ... also i synced the netfilter etc. options in the kernel config with the actual debian kernel options, which i guess should be quite well adjusted - https://github.com/hexdump0815/kernel-config-options/commit/3205fdb0a2ba8ecae604d1e730aa778eafec842f ... sadly that happened after the v6.1.11 kernel build, so will only be available in the next kernel builds ... to me it looks like debian has dropped support for the old ipfilter compatibility with bookman resulting in all such trouble - but anyway, it looks like we will have to find a way to deal with it somehow ...

gociii, 2023-02-23:
Not needed for docker - though I have switched to podman too, which plays nicely with its crun container engine.

In my case running ufw on bare metal odroid u3, not using podman on it. Will wait patiently for the next kernel and then build the kernel thanks to the awesome framework you've built :).

I had the feeling that it's something likely to do with bookworm, which by the way, seems very stable with podman.

hexdump0815, 2023-02-24:
@gociii - in case you build the kernel yourself, then maybe just get the latest v6.1.13 and build a kernel from it following my notes - the kernel config changes are already in the kernel-config-options repo (maybe pull that to make sure you have the latest version) ... it would be interesting if this is fixing your problem

good luck and best wishes - hexdump

gociii, 2023-02-24:
@hexdump0815 - thanks, am tempted to build that or v6.2. The own-kernel-build guide isn´t complex, judging by a first glance.... might give it a go in the coming days, when I find some time - thanks :+1:

hexdump0815, 2023-02-24:
@gociii - if anything is unclear, then please let me know - its good oportunity to improve the notes in case something is not really clear

gociii, 2023-02-26:
Thanks @hexdump0815  - managed to built kernel v6.2.1 using the guide and after booting with it got the following when running ufw:

ERROR: problem running ufw-init
Warning: Extension conntrack revision 0 not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
iptables-restore v1.8.9 (nf_tables):

line 2: RULE_APPEND failed (No such file or directory): rule in chain ufw-track-output
line 3: RULE_APPEND failed (No such file or directory): rule in chain ufw-track-output
Warning: Extension conntrack revision 0 not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension icmp revision 0 not supported, missing kernel module?
Warning: Extension udp revision 0 not supported, missing kernel module?
Warning: Extension addrtype revision 0 not supported, missing kernel module?
Warning: Extension addrtype is not supported, missing kernel module?
Warning: Extension addrtype is not supported, missing kernel module?
Warning: Extension limit revision 0 not supported, missing kernel module?
iptables-restore v1.8.9 (nf_tables):
line 25: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-input
line 26: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-output
line 27: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-forward
line 30: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-input
line 31: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-input
line 34: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-input
line 35: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-input
line 36: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-input
line 37: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-input
line 40: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-forward
line 41: RULE_APPEND failed (No such file or directory): rule in chain ufw-before-forward
line 63: RULE_APPEND failed (No such file or di
Warning: Extension udp revision 0 not supported, missing kernel module?
Warning: Extension tcp revision 0 not supported, missing kernel module?
Warning: Extension addrtype revision 0 not supported, missing kernel module?
iptables-restore v1.8.9 (nf_tables):
line 27: RULE_APPEND failed (No such file or directory): rule in chain ufw-after-input
Warning: Extension LOG revision 0 not supported, missing kernel module?
Warning: Extension limit revision 0 not supported, missing kernel module?
Warning: Extension conntrack revision 0 not supported, missing kernel module?
iptables-restore v1.8.9 (nf_tables):
line 39: RULE_APPEND failed (No such file or directory): rule in chain ufw-after-logging-input
line 40: RULE_APPEND failed (No such file or directory): rule in chain ufw-after-logging-forward
line 41: RULE_INSERT failed (No such file or directory): rule in chain ufw-logging-deny
line 42: RULE_APPEND failed (No such file or directory): rule in chain ufw-logging-deny
line 43: RULE_APPEND failed (No such file or directory): rule in chain ufw-logging-allow
Warning: Extension conntrack is not supported, missing kernel module?
ip6tables-restore v1.8.9 (nf_tables): Couldn't load match `conntrack':No such file or directory

Error occurred at line: 2
Try `ip6tables-restore -h' or 'ip6tables-restore --help' for more information.
Warning: Extension rt revision 0 not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
ip6tables-restore v1.8.9 (nf_tables): Couldn't load match `conntrack':No such file or directory

Error occurred at line: 29
Try `ip6tables-restore -h' or 'ip6tables-restore --help' for more information.
Warning: Extension udp revision 0 not supported, missing kernel module?
Warning: Extension tcp revision 0 not supported, missing kernel module?

Problem running '/etc/ufw/before.rules'
Problem running '/etc/ufw/after.rules'
Problem running '/etc/ufw/user.rules'
Problem running '/etc/ufw/before6.rules'

gociii, 2023-02-27:
Attach build.sh and log for reference
[build-own-kernel.sh.txt](https://github.com/hexdump0815/imagebuilder/files/10836424/build-own-kernel.sh.txt)
[kernel-build.log](https://github.com/hexdump0815/imagebuilder/files/10836425/kernel-build.log)

hexdump0815, 2023-02-28:
maybe there are still some kernel options required or it is broken in bookworm currently - most probably some searching on the net is required to find out what the problem is ... please keep me up to date here in case you find anything out.

gociii, 2023-03-12:
@hexdump0815  - managed to get past this issue with some help from https://conntrack-tools.netfilter.org/manual.html

Rebuilt in kernel v6.1.18 bullseye

- add kernel config flags
# CONFIG_FLAGS to add to /compile/doc/kernel-config-options/additional-options-generic.cfg
CONFIG_NF_CONNTRACK=m
CONFIG_NF_CONNTRACK_IPV4=m
CONFIG_NF_CONNTRACK_IPV6=m
CONFIG_NETFILTER_NETLINK=m
CONFIG_NF_CT_NETLINK=m
CONFIG_NF_CONNTRACK_EVENTS=y

- enable ipv6 in extlinux

-ensure that modinfo nf_conntrack_netlink will NOT result in:
modinfo: ERROR: Module nf_conntrack_netlink not found.

and run:

update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy
update-alternatives --set arptables /usr/sbin/arptables-legacy
update-alternatives --set ebtables /usr/sbin/ebtables-legacy

hexdump0815, 2023-03-13:
@gociii  - thanks a lot for your investigation - i'll have a closer look at it when i have some more time - most probably next weekend or the the one after

hexdump0815, 2023-10-21:
@gociii - could you maybe give my latest september image a try? it has more kernel options enabled and it might required to rerun the update-alternatives from above once more in the booted image ... please let me know if this improves things for you

gociii, 2023-10-29:
Will be testing this out in the coming days... Let me know if a more recent image becomes available.

hexdump0815, 2023-10-29:
there are new images from end of september - nothing newer is planned currently

gociii, 2024-03-27:
This is still an issue in the September 23 release bookworm. Tested on odroid-u3+:

root@host:~# ufw enable
Command may disrupt existing ssh connections. Proceed with operation (y|n)? y
ERROR: problem running ufw-init
Warning: Extension conntrack revision 0 not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?

I'd suggest adding the flags above for future builds. Thanks :)

closed 2024-03-27 by gociii