velvet-os/imagebuilder #132: Missing License.

Demon-Gyoza, 2023-02-26:
I Noticed the lack of a license for this... I would suggest licensing this under the GPL or Apache license.

hexdump0815, 2023-02-28:
you are right - its maybe time to add a proper license to it - i'll add a gpl one soon

hexdump0815, 2023-02-28:
added: https://github.com/hexdump0815/imagebuilder/commit/00386bf166c0291de01fbcf5863c6cf183f51c99

closed 2023-02-28 by hexdump0815

Demon-Gyoza, 2023-03-03:
Wait... Do I get to be a contributor because I contributed to this by suggesting to add a license?

hexdump0815, 2023-03-05:
i would count this as a suggestion, but not as a real contribution - but in the end its up to you how you see it

Demon-Gyoza, 2023-03-05:
Okay, I guess... I'm relatively new to Github and am lacking general knowledge about how the site operates.