velvet-os/imagebuilder #135: Clarification on x86 chromebook installs

unixabg, 2023-03-03:
Greetings,
I am using imagebuilder to upcycle and breath new life in arm chromebooks (ty @hexdump0815 ). But I also have x86 chromebooks I want to upcycle. Is it possible (or practical) to utilize original bios as is being done with the arm units on the x86 chromebooks?

I have done the https://mrchromebox.tech in the past and just did regular distro installs. fwiw, I am more satisfied to stay with stock bios and gbb settings if possible and the velvet os seems lighter than regular installs.

hexdump0815, 2023-03-03:
@unixabg - i tried to build an image for octopus chromebooks (gemini lake) based on the original chromebook bootloader (https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_octopus), but gave up on it due to several limitations like no initrd support on intel, some hw did not work well etc. ... i would recommend my approach to use uefi boot without having to reflash the bootloader working for at least octopus (gemini lake) and and coral (apollo lake) described here: https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_x86_uefi/doc/apollo-and-gemini-lake-chromebooks.md

if you do not want to flash the firmware, then it might be worth to give the rw firmware from mrchromebox (still keeps the chromebook bootloader intact) a try in case any is available for a certain device - otherwise i would recommend to really flash the uefi firmware from mrchromebox as its simply much better than using the chromebook bootloader for such devices.

spoelstraethan, 2023-04-09:
Depthboot is a great way to use the Depthcharge bootloader with a Linux distro. It was formerly known as Breath but maintenance was taken over by the Eupnea-Linux folks and has pretty good support for EFI Chromebooks without needing to replace the firmware with the MrChromebox one.

unixabg, 2023-04-28:
@spoelstraethan - thank you for the information. I'll try to look in to that when I get to the x86 units. I am still processing the some rockchip units currently.

unixabg, 2023-06-02:
@spoelstraethan - I just did take a quick look at the depthboot. It would appear that user who want to redistribute an image for a given chromebook and/or perhaps selling chromebooks that have had the modified image installed might be in violation of some  copyright issue. Very sad.

unixabg, 2023-11-22:
Greetings so I have been flashing mrchromebox uefi firmware and using the chromebook x86 uefi builds. They are very stable. @hexdump0815 - One follow up to the original is the idea of shipping image with two kernels on the uefi builds, one the main line kernel and a second for debug of the imagebuilder configured kernel? There may be reasons not to do that, but it would be handy to boot the imagebuilder configured kernel to test for hardware issue. Anyhow thank you so much for your work.

hexdump0815, 2023-11-24:
@unixabg - i once had the self built kernel shipped as .tar.gz in the /boot/extra dir, but then the resulting images got over 2gb (the github release upload limit) even if compressed with .xz and as a result i dropped it again ... in the end i think it can be downloaded easily and just upacked plus upgrate-grub

unixabg, 2023-11-24:
@hexdump0815 - cool and I agree. Do you build currently build the uefi kernel now? Please know I would try and write some instructions and test for others.

hexdump0815, 2023-11-25:
here: https://github.com/hexdump0815/linux-mainline-x86-64-kernel/releases are kernel builds i'm using on uefi x86 chromebooks - but for many even the default debian/ubuntu kernel is working quite well in my experience

PR #180 (2024-01-07): Update information on systems/chromebook_x86_uefi with eduGear CMT Touchscreen Closes #135
Update information on systems/chromebook_x86_uefi with eduGear CMT touchscreen
    and similar units Closes #135
commit 6eb80a4e (2024-01-07): Merge pull request #180 from unixabg/eduGearCMT-uefi

Update information on systems/chromebook_x86_uefi with eduGear CMT Touchscreen Closes #135
commit ccba830a (2024-01-07): Update information on systems/chromebook_x86_uefi with eduGear CMT touchscreen
and similar units Closes #135
closed 2024-01-07 by hexdump0815