velvet-os/imagebuilder #136: snapdragon_7c_woa: status

hexdump0815, 2023-03-04:
## notes from using a samsung galaxy book go and reports about an acer aspire 1 (from travmurav) running v6.1/v6.2:

here is list of working and non working features so far:

### working
- builtin display
- usb (it looks like some devices have to be connected at boot time, so that the bios is initializing usb properly on the samsung galaxybook go)
- cpu frequency scaling
- gpu (after windows firmware file installation, display shows random noise until xorg starts)
- emmc (acer aspire 1)

### untested
- hw video encode/decode (there are still some firmware errors related to it, but should be possible to get working maybe)
- external display (might partially work according to travmurav)

### broken
- wifi (working on aspire 1 according to travmurav)
- bluetooth (working on aspire 1 according to travmurav - maybe)
- audio (maybe partially working on aspire 1 according to travmurav)
- display backlight control/dpms/brightness (working on aspire 1 according to travmurav)
- suspend/resume (not sure about the status on aspire 1)
- battery state (partially working on aspire 1 according to travmurav)
- ufs storage (samsung galaxy book go)
- lte modem (samsung galaxy book go)
- touchpad (working on aspire 1 according to travmurav)

### installation
- latest image plus firmware installation after first boot with the image according to https://github.com/hexdump0815/imagebuilder/blob/main/systems/snapdragon_7c_woa/doc/required-firmware.md

### problems
- a lot of things do not work properly, only useable for trying to help improving the linuxsupport

hexdump0815, 2023-03-04:
some discussion about firmware files, backlight control etc.: https://oftc.irclog.whitequark.org/aarch64-laptops/2023-03-02#31938232

samsung galaxy book go issue on aarch64-laptops repo: https://github.com/aarch64-laptops/debian-cdimage/issues/21

hexdump0815, 2023-03-05:
samsung galaxy book go: due to the potential problems with usb filesystem corruption it might be a good idea to simply write the image to an sd card (remove the first two partitions afterwards) and a usb device (remove the third partition afterwards) so that it boots from usb (sd card not supported) and runs from sd card (hopefully no fs corruption)

hexdump0815, 2023-03-08:
new image with slightly newer kernel and no longer starting the pd-mapper service by default - it looks like this might get rid of the usb fs corruption i observed with the last image: https://github.com/hexdump0815/imagebuilder/releases/tag/230308-01

AliceCyber, 2023-05-17:
TCL Book 14 Go - image 230308-01 not working as is.
"galaxy book" grub entry -- boot stucks at 0.8s for ~20s, then kernel spews a warning about stall and that's it.
"aces aspire" grub entry -- seems to boot past that but something fails with screen initialization and it flicker twice and dies.

Not sure how to proceed past that. Unscrew case, get to UART and log everything from it?

hexdump0815, 2023-05-17:
@AliceCyber - maybe something like this https://github.com/Sonicadvance1/linux/issues/27#issuecomment-823668871 helps to get some more useful log output for debugging where your problem is ... i guess your system is most probably closer to the aspire 1 as i guess it is also using the sc7180 reference design as a base ... maybe @TravMurav (the author of the aspire 1 support) has an idea?

TravMurav, 2023-05-18:
> @AliceCyber - maybe something like this [Sonicadvance1/linux#27 (comment)](https://github.com/Sonicadvance1/linux/issues/27#issuecomment-823668871) helps to get some more useful log output for debugging where your problem is ... i guess your system is most probably closer to the aspire 1 as i guess it is also using the sc7180 reference design as a base ... maybe @TravMurav (the author of the aspire 1 support) has an idea?

B220G hw is very different from aspire1 I believe, notably it doesn't use the qsip module, instead having custom board layout and it's believed they are using different dsi-edp bridge than the one used in chromebooks and aspire1, which would result in no display support for sure.

There were a couple of people interested in the device, but I think no one ended up getting a far enough working system yet. You may find some notes in  https://github.com/MinimumLaw/TCL-B220G-Linux (in russian)

Please note that it's ///very/// not a good idea to run a wrong DT on a device, since you may assert some gpios or power regulators incorrectly, which /may/ result in hardware damage (Though most of the time this won't cause too many problems, the possibility is always there.)

AliceCyber, 2023-05-18:
> B220G hw is very different from aspire1 I believe, notably it doesn't use the qsip module, instead having custom board layout and it's believed they are using different dsi-edp bridge than the one used in chromebooks and aspire1, which would result in no display support for sure.
Cant we tell kernel to just _not_ touch the bridge, at least for now? It's already configured by UEFI and working. Unfortunately, `nomodeset` didn't help. Should I try to just cut everything about bridge from dts?
Windows .sys driver for the bridge is something like 40kb, and a good half of it is padding zeroes and constant strings. Worst case, decompiling and reverse engineering would be needed. Best case, there's compatible driver already and it would need only a small patch with proper hwids/regvalues.
> There were a couple of people interested in the device, but I think no one ended up getting a far enough working system yet. You may find some notes in https://github.com/MinimumLaw/TCL-B220G-Linux (in russian)
Видела уже.
The most interesting info there is about hardware debug serial on motherboard. I currently have only one USB-UART adapter and would prefer to use that hardware serial as output instead of searching for another adapter to get logs out of the machine.
> Please note that it's ///very/// not a good idea to run a wrong DT on a device, since you may assert some gpios or power regulators incorrectly, which /may/ result in hardware damage (Though most of the time this won't cause too many problems, the possibility is always there.)
Keyboard controller fucks up after "aces aspire" dts so much, that it stops working in uefi and windows until hard reset :lol:

TravMurav, 2023-05-18:
> Cant we tell kernel to just _not_ touch the bridge, at least for now? 

Yes, ideally you would start writing a completely new DT and disable everything related to SoC display subsystem (making Linux not reset it), + clk/pd_ignore unused to keep it clocked and powered on. UEFI sets up not only the bridge but also the SoC display pipeline in a "simple framebuffer" mode (aka just continuously displaying a memory region) which would work fine for bringup, but no gpu acceleration will be possible. 

> Видела уже. The most interesting info there is about hardware debug serial on motherboard. I currently have only one USB-UART adapter and would prefer to use that hardware serial as output instead of searching for another adapter to get logs out of the machine.
Yes, it will be /much/ easier for you if you have UART access, I'm not sure if it's easy to use it without soldering tho, but it's not too hard to i.e. get a multimeter probe to hold it on a testpoint safely.

> 
> Keyboard controller fucks up after "aces aspire" dts so much, that it stops working in uefi and windows until hard reset :lol:

Yeah, something probably upsets the embedded controller, and the firmware for those is usually hot garbage so it's really easy to do so. You'd need to figure out which i2c line is used for the keyboard HID from it and I think it would "just work" after defining it in the DT

AliceCyber, 2023-05-18:
Okay, I would need sooooo much help with device tree - I don't even know where to start when writing one. 
Some of the useful info *should* be in the acpi [dsdt.txt](https://github.com/hexdump0815/imagebuilder/files/11510805/dsdt.txt) which I just dumped and decompiled. Probably. I hope. ~Windows installer doesn't see any hid, storages or usb on boot at all lol~

https://github.com/MinimumLaw/TCL-B220G-Linux show firmware boot log from internal UART which calls eDPBridge LT8911 (more precisely, it's LT8911**EXB**) [Which has out-of-tree linux driver](https://github.com/aystshen/lontium_lt8911exb_driver)

So that should be supported. Technically. Code is four years old, and stable api is nonsense, after all.

@TravMurav are there any useful resources about translating acpi to dts and/or writing device trees in general?

TravMurav, 2023-05-19:
> Okay, I would need sooooo much help with device tree - I don't even know where to start when writing one. Some of the useful info _should_ be in the acpi [dsdt.txt](https://github.com/hexdump0815/imagebuilder/files/11510805/dsdt.txt) which I just dumped and decompiled. Probably. I hope. 

Yeah, acpi tables should have a bit of useful info (tho with a lot of noise around it sadly, it's pita to read)

> ~Windows installer doesn't see any hid, storages or usb on boot at all lol~

Unsurprising given qcom, being a shitty company, didn't even upload most 7c drivers to the MS repos, so you have to use special windows installer for the thing. Hopefully you did some ufs backup or still have the os installed, idk if tcl has recovery images... 

At least your board presumably has hw secure-boot disabled so there should be a chance of recovering it if the boot chain on the ufs is corrupted. (speaking of that, I hope you already know that you must not touch any special partitions other than uefi esp, windows c: or windows recovery)

> https://github.com/MinimumLaw/TCL-B220G-Linux show firmware boot log from internal UART which calls eDPBridge LT8911 (more precisely, it's LT8911**EXB**) [Which has out-of-tree linux [...]

That driver looks like something imported from old downstream and seems pretty annoying given it just throws some barely documented magic values in and doesn't even touch the DRM subsystem of the kernel, I suppose it might be workable at the start to write something sane from it, but sadly a quick skim didn't let me find a proper programming manual for the chip, which would have been much better to have :/

FWIW on the first steps it's probably better to ignore it and rely on efifb

> 
> @TravMurav are there any useful resources about translating acpi to dts and/or writing device trees in general?

For a general idea on how things go, pmOS has accumulated quite a lot of stuff for writing DTS for qcom based phones, which is /almost/ what you are up to. You can safely ignore pmOS specific things like packaging or anything related to bootloaders and android, but the general ideas and concepts should be described quite well.

These are the things I'd recommend to look at:
- https://wiki.postmarketos.org/wiki/Mainlining#Materials_to_look_into -- General index of many great resources and talks
- https://wiki.postmarketos.org/wiki/MSM8916_Mainlining#Initial_Device_Tree -- A topical guide for msm8916, which is a much older qcom SoC so you won't follow it as is, yet it tries to explain the basic concepts and (as an msm8916 enjoyer) I think it is pretty good in giving an idea of what is going on in the dt for someone who is just starting.
- https://wiki.postmarketos.org/wiki/SDM845_Mainlining#Initial_device_support -- The 845 is quite a bit closer to our 7c so this page, while not as detailed, has some hints on i.e. what to nuke from the display subsystem to keep efifb or what to poke to start the remoteprocs you'd need on later stages.
- https://elixir.bootlin.com/linux/latest/source/arch/arm64/boot/dts/qcom/sc7180-trogdor.dtsi -- This is the base dtsi for 7c based chromebooks so it should contain quite a lot of things you could just copy over, though note that cros is quite different wrt firmware so nothing that touches reserved-memory, sound, remoteprocs is valid for you.
- https://github.com/TravMurav/linux/blob/aspire1/arch/arm64/boot/dts/qcom/sc7180-acer-aspire1.dts -- You can probably use this as a base, keeping only the chipset specific bits and nuking peripherals, just to re-add them later one by one, verifying they work and are correct. Also, `sc7180_defconfig` from that branch should be more or less what you want to build your kernel with

Unfortunately I don't immediately know what to suggest on reading the dsdt, but I'd like to invite you to `#aarch64-laptops` on OFTC IRC (English, [has public log](https://oftc.irclog.whitequark.org/aarch64-laptops/), I'm bridging to it via [matrix](https://matrix.to/#/#_oftc_#aarch64-laptops:matrix.org)) as there are a couple other people who might have good suggestions on that front.

I think as a starting task, getting the board to boot from minimal dt with retaining efifb should be your goal, then you could probably get usb pretty trivially.

Feel free to poke me if you are confused with something, if one is willing to do this work, I'm happy to help :)

AliceCyber, 2023-05-20:
[image]
Weeell, easier than i through it would be.

hexdump0815, 2023-05-20:
@AliceCyber - congratulations! - can you maybe share some details (dts etc.)? :)

AliceCyber, 2023-05-21:
What is working:
- EFIFB
- Keyboard
- USB
- Bluetooth

What isn't:
- Touchpad
- GPU & eDP Bridge
- UFS!!!
  - It isn't even implemented in kernel for sc7180 properly.
  - All attempts to use ufs_phy from other chips, namely sdm845, sm6115 and even backported sm7150 driver from 6.4-rc failed
- I wasn't even trying to bring up Wifi and WWAN for now.

DTS: [sc7180-tcl-book-14-go.dts.txt](https://github.com/hexdump0815/imagebuilder/files/11525028/sc7180-tcl-book-14-go.dts.txt)

AliceCyber, 2023-05-22:
I've voided my warranty )
@TravMurav is that enough to start bringing up gpu&mipi without touching the bridge for now?
Also, anybody know how to properly work with `earlycon`?
```
Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.3.1.1-00010-SC7180XWZB-1
S - IMAGE_VARIANT_STRING=RennellWP
S - OEM_IMAGE_VERSION_STRING=SM042022201
S - Boot Interface: UFS
S - Secure Boot: Off
S - Boot Config @ 0x00786070 = 0x000000c8
S - JTAG ID @ 0x00786130 = 0x1012c0e1
S - OEM ID @ 0x00786138 = 0x00000000
S - Serial Number @ 0x00786134 = 0x2c4aa279
S - OEM Config Row 0 @ 0x007841b8 = 0x0000000000000000
S - OEM Config Row 1 @ 0x007841c0 = 0x0000000000000000
S - Feature Config Row 0 @ 0x007841d0 = 0x0840200015300400
S - Feature Config Row 1 @ 0x007841d8 = 0x0014000000008080
S - Core 0 Frequency, 1516 MHz
S - PBL Patch Ver: 5
S - PBL freq: 600 MHZ
D -      5134 - pbl_apps_init_timestamp
D -     63890 - bootable_media_detect_timestamp
D -      1339 - bl_elf_metadata_loading_timestamp
D -       703 - bl_hash_seg_auth_timestamp
D -      6425 - bl_elf_loadable_segment_loading_timestamp
D -      5425 - bl_elf_segs_hash_verify_timestamp
D -      7210 - bl_sec_hash_seg_auth_timestamp
D -       890 - bl_sec_segs_hash_verify_timestamp
D -        37 - pbl_populate_shared_data_and_exit_timestamp
S -     91053 - PBL, End
B -    112362 - SBL1, Start
B -    231434 - SBL1 BUILD @ 14:16:36 on Jul 13 2022
B -    237534 - boot_flash_init, Start
D -         0 - boot_flash_init, Delta
B -    242444 - xblconfig_init, Start
B -    403423 - UFS INQUIRY ID: MICRON  MT128GASAO4U21  0101
D -      1098 - Auth Metadata
D -    193858 - xblconfig_init, Delta
B -    439871 - boot_config_data_table_init, Start
B -    443500 - Using default CDT
D -      4636 - boot_config_data_table_init, Delta - (54 Bytes)
B -    456859 - CDT Version:3,Platform ID:29,Major ID:1,Minor ID:0,Subtype:0
B -    464271 - pm_device_init, Start
B -    469303 - PM: PM 0=0x800808000000080:0x0 
B -    469364 - PM: PM 2=0x400008000000020:0x0 
B -    473726 - PM: POWER ON by KPDPWR,POWER OFF by RAW_DVDD_SHD
B -    564463 - PM: SET_VAL:Skip
B -    564768 - PM: PSI: b0x30_v0x17
B -    572271 - PM: Device Init # RTC: 0, #SDAM: 0
B -    572332 - PM: Device Init # SPMI Transn: 5258
D -    112636 - pm_device_init, Delta
B -    581665 - pm_driver_init, Start
B -    590327 - PM: Driver Init # SPMI Transn: 362
D -      5215 - pm_driver_init, Delta
B -    595024 - vsense_init, Start
D -         0 - vsense_init, Delta
B -    604937 - sbl1_ddr_set_params, Start
B -    605120 - Pre_DDR_clock_init, Start
D -        91 - Pre_DDR_clock_init, Delta
D -      8265 - sbl1_ddr_set_params, Delta
B -    616588 - sbl1_ddr_init, Start
D -      8052 - sbl1_ddr_init, Delta
B -    628605 - DSF version = 104.0.19, DSF SHRM version = 59.5
B -    637694 - Manufacturer ID = ff, Device Type = 7
B -    637846 - Rank 0 size = 4096 MB, Rank 1 size = 0 MB
B -    642909 - Row Hammer Check : DRAM supports unlimited MAC Value : MR24[OP2:0 = 0] & MR24[OP3 = 1] for CH0 & CS0
B -    653737 - Row Hammer Check : DRAM supports unlimited MAC Value : MR24[OP2:0 = 0] & MR24[OP3 = 1] for CH1 & CS0
B -    664412 - do_ddr_training, Start
[... 245 lines cut ...]
Successfully loaded Tools FV
Current is disabled
WaitForTimeout:0
HotKey: 0x11, 0x0
Boot menu Hotkey detected
FV mounted from memory
BattShow  don't need to show low batt tips
[DppDxe] Failed to read DPP block data with RawFSProtocol.
: BlockGetDPP returned Not Found
LoadEC failed. Status: Not Found.
ENEFlashApp force to flash EC.bin failed. Status: Not Found.
Start image failed
Failed to launch ENEAutoFlash app, status: Not Found
No pending capsules found in EFI Raw file
ACPI: Loaded 1 tables
TPM type 0x6654504D successfully updated to ACPI variable.
[DppDxe] Failed to read DPP block data with RawFSProtocol.
: BlockGetDPP returned Not Found
: DppProtocol->GetDPP returned Not Found
 HandleMorPpi delta: 4
Platform Init End : 5101
-----------------------------
JLQ_Gpio_Init_2
Entering Boot Menu
```

TravMurav, 2023-05-23:
> Okay, I should've also added console=ttyMSM0 because kernel driver brings
> up device and breaks earlycon so after that console starts working.
> Boot log is there: boot.log
> Most interesting piece:
>
> [    2.491423] ufshcd-qcom 1d84000.ufshc: Adding to iommu group 2
> [    2.498737] ufshcd-qcom 1d84000.ufshc: ufshcd_populate_vreg: Unable to find vdd-hba-supply regulator, assuming enabled
> [    2.510605] ufshcd-qcom 1d84000.ufshc: ufshcd_populate_vreg: Unable to find vcc-supply regulator, assuming enabled
> [    2.522108] ufshcd-qcom 1d84000.ufshc: ufshcd_populate_vreg: Unable to find vccq-supply regulator, assuming enabled
> [    2.533706] ufshcd-qcom 1d84000.ufshc: ufshcd_populate_vreg: Unable to find vccq2-supply regulator, assuming enabled
> [    2.545938] scsi host0: ufshcd
> [    2.734873] PM: genpd: Not disabling unused power domains
> [    2.742312] clk: Not disabling unused clocks
> [    3.166879] ufshcd-qcom 1d84000.ufshc: ufshcd_verify_dev_init: NOP OUT failed -11
> [    3.185339] ------------[ cut here ]------------
> [    3.191382] gcc_ufs_phy_axi_clk status stuck at 'off'
> [    3.191417] WARNING: CPU: 0 PID: 63 at drivers/clk/qcom/clk-branch.c:86 clk_branch_wait+0x13c/0x154
> [    3.208574] Modules linked in:
> [    3.213065] CPU: 0 PID: 63 Comm: kworker/u17:0 Not tainted 6.4.0-rc2+ #1
> [    3.221319] Hardware name: TCL Communication Ltd. B220G/B220G, BIOS 1.21 07/29/2022
> [    3.230573] Workqueue: ufs_clk_gating_0 ufshcd_ungate_work
> [    3.237616] pstate: 604000c5 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
> [    3.246185] pc : clk_branch_wait+0x13c/0x154
> [    3.251997] lr : clk_branch_wait+0x13c/0x154
> [    3.257789] sp : ffff800008b53c80
> [... 18 lines cut ...]
> [    3.388294]  worker_thread+0x220/0x428
> [    3.393201]  kthread+0x110/0x114
> [    3.397577]  ret_from_fork+0x10/0x20
> [    3.402289] ---[ end trace 0000000000000000 ]---
> [    3.408203] ufshcd-qcom 1d84000.ufshc: ufshcd_setup_clocks: core_clk prepare enable failed, -16

Ah, seems like the UFS driver tries to init the chip and fails. Then it
starts to tear down the configured and enabled clocks, which it for some
reason fails to do, resulting in gcc being upset, and when gcc is sad
clocks do break terribly everywhere...

I think the most obvious thing to look at is the fact that the driver
complains about the missing power lines. Maybe efi turns those supplies off
and Linux has to re-enable them before touching the ufs chip.

mrintrepide, 2024-01-09:
Hi, I have an Acer Aspire 1, and I tried old and new builds with a USB stick (Samsung USB 3.1 FIT Plus 256 GB), but I got the same result. After GRUB model selection, I have some lines about dtb and get a permanent black screen.
Did I do anything wrong? (I successfully boot PostMarketOS)

hexdump0815, 2024-01-14:
@mrintrepide - can you please try to unplug/replug the usb device at the piint when it hangs? i have seen cases where this was required for some system/usb disk combinations for the usb disk to be properly recognized by the kernel ...

mrintrepide, 2024-01-15:
@hexdump0815 thanks for your reply.
I use the latest build for 7c_WOA with 6.6.9 kernel.
Unplug/Replug do nothing visible in screen.
But with plug/unplug some device (usb key or usb ethernet adapter) and reboot and reset sometime it boot. I don't know if it related.

trustytrojan, 2024-08-22:
Any progress with the Sd7c? The main killer is having no wifi. I'd like to contribute some time to get it working. Just got my hands on a Galaxy Book Go, so I can do some experimentation on it, though I have no idea where to start. As I read through most of the comments here I hear a lot about ACPI tables and device trees (DTs), and ripping drivers from Windows images/installs. What can I do to push the effort forward?

hexdump0815, 2024-08-22:
@trustytrojan - wifi meanwhile actually works if you add the proper firmware files from windows - see https://github.com/hexdump0815/imagebuilder/blob/main/systems/snapdragon_7c_woa/doc/required-firmware.md ... i'll update the top page of this gh issue

trustytrojan, 2024-08-22:
> @trustytrojan - wifi meanwhile actually works if you add the proper firmware files from windows - see https://github.com/hexdump0815/imagebuilder/blob/main/systems/snapdragon_7c_woa/doc/required-firmware.md ... i'll update the top page of this gh issue

I followed the instructions there to find the firmware files but unfortunately I couldn't find any `*.mbn` files inside `Windows/System32/DriverStore/FileRepository`. Is there something I need to do on Windows to get those files downloaded?

trustytrojan, 2024-08-22:
@hexdump0815 I got wifi working! Thanks for the link. The next obstacle is the very low brightness... Do you guys have any leads on how to fix that? I can experiment with any steps you give me.

thenameisluk, 2024-08-22:
@trustytrojan you can try
```
echo 1000 | sudo tee /sys/class/backlight/*/brightness
```
it should set your brightness to maximum your kernel can set to

trustytrojan, 2024-08-22:
> @trustytrojan you can try
> ```
> echo 1000 | sudo tee /sys/class/backlight/*/brightness
> ```
> it should set your brightness to maximum your kernel can set to

@thenameisluk This I already know. I'm talking about how to increase that kernel limit, because it's not as high as what Windows provides.

thenameisluk, 2024-08-22:
@trustytrojan you will likely need to modify .dts file for your device before compiling the kernel
can't really help u with that since i don't have one of these

trustytrojan, 2024-08-22:
How does one get a `.dts` file? Is there a guide for how to do this here?

hexdump0815, 2024-08-22:
sadly for the galaxy book go the backlight control has not yet been found, so there is no way to change the brightness right now ... one thing worth a try is to set it to a certain value in windows and hope that it keeps remembered after a restart to linux ... otherwise there is no way to change it as it first needs to be reverse engineered how it works which is a very complex task i'm for instance not able to get solved

trustytrojan, 2025-05-25:
hey guys, i finally came back to this project and got my galaxy book go with wifi & gpu working! i'd like to help out with this device specifically in any way i can, though i'll need some guidance. first thing would be the display brightness issue, but if that's too difficult then i can focus on trying to build a newer debian release.

nbe1233, 2025-10-23:
@hexdump0815 I've been trying to install debian on my sd card via debootstrap,  which builds a minimal rootfs, and booting via the kernels and modules provided in the images. I'm so close to getting a useable laptop, I've disabled the gpu driver with modprobe.blacklist=msm to keep the brightness. The only thing is WIFI, however even after I've installed the firmware, I am unable to get a wlan0 network interface. WIFI works when I install the firmware on the prebuilt images, do you know what is the difference between velvet-os and a raw debian install?

NeonMan, 2025-10-26:
It works. After installing `xserver-xorg-input-evdev` as suggested and editting the xorg conf.

I also changed the match device path to a wildcard, restarted display manager, and it also works. So this is currently working:

```
#/etc/X11/xorg.conf.d/52-keyboard.conf
Section "InputClass"
        Identifier "laptop keyboard"
        MatchIsKeyboard "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection
```

TheVorkMan, 2025-12-03:
I don't know anything about kernel modules that qualcomm uses. lsmod doesn't show anything related to i2c_hid, so I need help with that one.
I think, compiling the module with this message removed will fix the issue. Atleast it will remove the sympthomes :)

nbe1233, 2026-03-28:
I found a workaround for resolving the problem on the Samsung Galaxy Book Go that the  max brightness is too dim; I compiled the mainline kernel 6.19.10 and disabled all backlight drivers and modules in the kernel config file.

Wayland compositors like sway and weston keyboard input don't work as they use libinput which detects event0 as tablet_pad, so I've been using weston's x11 backend and kiosk mode to run sway or plasma wayland.

hexdump0815, 2026-05-02:
@NeonMan - the galaxy book go support is somewhat limited - the best way to dual boot is to boot windows from the internal ufs and linux from the external usb via the bios boot selector (see which f-key to press in the readme) ... i use it this way and have my linux roofts on an sd card

aungmoe32, 2026-09-08:
Anyone got audio updates on galaxy book go?

hexdump0815, 2026-09-15:
@aungmoe32 - nothing from my end, but if anyone makes any kind of progress on the galaxy book go, please add a note here