velvet-os/imagebuilder #138: Unable to boot u3 & u3+ with release 230224-05

gociii, 2023-03-09:
Hi hexdump - thought I'd let you know that having tried burning this img twice for u3 & u3+, I was unable to boot. Blue LED not flashing at all after power up.

hexdump0815, 2023-03-09:
@gociii - i think i know what the problem is: this commit https://github.com/hexdump0815/imagebuilder/commit/793423b50929fd9b66d7b19585fe1a4ecf276a63 is breaking the boot blocks at least on the odroid, but maybe on other systems too - i'll build and upload new images and let you know when they are ready

hexdump0815, 2023-03-09:
@gociii - i have uploaded an image with fixed bootloader which should work (at least it does for me now) ... as a bonus there is now also an ubuntu jammy image as well :)

https://github.com/hexdump0815/imagebuilder/releases/tag/230224-05

thanks once more for reporting this problem

gociii, 2023-03-09:
You're a star🌟 ! Will try it in a couple of hours and let you know.

gociii, 2023-03-09:
This is fixed :) Thanks.

closed 2023-03-09 by gociii