velvet-os/imagebuilder #141: misc: all good - i'm happy :)

hexdump0815, 2023-03-22:
this issue is here in case someone wants to leave a comment when everything with an image is working well, so there would be no other reason for opening an issue here :)

jakubtalich, 2023-03-24:
All good, boots just fine from a microSD card. (firmware v 2.10.4). Thanks!

jakubtalich, 2023-03-24:
Update: so far the best build of headless Linux image out there. Everything working out-of-the box. Even got this https://github.com/eswincomputing/eswin_6600u running without having to build it myself. Again, many thanks! You made my day easier :-)

LarsLinux93, 2023-03-27:
Hi @hexdump0815,
I initially became aware of your work when I was searching for a way to make music using a raspberry pi zero. I had dabbled a little in pure data and supercollider and build a MeeBlip synthesizer, but I learned about VCV Rack and it didn't run on my netbook. I built VCV Rack using your scripts and I only remember that it took forever to build on the raspberry pi itself.  Traded the Raspberry Pi for a Nokia N900 somewhere afterwards and fell in love with SunVox. I tried to install mainline Debian with some success, always returned back to stock to make phone calls. weird flash formats, missing GPU/DRM support and the lack of a stable internet connection were also discouraging. Changed from the N900 to a Sony Xperia, ostensibly to use Sailfish OS. Didn't even install LineageOS after actually using the device. 

When I saw the announcement regarding mainline support for powervr  I bought a Lenovo Yoga N23 Chromebook (hana). I had tried a few times to boot debian with the chromeos kernel as outlined here for example: https://elinux.org/User:Uli/Acer_Chromebook_R13_GPU_test_system
There are other bits of information that helped me familiarize myself with bootable images for chromeos devices, even if they don't look like success stories: https://gitlab.tails.boum.org/tails/blueprints/-/wikis/ARM_platforms/Acer_Chromebook_R_13_CB5-312T/
The A/B kernel partitions and immutable root remind me of Fedora Silverblue, right now, Fedora doesn't really feel at home on MediaTek cbs: https://fedoraproject.org/wiki/Architectures/ARM/Chromebook

Your builds for hana are the most viable option to test the current mainline status before committing to turning off the beep with a suzyqable and installing to EMMC. I've run Renoise and SunVox after setting up JACK2 with you scripts and recorded electronic jam sessions with OBS Studio, Reaper and the Sony mentioned earlier as network camera.

Until we see some mainline work from IMG regarding the GPU and google is still offering updates, I don't mind that external displays are not working yet. I imagine the work to support external displays via USB-C/HDMI depends a lot on the DRM kernel driver, so I bought a Tanix TX6S to learn a little about that. 

From your notes I thought my hassel would be to get panfrost working, but it seems I got a different hardware revision as you suggested here:  https://github.com/hexdump0815/imagebuilder/issues/130#issuecomment-1472493061
If I had seen that earlier, I might have chosen a model with an H6 processor instead - hate to open a new device to add UART, but after the experience with the chromeos and maemo this is still worth the trouble. 
I'll also test the MiniArch and MiniMyth builds, to see if a patch arrives there early. For the H616 TV Box, I might want to build a minimal image using your scripts to run a media sever:  https://wiki.debian.org/FreedomBox/Hardware/Debian 

Once I have checked all troubleshooting resources that are available, I will start leaving some notes and try my luck to make my own builds. I wonder if I could build Ubuntu Focal still, now that UBports finally moved to that Ubuntu version, there might be some interesting things to try. Haven't really seen LoMiri run on 2-in-1 devices yet. Firmware blobs from ChromeOS or Android are meant for 32-bit rootfs in both of my devices and I didn't see anyone in the comments here try what's written on elinux. 

At the end, I wanted to share a magazine that has been a source of inspiration for my journey with linux on ARM devices:
https://solar.lowtechmagazine.com/2018/09/how-to-build-a-lowtech-website.html

Thanks to your work, I am able to use some low-powered consumer devices as development and education tools.
I hope to see support for the MT81xx with a legacy desktop distro such as MX Linux or Ubuntu Mate to be the fruition of work done here.  Compared to the first 32-bit armhf chromebooks, the mt8173 can be found in a number of education editions which I expect to be around for at least as long as that darn T430 and last 8 hours a charge with the usage outlined in your daily driver doc.

aneeshlingala, 2023-04-03:
One day, I needed a budget Chromebook. I found one on sale on a trusted online marketplace in India. Since the price was so good, I forgot to check for Linux support. When I wanted Linux, I only found guides for x86. So, I dived a little bit deeper and found your awesome repo. I tried Cadmium but there was no support for kappa. Your image worked without sound, so I raised an issue. Thanks to your quick response, I was able to know that you got sound working and you were going to release an image with sound support soon. One suggestion though. It would be nice if we had Arch Linux or Fedora support. The only true mainline Linux alternative for kappa is [postmarketOS](https://wiki.postmarketos.org/wiki/HP_Chromebook_11a_(google-kappa)), which has no Bluetooth support. Recently, under load the freezing problem was fixed with the 6.0+ kernels. Now the images could be used better. Thank you for making a streamlined process to run Linux on ARM Chromebooks.

Thanks,
Aneesh ([Link to my website](https://aneeshlingala.github.io))

smoti17, 2023-05-09:
Just to say Thank You for this project :-) And all the useful notes. Pleased to report your 230220-03 build booted fine first time from μSD on my gru/kevin Samsung Chromebook Plus.

Despite it being slightly underpowered, I've been running native Linux (SuSE Leap 15.x) on this cute laptop for 5 years now, with help from  [this](https://archlinuxarm.org/platforms/armv8/rockchip/samsung-chromebook-plus) Arch Linux build, but have not had any luck booting more recent kernels than 4.4.159. It's great to finally have Mali GPU support.

(I've started playing with Rockchip RK3588 hardware, and looking forward to seeing its much faster performance in a laptop..)

Anyway, please keep up the good work !

aneeshlingala, 2023-05-11:
Hello Hexdump0815,

Thanks for the great work. I have gotten Arch Linux, Fedora, and OpenSuse on kappa to run by copying the new kernel files for mt81xx from your repo (6.1.11). I also made a good fstab. But, the system boots up in read-only mode and everything fails. I also tried few non-systemd distros like Artix and Void, but there is still a problem. Any solution to get this working. Would love to have Fedora on this! If this is possible, we could have more distros.

Thanks,
Aneesh

hexdump0815, 2023-05-13:
@aneeshlingala - i have no idea what might cause the readonly system - it could be that those systems rely on an initrd/initramfs to work properly and by default the kernel packages are not using innitrd/initramfs

aneeshlingala, 2023-05-13:
I have removed the default kernel packages by chrooting into the system.
What do mean by "those systems rely on an initrd/initramfs to work
properly?"

hexdump0815, 2023-05-14:
some things require an initial ram disk (initrd/initramfs) to work like for instance encryption of the rootfs via luks - i have no idea how those other distributions work, but maybe they also require such an initial ram disk for doing some stuff before the rootfs is mounted - i have no real idea, as i do not know those others dists in detail

aneeshlingala, 2023-05-14:
I have gotten the init scripts from a non-systemd distro and implanted it
into the initramfs. Still didn't work.

Touchgg, 2023-06-12:
Installed on HP 14G3 TEGRA K1, with u-boot for nyan blaze 2GB ram on a 32GB microSD card. Everything is working OK, sound need a little configuration from alsamixer, enabling DAC for speakers, left and right. Hdmi is working OK too. Much more faster compared with Postmarket OS. Upgraded to Debian12. If you choose keyboard layout for chromebook, from settings,  a part of dedicated key's are activated.Thank you for your great work ! The single limitation is due to 32 bit architecture,  but that's another discussion!!
Thank you again !!

alxdolphin, 2023-06-14:
You are a machine. Booted image off of USB and installed to internal with BT/WiFi/Trackpad/etc. working out the gate. Genius fr.

slfdstrctn, 2023-08-03:
Hello! Thx you for your job! Device: Acer CB311-11H (kenzo), don't work touchpad. Tested in Deban and Ubuntu. Other things seems work fine (sound, wifi-bt, touchscreen, etc). Trying this image https://github.com/hexdump0815/imagebuilder/releases/tag/230218-01

hexdump0815, 2023-08-03:
@slfdstrctn - thanks for the feedback and report - do you see anything touchpad related in dmesg? (mostly its i2c related stuff usually)

slfdstrctn, 2023-08-05:
> @slfdstrctn - thanks for the feedback and report - do you see anything touchpad related in dmesg? (mostly its i2c related stuff usually)
Here dmesg logs:
[    0.817478] i2c_dev: i2c /dev entries driver
[    0.928814] mediatek-dpi 14015000.dpi: Found bridge node: /soc/i2c@1100f000/it6505dptx@5c
[   15.297135] elants_i2c 0-0010: supply vcc33 not found, using dummy regulator
[   15.301754] elants_i2c 0-0010: supply vccio not found, using dummy regulator
[   15.404140] elan_i2c 2-0015: supply vcc not found, using dummy regulator
[   15.450020] input: Elan Touchscreen as /devices/platform/soc/11007000.i2c/i2c-0/0-0010/input/input7
[   15.530407] elan_i2c 2-0015: reading cmd (0x0102) fail.
[   15.532912] elan_i2c 2-0015: failed to get FW version: -6
[   15.568174] i2c_hid_of 2-002c: supply vdd not found, using dummy regulator
[   15.577482] i2c_hid_of 2-002c: supply vddl not found, using dummy regulator

slfdstrctn, 2023-08-05:
@hexdump0815, here dmesg logs:

[    0.817478] i2c_dev: i2c /dev entries driver
[    0.928814] mediatek-dpi 14015000.dpi: Found bridge node: /soc/i2c@1100f000/it6505dptx@5c
[   15.297135] elants_i2c 0-0010: supply vcc33 not found, using dummy regulator
[   15.301754] elants_i2c 0-0010: supply vccio not found, using dummy regulator
[   15.404140] elan_i2c 2-0015: supply vcc not found, using dummy regulator
[   15.450020] input: Elan Touchscreen as /devices/platform/soc/11007000.i2c/i2c-0/0-0010/input/input7
[   15.530407] elan_i2c 2-0015: reading cmd (0x0102) fail.
[   15.532912] elan_i2c 2-0015: failed to get FW version: -6
[   15.568174] i2c_hid_of 2-002c: supply vdd not found, using dummy regulator
[   15.577482] i2c_hid_of 2-002c: supply vddl not found, using dummy regulator

bserret, 2023-08-09:
I got it working fine off a usb and was able to use a usb dongle with a network cable - but how do you access the internal eMMC drive?

bserret, 2023-08-09:
> You are a machine. Booted image off of USB and installed to internal with BT/WiFi/Trackpad/etc. working out the gate. Genius fr. (ASUS C201PA)

How were you able to install to internal drive?

hexdump0815, 2023-08-09:
@slfdstrctn - i have no real idea yet about your touchpad problem - something similar was the case with cozmo as well and with some kernel version it started to work then - maybe trying to build a v6.4 kernel might work better ... i have adjusted my kernel compile info for kukui recently to v6.4 so that it compiles, but i did not yet get to testing if the resulting kernel will work (but i guess it should do) - see: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/master/readme.mt8 ... some generic info about building your own kernel: https://github.com/hexdump0815/imagebuilder/blob/main/doc/building-own-kernels.md

bserret, 2023-08-09:
> the internal emmc appears just as another /dev/mmcblk device like an sd card ... it is indeed soldered on the mainboard on all arm chromebooks

Thanks for the reply - I don't show any such devices - this is a samsung galaxy book pro wifi
cat /proc/partitions only shows the usb drives ( I have an extra on the dongle)  
parted --list also only shows the usb drives  as does lsblk -f
Any suggestions?

slfdstrctn, 2023-08-10:
> @slfdstrctn - i have no real idea yet about your touchpad problem - something similar was the case with cozmo as well and with some kernel version it started to work then - maybe trying to build a [...]

thx for replay, i'll try.

smoti17, 2023-08-18:
Got an openSUSE Leap 15.5 aarch64 rootfs to boot and run successfully on my gru/kevin Samsung Chromebook Plus using [hexdump0815](https://github.com/hexdump0815)'s /boot and kernel. Did a number of things to make this work:

- re-did mkfs on the root and /boot partitions (needed to re-size them anyway) in the  6.1.11-stb-cbg+ install image
- copied /lib/modules/ and /lib/firmware/ from  the 6.1.11-stb-cbg+ root partition to the target rootfs
- copied all the files from the /boot directory on the SuSE rootfs to the /boot partition
- restored all the /boot files from the 6.1.11-stb-cbg+ install image
- adjusted /etc/fstab on the target rootfs accordingly

I saw the read-only FS problem, but only on the ext4 /boot partition (not the btrfs root), which seemed to have some ext4fs features the SuSE rootfs did not like. Not sure if **all** the above steps were necessary to get things running, but they appear to be sufficient. Looks like the Leap 15.5 Mesa Panfrost Mali GPU userland support plays nice with the 6.1.11-stb-cbg+  kernel (though some geometry weirdness going dual-head display).

This is of course a very hacky way of doing things compared to getting the imagebuilder stuff to run properly in SuSE, but it at least gets me up and running..

RaylynnKnight, 2023-08-19:
My ODROID-X is finally useful again!  Successfully installed the 230224-05 image to my 16GB eMMC module and so far everything appears to be functioning.  Will be updating my X2 and U3 devices next.

TheNathan0, 2023-08-23:
I remember almost two years ago trying to run Linux on my Peach Pit ChromeBook, trying to get GPU acceleration on it.  I didn't know very much about Linux and this is where my inspiration started and thus, my Linux journey truly began.  From hardly knowing how to setup a Linux environment to build a functioning kernel to now basically being able to build Linux kernels blindfolded.  With my newly learned skills do I hope to effectively and positively contribute back to the very project that helped me reach the inspiration to learn more about Linux in-depth (and Linux on ARM).  You do fantastic work HexDump, God bless.

rakslice, 2023-08-27:
The kukui 230218-01 bookworm image is working great on lenovo 10e (kodama sku288); [hw probe](https://linux-hardware.org/?probe=7c4afe8e55), running from usb

There's a quirk to enable dev mode on this model: the internal screen is not initially enabled; to work around, hold down vol buttons and power on, letting go of the buttons after 5s, then press vol up ten (10) times, then power, then wait 5s; the screen will turn on and you will be at the dev mode menu with os verification disabled, and you can proceed with the rest of the instructions as usual booting from internal drive for the dev mode wipe. Subsequent regular boots will show the dev mode menu and turn the screen on automatically. 

(This workaround courtesy of [reddit](https://www.reddit.com/r/chromeos/comments/r7iph5/comment/hncaoa0/?utm_source=share&utm_medium=web2x&context=3); even the hardware maintenance manual  for this model's recovery instructions didn't account for it.)

Working: 

sound, video, touchscreen, usb, bt/wifi, bat, 2x thermal sensors, graphics accel and video decode in chromium.

Notes:

- Using dropdowns with the touch screen is difficult; idk if this is normal for xfce
- The sound setup is a bit flaky, where sound plays through the speakers when the speakers are not selected in pulseaudio mixer; just switching the output to speakers and back fixes the problem

hexdump0815, 2023-08-27:
@rakslice - thanks a lot for the report and the details about getting it working - i marked kodama as tested now too - https://github.com/hexdump0815/imagebuilder/commit/486175f026b64cbdf34de5f7c0bd264071601cc9 ... the images are not ready yet for proper touch operation (they are just prepared for basic operation in case no keyboard is around or detected) and i think xfce is not the best choice for that right now - i guess gnome or kde are most probably better prepared for that and even then some extra work for automatic screen rotation etc. will most probably be required

antonialoytorrens, 2023-08-28:
> @RaylynnKnight My ODROID-X is finally useful again! Successfully installed the 230224-05 image to my 16GB eMMC module and so far everything appears to be functioning. Will be updating my X2 and U3 devices next.

Just for the report: my Odroid U3 works flawlessly with the same image, well done hexdump :)

aneeshlingala, 2023-08-31:
@rakslice You have to update the kernel to fix freezing issue. Hexdump should have a guide on that. This also happened to me, while playing a video, etc, on kappa.

slfdstrctn, 2023-09-22:
Upd. Device: Acer CB311-11H (kenzo). Image: https://github.com/hexdump0815/imagebuilder/releases/tag/230218-01 
The previous report about the touchpad not working is no longer relevant. Update Debian via apt upgrade fix it. Ubuntu image not tested.

scateu, 2023-09-28:
Samsung Chromebook Plus V1 (XE513C24) (gru-kevin)

works like magic. Sound, touchscreen, WiFi and hardware decoding all good!

After spending 24 hours trying different seafood and poison, I nearly burst into tears.

Thank you so much.

Carlosmaleal, 2023-10-07:
Hi, first off all Thank You for sharing your knowledge, in my case it was successful:

Laptop: Acer CB311-11H KENZO Mediatek mt8183
Your File: chromebook_kukui-aarch64-bookworm.img.gz (https://github.com/hexdump0815/imagebuilder/releases/tag/230218-01)

Developer mode set 
Disconect Battery (no screw I think)
login as guest
- Ctrl Alt --»
login: chronos (no pass)
# sudo -s (root)
# flashrom –wp-disable
# flashrom –wp-status 
(chek OK)
Gbb Flags
# /usr/share/vboot/bin/set_gbb_flags.sh 0x19
# usr/share/vboot/bin/get_gbb_flags.sh
(chek OK)
Write protect
# fmap_decode bios.bin 
(WP_RO range)
# flashrom --wp-enable –wp-range=0x00000000,0x00400000
# flashrom –wp-status
 (chek OK)
Disconect the Laptop

Conect Battery
login as guest
- Ctrl Alt --»
login: chronos (no pass)
# sudo -s 
(root)
# flashrom –wp-status 
(chek OK)
# usr/share/vboot/bin/get_gbb_flags.sh
(chek OK)
# sudo crossystem dev_boot_usb=1 dev_boot_signed_only=0
Disconect the Laptop

insert USB (chromebook_kukui-aarch64-bookworm.img.gz wrote with Rufus)
at Developer scren
- Ctrl+u (or Ctrl+Uparrow+u I don´t now what get to work)
Debian launches
User login: linux
password: changeme
Open terminal
(see https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md , in my case the report is equal)
# sudo -i
# ls -d /dev/mmcblk* /dev/sd* | cat
# df -h /boot
# fdisk -l
# /scripts/extend-rootfs.sh
# wget https://github.com/hexdump0815/imagebuilder/releases/download/230218-01/chromebook_kukui-aarch64-bookworm.img.gz
# export TGTDEV=mmcblk0
# zcat chromebook_kukui-aarch64-bookworm.img.gz | dd of=/dev/$TGTDEV bs=1024k status=progress

Disconect Laptop
Remove USB
Turn On 
Debian launches Ok
User login: linux
password: changeme
Open terminal
# sudo apt-get update (& upgrade)
All Ok till now!
Thank You!!

robanastasi, 2023-10-08:
231001-01 on Samsung Chromebook Plus / gru kevin

Works perfectly booting off USB or microSD.  Thank you very much for you work on this!

hexdump0815, 2023-10-21:
@LarsLinux93 - i think the framework should allow to build oak images with armv7l userland - which problems did you run into? (please open a new github issue for that topic then)

ryjelsum, 2023-10-25:
All is good on my end with the new gru-kevin image as well. 

For what it's worth regarding the graphical issues with gru-kevin mentioned on the device page for the past while (which I have been experiencing - windows were not updating as they should have, they only updated after moving the mouse) these issues seem to be greatly alleviated running under sway - kind of makes me want to try the GNOME and KDE Wayland sessions and see how they run compared to XFCE. USB-C HDMI output also worked much better under sway. Not sure why.

fockr, 2023-10-28:
Thanks a lot @hexdump0815 ! The Odroid U2 image works flawlessly. Incredible I can still use the 10 year old device which - still - has fairly much power under the hood. Great work!

P.S.: Do you have plans for a basic headless image for servers?

hexdump0815, 2023-10-28:
@fockr - no plans for a headless image right now due to the lack of time

hexdump0815, 2023-10-29:
@fockr - as the question came up and comes up again and again from time to time i have now added a script to trim down a running system to a more server oriented setup - it will be part of all future images and you can just run the commands on older ones in case you want to have it already now: https://github.com/hexdump0815/imagebuilder/commit/c2baf9d6038dd943529435f0d2d8e9811fc536ae

gociii, 2023-10-29:
@fockr you can use dietpi after installing the image. It will get rid of the GUI and other items to strip it to a headless.

fockr, 2023-11-01:
@hexdump0815: great! exactly what i've been looking for. i removed big parts of the gui by googling the packages but even after that your script removed x11* and other packages. thanks a lot, that's very handy! if i may suggest a feature: you could add a login notice with a hint to this command for anybody who's not familiar with it - i.e. "This installation contains a complete Desktop Environment. If you need a headless install please execute desktop-to-server.sh."

@gociii: Will look into it. Thank you for your suggestion!

gociii, 2023-11-04:
@fockr - on testing the image for odroid u3 I've used the script desktop-to-server.sh - it works fine. I notice that **agetty** may be removed for many as it is not used by ssh. @hexdump0815 would be handy to include this script going forward :)

hexdump0815, 2023-11-07:
@gociii - do you mean that there was no login on serial console anymore as agetty was gone?

NightFurySL2001, 2023-11-12:
Sorry to disturb but does anyone had success running the image on Odroid U3 with eMMC? I've followed https://forum.odroid.com/viewtopic.php?p=325731#p325731 by putting the eMMC on a SD-card convertor and boot it through the SD card slot, then run the commands 1-7. However command 6 `sudo echo 0 > /sys/block/mmcblk0boot0/force_ro` failed  (which I then skipped) and still can't boot the system from eMMC slot.

rakslice, 2023-11-12:
@NightFurySL2001 Running `sudo echo 0 > /sys/block/mmcblk0boot0/force_ro` doesn't make any sense; that echoes 0 as root and then write it to the file as the user your shell is running as.  Try it another way, like `sudo bash -c 'echo 0 > /sys/block/mmcblk0boot0/force_ro'`, or `echo 0 | sudo tee /sys/block/mmcblk0boot0/force_ro`.

vitali2y, 2023-11-12:
Everything works out-of-the box: _Jammy_ @ _Odroid U3_ , and _desktop-to-server.sh_ made my day to make the device desktop-less. Thank you!

NightFurySL2001, 2023-11-13:
@rakslice Thanks for the pointer, I think i changed the file successfully. However, the image still doesn't boot on the eMMC slot after following the tutorial, the blue LED doesn't flash and the fan keep spinning.

slfdstrctn, 2023-12-31:
> @slfdstrctn - i have no real idea yet about your touchpad problem - something similar was the case with cozmo as well and with some kernel version it started to work then - maybe trying to build a [...]

Hi again! I updated the kernel to 6.5.5 and still have problems with the touchpad. During a cold start, the touchpad works, but after a reboot it stops working. If turn off the device and turn it on again, the touchpad still does not work.
I took the kernel from the archive here - https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.5.5-stb-mt8%2B
Any advice?

aneeshlingala, 2023-12-31:
Did you copy /lib/modules?

Works completely fine for me on Debian.

If you are using the Jammy image, and you are fine with moving to Debian,
try it. Jammy is known for color issues on some Wayland desktops and is
unstable with kernels newer than 6.1.11 (in my experience)

TheNathan0, 2023-12-31:
It could be a hardware and/or driver related issue.  6.1 LTS from what I've seen online and from my own personal experience works fine on Jammy.

PinkFreud, 2024-01-02:
Device: Amazon Fire HD 10 Tablet Gen 7 (suez - mt8173)

I've managed to get debian bookworm 231113-05 booting on this tablet, so that's definitely a start!  I wrote the root image to the data partition on mmc, rather than an sd card.  I'm noticing that update_secureti is a bit unhappy with this decision, given that it's save path was previously on /data:

```
[  120.366326] .(0)[32:update_secureti]enter  TEE_update_gb_time_intee
[  120.366443] .(0)[32:update_secureti]enter  TEE_update_gb_time_infile
[  120.367500] .(0)[32:update_secureti]FILE_Open GB_TIME_FILE_SAVE_PATH return NULL
[  120.367522] .(0)[32:update_secureti]enter  TEE_update_gb_time_intee
```

I'm not sure if this affects / breaks anything.

One thing I've noticed that isn't functional on here is internal wifi, though I think you're aware of this, given that I seem to recall postmarketos's wiki entry mentioning it, and I see it mentioned in https://github.com/hexdump0815/imagebuilder/blob/main/systems/tablet_amazon_suez/readme.md as well.  dmesg log entries show what appears to be gpio conflicts between pinctrl and mediatek,connectivity-combo:

```
[    0.967240] .(2)[1:swapper/0]enter mtk_wcn_cmb_sdio_request_eirq
[    0.967436] .(2)[1:swapper/0]WIFI EINT irq 102 !!
[    0.967447] .(2)[1:swapper/0]exit mtk_wcn_cmb_sdio_request_eirq
...
[    1.582774] .(3)[131:kworker/u8:3][WMT-DETECT][E]wmt_detect_probe(244):platform name: mediatek,connectivity-combo
[    1.585001] .(3)[131:kworker/u8:3]wmt_gpio: gpio init start!
[    1.586001] .(3)[131:kworker/u8:3]wmt_gpio:set GPIO_COMBO_PMU_EN_PIN out to 0: 0!
[    1.587349] .(3)[131:kworker/u8:3]wmt_gpio:set GPIO_COMBO_RST_PIN out to 0: 0!
[    1.588608] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin EINT2 already requested by 1000b000.pinctrl; cannot claim for mediatek,conn
ectivity-combo
[    1.591206] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin-2 (mediatek,connectivity-combo) status -22
[    1.593110] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: could not request pin 2 (EINT2) from group EINT2  on device 1000b000.pinctrl
[    1.595456] .(3)[131:kworker/u8:3]mediatek,connectivity-combo mediatek,connectivity-combo: Error applying setting, reverse things back
[    1.597545] .(3)[131:kworker/u8:3]wmt_gpio:set GPIO_WIFI_EINT_PIN to GPIO_IN_PULLUP done!
[    1.598962] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin DAICLK already requested by 1000b000.pinctrl; cannot claim for mediatek,connectivity-combo
[    1.601574] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin-33 (mediatek,connectivity-combo) status -22
[    1.603493] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: could not request pin 33 (DAICLK) from group DAICLK  on device 1000b000.pinctrl
[    1.605882] .(3)[131:kworker/u8:3]mediatek,connectivity-combo mediatek,connectivity-combo: Error applying setting, reverse things back
[    1.607969] .(3)[131:kworker/u8:3]wmt_gpio:set GPIO_PCM_DAICLK_PIN to GPIO_PULL_DIS done!
[    1.609386] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin DAIPCMIN already requested by 1000b000.pinctrl; cannot claim for mediatek,connectivity-combo
[    1.612027] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin-34 (mediatek,connectivity-combo) status -22
[    1.613945] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: could not request pin 34 (DAIPCMIN) from group DAIPCMIN  on device 1000b000.pinctrl
[    1.616402] .(3)[131:kworker/u8:3]mediatek,connectivity-combo mediatek,connectivity-combo: Error applying setting, reverse things back
[    1.618481] .(3)[131:kworker/u8:3]wmt_gpio:set GPIO_PCM_DAIPCMIN_PIN to GPIO_PULL_DIS done!
[    1.619927] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin DAIPCMOUT already requested by 1000b000.pinctrl; cannot claim for mediatek,connectivity-combo
[    1.622582] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin-35 (mediatek,connectivity-combo) status -22
[    1.624500] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: could not request pin 35 (DAIPCMOUT) from group DAIPCMOUT  on device 1000b000.pinctrl
[    1.626987] .(3)[131:kworker/u8:3]mediatek,connectivity-combo mediatek,connectivity-combo: Error applying setting, reverse things back
[    1.629065] .(3)[131:kworker/u8:3]wmt_gpio:set GPIO_PCM_DAIPCMOUT_PIN to GPIO_PULL_DIS done!
[    1.630526] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin DAISYNC already requested by 1000b000.pinctrl; cannot claim for mediatek,connectivity-combo
[    1.633152] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: pin-36 (mediatek,connectivity-combo) status -22
[    1.635070] .(3)[131:kworker/u8:3]mediatek-mt8173-pinctrl 1000b000.pinctrl: could not request pin 36 (DAISYNC) from group DAISYNC  on device 1000b000.pinctrl
[    1.637497] .(3)[131:kworker/u8:3]mediatek,connectivity-combo mediatek,connectivity-combo: Error applying setting, reverse things back
[    1.639576] .(3)[131:kworker/u8:3]wmt_gpio:set GPIO_PCM_DAISYNC_PIN to GPIO_PULL_DIS done!
[    1.641004] .(3)[131:kworker/u8:3]wmt_gpio: gpio init done!
```

Lastly, it appears mainline kernel support for mediatek devices is improving by small steps.  https://patchwork.kernel.org/project/linux-mediatek/patch/20231220121110.1441160-3-vignesh.raman@collabora.com/ mentions a powervr driver which has been merged into the kernel, though there's still some work to do for actual mediatek support.  Still, there's hope that we might be able to (eventually?) migrate away from the now-ancient 3.18.19 kernel in use for suez.

hexdump0815, 2024-01-05:
@PinkFreud - i do not want to be too pssimistic, but i think there will be no newer/better kernel for those amazon tablets as the bootloader is too non-cooperative, there is no documentation available and the legacy mediatek kernel (source code) used on those devices is some of the worst code i have seen so far :)

toywood, 2024-01-14:
Very very thank you! hexdump0815.This img is perfect for my arm chromebook-samsung chromebook plus.The xfce system is design so smart:video play,sound volume key ,backlight key,wifi,bluetooth,2d accessary,touchscreen,touchpad,and magnet pen,all of them is work so  wonderful ,it has full function drived to runs much better than x86 chromebook by so easy to do,just dd the img to a sd card.So thank you again!This machine is wonderful eighter ,for light weight ,little power consume ,elegant screen with touchscreen and magnet screen,and can insatll linux in sd card ,so can add no external disk outside with  nowipe the chromeOS,it is perfect!It  better than an apple arm Macbook,for slim magnet pen, elegant screen,so many working linux app.Your work is fit the arm chromebook excellent!
     I have a sugest:can it run as debian live as an overlayfs union lay to avoid the too many write data to sd card flash media ? I have find a project:https://github.com/fitu996/overlayroot.sh.it can do this job ,but wo can not change the img file. All i can do is change home,log,tmp to tmpfs by usr fstab.

toywood, 2024-01-14:
Very very thank you! hexdump0815.This img is perfect for my arm chromebook-samsung chromebook plus.The xfce system is design so smart:video play,sound volume key ,backlight key,wifi,bluetooth,2d accessary,touchscreen,touchpad,and magnet pen,all of them is work so  wonderful ,it has full function drived to runs much better than x86 chromebook by so easy to do,just dd the img to a sd card.So thank you again!This machine is wonderful eighter ,for light weight ,little power consume ,elegant screen with touchscreen and magnet screen,and can insatll linux in sd card ,so can add no external disk outside with  nowipe the chromeOS,it is perfect!It  better than an apple arm Macbook,for slim magnet pen, elegant screen,so many working linux app.Your work is fit the arm chromebook excellent!
 I have a sugest:can it run as debian live as an overlayfs union lay to avoid the too many write data to sd card flash media ? I have find a project:https://github.com/fitu996/overlayroot.sh.it can do this job ,but i can not edit the img file. All i can do is change home,log,tmp to tmpfs by usr fstab.Can you post a new img with the overlayfs and ro-root disk?

PinkFreud, 2024-01-14:
> @PinkFreud - i do not want to be too pssimistic, but i think there will be no newer/better kernel for those amazon tablets as the bootloader is too non-cooperative, there is no documentation [...]

Yeah, I know not to hold my breath, but as long as the chipset is slowly gaining support in the kernel, one can hope.  ;)

I've already worked around the DPMS issue where the screen never re-draws after turning the display back on with a script that watches DPMS status, and forces a redraw by doing a `chtvt 1 ; chvt 7`.  This image is still far from perfect, but good enough to be a Nextcloud client, so that's something, I suppose.

cdd1000, 2024-02-11:
it's pretty cool you were able to get full linux working on these arm chromebooks, definitely not something i thought would ever work!

hexdump0815, 2024-02-11:
just a little note to put the credits where they belong: i mostly just put together the work of lots of other people without which something like this project would not be possible :)

gertk64, 2024-03-18:
Tried the debian bookworm image on a Lenovo S330 and it seems to work fine. Sound (after selecting speakers/headphones), WiFi, charging/mains detect. Great work. Now trying to make the install permanent.

gociii, 2024-03-27:
Tested bookworm image on **odroid-u3+** https://github.com/hexdump0815/imagebuilder/releases/tag/230915-01

Areas of interest:

-IPv4 only
-CONNTRACK & NETFILTER for use with WireGuard
-CLI reset using shutdown / reboot
-Hard reset (from experience these devices had failed rebooting in the past due to not having peripheral connections such as HDMI connected).

All good :) Thanks....

Looking for the set-server-script.sh shared by someone time ago to get rid of the GUI. @hexdump0815  any pointers for this please?

hexdump0815, 2024-03-29:
you mean this one? - https://github.com/hexdump0815/imagebuilder/blob/main/files/extra-files-bookworm/scripts/desktop-to-server.sh or https://github.com/hexdump0815/imagebuilder/blob/main/files/extra-files-jammy/scripts/desktop-to-server.sh ... the next round of images will have it by default, not sure yet when those will be done - in the meantime just take the file from the repo

NightFurySL2001, 2024-03-31:
@gociii just wondering did you boot the image from eMMC or SD card?

hexdump0815, 2024-03-31:
just to add a little note here myself: i'm using my kukui kappa chromebook as my usual daily machine running my debian bookworm setup on it for a fewmonths now, always putting it to sleep over night or when i do not use it for longer. this way i'm at an uptime of 48 days now without any real reason to reboot it yet - those chromebooks might not yet be perfect with mainline, but they are very stable, reliable and useable and thanks to things like zswap with zstd compression for memory and btrfs with zstd compression for the disk (luks encrypted in my setup) even this little machine with only 4gb of ram and 32gb of emmc is surprisingly useable with firefox and sometimes around 50-100 tabs open :)

[31 more replies]