velvet-os/imagebuilder #142: No installer Icon

Rcjohn003, 2023-03-24:
I have recently put Ubuntu on my arm Chromebook but it has no installer Icon on any of the desktops if anybody has any informations on this issue that would be great thanks!

hexdump0815, 2023-03-24:
@Rcjohn003 - not sure if i understand you correctly, but those images here are not supposed to run an installer, but instead they are completely installed systems already and ready to run (so they are kind of what you would get after running the installer)

Rcjohn003, 2023-03-29:
Yes got it Thanks!

techorix, 2024-01-21:
I looked extensively and wanted to ask if there is a possible way to install from the SD card to the hard drive you could point out? 

I mean this doesn't seem to be the purpose of the image but I was so happy that it ran on my Veyron/mighty I want to use it as exclusive as possible now that the original OS has long reached EoL.

hexdump0815, 2024-01-21:
for arm chromebooks there is some info here: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

Kekobi, 2025-07-13:
> for arm chromebooks there is some info here: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

This doesn't seem to exist anymore. I would like to install from the sd card to the hard drive too, how can I do this?