velvet-os/imagebuilder #143: The wifi is not working

Patola, 2023-03-25:
Hello, it seems your image does not include the firmware file for the VisionFive kit that comes with wifi. It shows the following in dmesg:
```
[Sat Mar 25 15:45:31 2023] usb 1-1.2: new high-speed USB device number 6 using xhci_hcd
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_usb_probe entry, reset slave !!
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_usb_probe entry, func: 0, g_usb: 0000000002807abe !!
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_tx_comp_thread entry
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_usb_probe kthread_run success.
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_rx_comp_thread entry
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_usb_probe exit!!
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_usb_probe entry, reset slave !!
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_usb_probe entry, func: 1, g_usb: 0000000002807abe !!
[Sat Mar 25 15:45:32 2023] [ecrnx] fw file name: ECR6600U_transport.bin
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_fw_file_chech, Checking firmware... (ECR6600U_transport.bin)
[Sat Mar 25 15:45:32 2023] eswin_usb 1-1.2:1.0: Direct firmware load for ECR6600U_transport.bin failed with error -2
[Sat Mar 25 15:45:32 2023] [ecrnx] eswin_fw_file_chech, error status = -2
[Sat Mar 25 15:45:32 2023] [ecrnx] MAC Address is:00:11:22:33:44:5f
[Sat Mar 25 15:45:32 2023] [ecrnx] host driver log level is:3 
[Sat Mar 25 15:45:32 2023] [ecrnx] firmware log level is:2 
[Sat Mar 25 15:45:32 2023] [ecrnx] firmware log level type:0 (print to chip's uart) 
[Sat Mar 25 15:45:38 2023] [ecrnx] cmd timed-out queue_sz:1
[Sat Mar 25 15:45:38 2023] [ecrnx] tkn[0]  flags:001a  result: -4  cmd:8299-MM_GET_CAL_RESULT_REQ    - reqcfm(8300-MM_GET_CAL_RESULT_CFM)
[Sat Mar 25 15:45:38 2023] ieee80211 phy2: get cal result failed
[Sat Mar 25 15:45:38 2023] [ecrnx] q_sz/max:  0 /  8 - next tkn: 1
```

Patola, 2023-03-25:
It would be package `firmware-misc-nonfree` but I don't see anything named ECR6600U_transport.bin in there.

hexdump0815, 2023-03-26:
i think it will take some time until it lands in firmware-misc-nonfree - i'll most probably add a script to /scripts to install the firmware until this happens - thanks a lot for reporting this

Patola, 2023-03-26:
You're welcome. I've downloaded the firmware from https://github.com/eswincomputing/eswin_6600u (files ECR6600S_transport.bin for SDIO and ECR6600S_transport.bin for USB) and put it under /lib/firmware, and it seemingly works but times out afterwards, know what can be done?

I've pasted a partial dmesg shortly after and before the firmware is loaded and fails by timeout.
[dmesg.txt](https://github.com/hexdump0815/imagebuilder/files/11071944/dmesg.txt)

hexdump0815, 2023-03-26:
maybe @jakubtalich from https://github.com/hexdump0815/imagebuilder/issues/141#issuecomment-1482527646 has an idea as he seems to have gotten it working? i do not have the wifi dongle, so i cannot test anything myself.

jakubtalich, 2023-03-27:
There are multiple versions of the firmware, StarFive added their own patches.
Try [this one](https://github.com/dtometzki/starfive/tree/master/firmware) or the one from StarFive.
More info on the versions here: https://forum.rvspace.org/t/ecr6600u-firmware/2368
Discussion [here](https://forum.rvspace.org/t/how-to-compile-stuff-in-my-case-wifi-module-on-the-vf2/1436) and [here](https://forum.rvspace.org/t/usb-module-eswin-6600u/979).