velvet-os/imagebuilder #144: cgpt error when creating partition on veyron-speedy (C201)

gregordinary, 2023-03-27:
**Issue**
Using [Feb 18th image](https://github.com/hexdump0815/imagebuilder/releases/tag/230218-05) and following steps to [install to emmc for veyron](https://github.com/hexdump0815/imagebuilder/blob/8102dc8fa2376ead856d74ad0e4e4d5b53fc518f/doc/install-to-emmc-on-arm-chromebooks.md#veyron-is-special), the following command:

`root@changeme:~# cgpt create /dev/$TGTDEV`

Results in the following error:
```
> ERROR: Can't seek: Invalid argument
> ERROR: Cannot read secondary GPT header
```

**Cause**
The issue seems to be with the version of cgpt bundled with Bookworm (and Sid). 

**Solution**
Fix was to downgrade to the version bundled with Bullseye. After doing so I was able to execute the cgpt commands successfully. 

**Source for Resolution** 
[Same/similar issue](https://github.com/Maccraft123/Cadmium/issues/127#issuecomment-1207237595) was identified on Cadmium Linux.

hexdump0815, 2023-03-27:
@gregordinary - thanks a lot for the feedback ... i'll try to find a solution of how to handle it for future images

hexdump0815, 2023-09-30:
fyi: as a temporary solution i have now at least added some info and the workaround here - https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md#23-09-25-cgpt-seems-to-be-broken-on-32bit-armv7l-systems-in-debian-bookworm