velvet-os/imagebuilder #147: Kukui Image

Rcjohn003, 2023-04-15:
I was just wondering how long are these Images or OS updated for?  Or when a new kernel will develop if any body has any info about this I was just curious Thanks!

hexdump0815, 2023-04-15:
there is no fixed schedule for updates - i'm building new images usually 2-3 times a year and often new images get updated kernels as well, sometimes i build new kernels inbetween as well when i discover some important patch or some missing config options ... all in all this is not intended as a real distribution with regular updates etc., but rather ment as a starting point for people to get started with the systems and then i try to encourage people to build their own kernels (see the notes in the doc dirs for some info about such topics) - the base os (debian/ubuntu) usually is supported for at least three years (often more) with updates, so that will just go on and on (see /scripts/update-systems.h)

Rcjohn003, 2023-04-15:
Alright very good Thanks! 👍