velvet-os/imagebuilder #148: HP Chromebook x2 11 ARM 7c Strongbad (not Trogdor)

mohuddle, 2023-04-20:
I stumbled on this after posting a message to Reddit. If I were to try and test this out, is it just a matter of booting the image from Developer Mode on the Chromebook of my microSD?

hexdump0815, 2023-04-21:
@mohuddle - it should work - see: https://github.com/hexdump0815/imagebuilder/issues/44 ... strongbad are the trogdor tablets basically but both are very similar

mohuddle, 2023-04-21:
No issues booting from sdcard, still poking around. Resolution scaling is too small for these old eyes ;) but I scaled down to .5 and it works.

Are there plans to have an installer to replace chromeOS that's installed on the device? Is that possible?

unixabg, 2023-04-22:
@mohuddle - this document might help you install to the emmc: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

mohuddle, 2023-04-22:
Thanks @unixabg , appreciate it.

hexdump0815, 2023-04-23:
@mohuddle - just out of curiosity: which reddit thread it was mentioned in? thanks to @unixabg for answering the other questions already ...

mohuddle, 2023-04-23:
> @mohuddle - just out of curiosity: which reddit thread it was mentioned in? thanks to @unixabg for answering the other questions already ...

/r/chromeos

junnikokuki, 2023-06-14:
Does touchscreen work out of box? Or I have to do sth to make it work?

hexdump0815, 2023-06-14:
@junnikokuki - according to https://github.com/hexdump0815/imagebuilder/issues/44 it should be possible to make it work i think

junnikokuki, 2023-06-15:
Maybe that document needs update. It saids 5.19.

hexdump0815, 2023-06-16:
it should hopefully still apply