velvet-os/imagebuilder #150: Arch Linux on Kukui with mainline Arch Linux Kernel

aneeshlingala, 2023-06-12:
I found a mainline kernel in ArchLinuxARM supporting kukui devices. It is the linux-aarch64-chromebook kernel.
I downloaded an EndeavourOS rootfs, installed the kernel with "sudo pacman -S linux-aarch64-chromebook" Then, I fixed the bluetooth and sound by downloading the extra files folder for kukui on your repo and copying it. Then, I fixed bluetooth by created /etc/modprobe.d/blacklist.conf with the contents "install bluetooth /bin/false". Then, I updated the initramfs with "sudo mkinitcpio -P." Then I created two partitions. One for the kernel (/dev/mmcblk0p1), and one for the root (/dev/mmcblk0p2), formatted as ext4. Then, I made an fstab for the root partition. Then, I copied the files and boot into the system, connected to the internet with nmtui and installed GNOME. It runs like a charm. This guide works for ArchLinuxARM and EndeavourOS ARM.

rissole768, 2023-06-12:
bro thanks i'm gonna try this

rissole768, 2023-06-12:
I'll try this on my end with my (also kukui based) device.

aneeshlingala, 2023-06-12:
Sure

aneeshlingala, 2023-06-12:
How's it going?

aneeshlingala, 2023-06-14:
Is everything good?

rissole768, 2023-06-16:
I sorta forgot to post an update here so i guess i'll post it now.
My USB drive died.
I can't test it.

aneeshlingala, 2023-06-16:
Tell me when you can, and also which kukui device are you using? If it is something other than camera, then please give me feedback when you get back to this.

sea768, 2023-06-21:
I'm using juniper. Sorry for not mentioning the specific device

aneeshlingala, 2023-07-21:
I am currently making a script for this

Also note that only GDM and GNOME works for some reason

sea768, 2023-09-11:
Now that i have a USB drive that works i'll test it as soon a i can.

aneeshlingala, 2023-09-13:
Sure, Good luck!

Ask me if you have any problems

AT14C, 2023-12-12:
This does not work anymore, it simply leaves me with a black screen on boot

PD-Kerman, 2024-06-11:
> Sure, Good luck! Ask me if you have any problems
> […](#)

Hellow there, it’s good to see a chance that kukui can have arch Linux support, is there any updates on this project~? :D

aneeshlingala, 2024-06-11:
I have issues on this project, summer break is coming up in a few weeks so
will look back into this project with downloadable images

PD-Kerman, 2024-06-11:
> I have issues on this project, summer break is coming up in a few weeks so
> 
> will look back into this project with downloadable images
> 
> 
> 
> 
> 
> 
> > Sure, Good luck! Ask me if you have any problems
> 
> > … <#m_6385623511292518125_>
> 
> > On Sun, Sep 10, 2023 at 9:39 PM c h e l s e a *@*.*> wrote: Now that i
> 
> > have a USB drive that works i'll test it as soon a i can. — Reply to this
> 
> > email directly, view it on GitHub <#150 (comment)
> 
> > <https://github.com/hexdump0815/imagebuilder/issues/150#issuecomment-1713151833>>,
> 
> 
> 
> 
> 
> >
> 
> > Hellow there, it’s good to see a chance that kukui can have arch Linux
> 
> > support, is there any updates on this project~? :D

That would be really great! I'm very looking forward for it, thank you for the reply and the great work!