velvet-os/imagebuilder #153: chromebook_snow-armv7l-ubuntu.img and chromebook_snow-armv7l-focal.img don't boot for me.

Marietto2008, 2023-07-02:
Hello.

I've dd'ed those images to my sd card,but they didn't boot :

chromebook_snow-armv7l-ubuntu.img : from 200206-01
chromebook_snow-armv7l-focal.img : from 210321-01

do you know the reason ? how can I fix it ? It seems that these images use u-boot as bootloader. Instead the images created with syslinux as bootloader are able to boot correctly for me.

hexdump0815, 2023-07-02:
maybe it is because of "the snow chromebooks seem to not boot properly with all sd cards ... if booting does not work it might be worth to try different sd cards ... older and smaller ones seem to have the higher chance to work well" - from https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_snow#readme - otherwise they should boot just fine as they worked well for me as for others too ... good luck

hexdump0815, 2023-07-02:
little update: at least the later snow images have two different u-boot images in /boot/extra of the written sd card - it might be worth trying both separately even on the older images - maybe one works better than the other for certain sd cards

Marietto2008, 2023-07-02:
Which sd card model do you suggest to me to buy ?

hexdump0815, 2023-07-02:
i have no real overview which works and which not - its a good idea to try both of the u-boot images which are inside of the latest images, maybe one of them already works with a given card  ... in general it looks like older cards seem to work more reliable than newer ones maybe, but in the end it depends on the actual card in the end

Marietto2008, 2023-07-03:
Are you sure that it's enough to change the type of the sd card to make your images to boot ? because I'm trying a lot of images on different sd cards but any image has been able to boot. Probably I should completely erase the sd card with sudo dd if=/dev/zero of=/dev/mmcblk1 status=progress bs=2M ?

hexdump0815, 2023-07-03:
sgdisk -Z /dev/xyz should be enough to wipe gpt partition info ... the images should work for sure as i test booted them all and others used them too - did you try both u-boot images from /boot/extra?

Marietto2008, 2023-07-03:
Can you imagine how many combinations there are ? I can create a matrix putting on the vertical line some kind of sd cards and on the horizontal lines the u-boot images.