velvet-os/imagebuilder #154: What's the bootloader used on your newest images ?

Marietto2008, 2023-07-03:
Hello.

Its again me. I'm still working on the old project to enable KVM on the kernel 5.4 on the samsung arm chromebook. I saw that you have two kinds of images on your repository. The images that uses u-boot as bootloader and the the other ones,that are more recent,that use another bootloader. I don't know what is this. I imagined that it was syslinux,but it isn't. I've got an idea to enable KVM but I'm not sure if I can develop it. Since the images that use u-boot aren't patched to enable KVM and only the virtual open system custom u-boot can,I've thought to chainload that version of u-boot with the bootloader that you have used on your newest images. To do this I need to know what's the name. Very thanks.

hexdump0815, 2023-07-04:
it is u-boot - the extlinux functionality in u-boot is very similar to syslinux i think

Marietto2008, 2023-07-04:
Did you understand my point ? I want to explain it as best as possible. Let's take this tutorial as a base :

http://www.virtualopensystems.com/en/solutions/guides/kvm-on-chromebook/

it does the following tasks :

1) it installs the unlocked u-boot custom version created by them on the first partition
2) the script stores on the 3 partition two only files : uImage and exynos 5250-snow.dtb (and not :

d) config.5.18.1-stb-cbe+ : file
e) initrd.img-5.18.1-stb-cbe+ : file
f) System.map-5.18.1-stb-cbe+ : file
g) vmlinux.kpart-5.18.1-stb-cbe+ : file
h) zImage-5.18.1-stb.cbe+ : file

3) default bootstring is hardcoded there :

https://github.com/virtualopensystems/u-boot/blob/eecfeb578e296ef3b739ac918a39b6c177dff58a/include/configs/snow.h#L101

but :

#define CONFIG_EXTRA_ENV_SETTINGS \
EXYNOS_DEVICE_SETTINGS \
"dtaddr=0x43000000\0"\
"initrdaddr=0x44000000\0"\
"boot_noinitrd=mmc dev 1 ; mmc rescan 1 ; ext2load mmc 1:3 ${loadaddr} uImage ; ext2load mmc 1:3 ${dtaddr} exynos5250-snow.dtb ; bootm ${loadaddr} - ${dtaddr}\0"\
"boot_initrd=mmc dev 1 ; mmc rescan 1 ; ext2load mmc 1:3 ${loadaddr} uImage ; ext2load mmc 1:3 ${initrdaddr} initrd ; ext2load mmc 1:3 ${dtaddr} exynos5250-snow.dtb ; bootm ${loadaddr} ${initrdaddr} ${dtaddr}\0"\
"bootdelay=3\0"

as you can see,it does not define any file like those shown above. I mean,these files aren't hardcoded :

d) config.5.18.1-stb-cbe+ : file
e) initrd.img-5.18.1-stb-cbe+ : file
f) System.map-5.18.1-stb-cbe+ : file
g) vmlinux.kpart-5.18.1-stb-cbe+ : file
h) zImage-5.18.1-stb.cbe+ : file

so,my question is : if in both cases u-boot is used,how can I hardcode even the missing files ? thanks.