velvet-os/imagebuilder #156: Lots of Btrfs warning and error messages.

EMLommers, 2023-07-23:
Using the latest release I got lots of btrfs warnings and messages
checked with btrfsck and btrfs check --repair --force /dev/mmcblk0p4 , but no errors or faults found.

It is a bug?

hexdump0815, 2023-07-23:
@EMLommers - can you try "btrfs scrub start /dev/xyz" - this will recalculate and compare the checksums of all files - you can follow the progress with "btrfs scrub status /dev/xyz" ... i guess this is on usb? it seems the usb implementation of various chromebooks is not really clean if used as rootfs as a chromebook with chromeos is never dong this - this might in some case result in fs corruption. i saw on a kukui krane with broken emmc which i'm running from usb and krane only has a single usb connector to which i connected a usb-c hub with usb-c-power - whenever i connect/disconnect the power usb goes down the drain and with it the filesystem.

EMLommers, 2023-07-23:
Actually it was on mmcblk0, internal mmc, it did not happen on sdcard, for a strange reason my OAK will not boot any USB sticks.
I am currently running from sdcard and when dd to internal mmc. the messages appear. Will dd again and see if it happens again. (I killed my other installation on internal due to building 6.4 kernel)

hexdump0815, 2023-07-23:
oh - dd to emmc is risky with btrfs - see: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md ... that could explain the btrfs errors actually

EMLommers, 2023-07-23:
I have again dd to internal mmc (mmcblk0) and sync, and run the following /scripts in the following order:
rename-default-user.sh
extend-rootfs.sh
recreate-swapfile.sh

The brtfs error from above did not occur , however errors among:
- systemd-gpt-auto-generator[200]: File system behind root file system is reported by btrfs to be backed by pseudo-device /dev/root, which is not a valid userspace accessible device node. Cannot determine correct backing block device.
 -systemd[195]: /usr/lib/systemd/system-generators/systemd-gpt-auto-generator failed with exit status 1.
- BTRFS info: devid 1 device path /dev/root changed to /dev/mmcblk0p4 scanned by systemd-udevd (258)
- rtkit errors.

[dmesg-internal.txt](https://github.com/hexdump0815/imagebuilder/files/12139698/dmesg-internal.txt)
[journalctl-internal.txt](https://github.com/hexdump0815/imagebuilder/files/12139699/journalctl-internal.txt)