velvet-os/imagebuilder #157: Running script install_gl4s.sh failed

EMLommers, 2023-07-23:
File not found....

hexdump0815, 2023-07-29:
which system and version is this on?

EMLommers, 2023-07-30:
Your latest oak version.
Chromebook R13.

hexdump0815, 2023-07-30:
thanks for the info - i have noted it down, will have to rework all this a bit - not sure when i'll get to it ... my latest gl4es build can be found here: https://github.com/hexdump0815/mesa-etc-build/releases/tag/220604-01 ... on oak it does not make any sense, as there is no gpu accel due to the imgtec gpu yet - one day it will hopefully be there (its currently being open sourced, but that will still take some time until it will get useable)

EMLommers, 2023-07-31:
Therefore was also my comment at issue #34., and imagination kernel 6.4