velvet-os/imagebuilder #158: problem installing kuku imagei to emmc

linreigu, 2023-08-19:
I've been trying to install a linux version on my Lenovo Duet for much of the past month, but your imagebuilder is the only thing that works. There is so much useless advice out there. You are great!

Stupid problem remains. I can't install to emmc (following your "very easy advice") since I can't fetch the image with wget. It stops and times out. I guess my Duet takes too long to read from the USB or perhaps there is a memory problem.

Is there some other way to get the image file? As in, could I install it on emmc of the Chromebook and take it from there or could I preload it onto the USB?

Again, thanks for your hard work and excellent instructions.

Erik

hexdump0815, 2023-08-19:
i have no idea what your setup is - if you have some kind of hub connected to your krane then you could put the image onto another usb device or an sd card and mount/take it from there ... if you only have the usb disk you booted from you might be able to create an extra partition on it with some filesystem onto which you put the image maybe?

linreigu, 2023-08-19:
Sorry, I should have explained -- no hub. But I'll try your extra partition. (It would be easier if I can launch Firefox and go to your website, but there isn't enough memory for that ...).

hexdump0815, 2023-08-20:
oh wait - you can just run /scripts/extend-rootfs.sh to extend the rootfs to the full usb size - then it should work with firefox as well i guess ...

linreigu, 2023-08-21:
Yes, of course. I'll try.

linreigu, 2023-08-21:
Hi back,

Great news: the Jammy installation on my Lenovo Duet is now up and running. It took me a months to set this up (I'm just a happy amateur when it comes to computers) but now it's working beautifully. Yours were the only images and instructions that were any good. A million thanks!

I didn't try to download with wget in the end -- it kept on stalling. Instead I took your advice and expanded rootfs to 100%, then I took the USB out, mounted it on my desktop Ubuntu, mounted /sda3 and wrote the image.gz file to /home/linux/Downloads. Then I dd:ed the file from there to emmc.

Small issue remaining: Jammy is now mounted on dev/mmcblk03 on the SD card. There is next to no disk space left. How can I increase it? Gparted?

Again, thanks so much. You are the best!

closed 2023-08-21 by linreigu