velvet-os/imagebuilder #163: mwifiex_sdio mmc2:0001:1: CMD_RESP: cmd 0x23f error, result=0x2 on your image Debian Bookworm for the ARM Chromebook

Marietto2008, 2023-09-03:
Hello.

on my last experiment I've switched from Devuan 4 to Debian bookworm and I've used your last image (I've got the image file here : 

[](https://github.com/hexdump0815/imagebuilder/releases/download/230220-01/chromebook_snow-armv7l-bookworm.img.gz)

I'm using it,but I see some huge errors that are destroying my system files reboot after reboot,because they are produced at a regular intervals :

```
[   10.480204] mwifiex_sdio mmc2:0001:1: info: FW download over, size 533976 bytes
[   10.719132] mwifiex_sdio mmc2:0001:1: WLAN FW is active
[   10.719369] Bluetooth: vendor=0x2df, device=0x912a, class=255, fn=2
[   11.089624] btmrvl_sdio mmc2:0001:2: sdio device tree data not available
[   11.091551] mwifiex_sdio mmc2:0001:1: CMD_RESP: cmd 0x242 error, result=0x2
[   11.091563] mwifiex_sdio mmc2:0001:1: mwifiex_process_cmdresp: cmd 0x242 failed during initialization
[   11.133736] mwifiex_sdio mmc2:0001:1: info: MWIFIEX VERSION: mwifiex 1.0 (14.68.29.p59) 
[   11.135281] mwifiex_sdio mmc2:0001:1: driver_version = mwifiex 1.0 (14.68.29.p59) 
[   12.914015] NET: Registered protocol family 38
[   19.058217] mwifiex_sdio mmc2:0001:1: info: trying to associate to 'ziomario' bssid 
[   19.093260] mwifiex_sdio mmc2:0001:1: info: associated to bssid  successfully
[   19.116924] mwifiex_sdio mmc2:0001:1: CMD_RESP: cmd 0x23f error, result=0x2
[   19.129679] IPv6: ADDRCONF(NETDEV_CHANGE): mlan0: link becomes ready
[   19.134791] mwifiex_sdio mmc2:0001:1: CMD_RESP: cmd 0x23f error, result=0x2
[   31.849443] TPS65090_RAILSDCDC1: disabling
[   31.858024] TPS65090_RAILSDCDC2: disabling
[   31.868245] TPS65090_RAILSDCDC3: disabling
[   31.878250] TPS65090_RAILSLDO1: disabling
[   31.887352] TPS65090_RAILSLDO2: disabling
[  115.862293] mwifiex_sdio mmc2:0001:1: CMD_RESP: cmd 0x23f error, result=0x2
```

how can I fix it ? If I don't get wrong,it is the wireless driver that you have added to the system that's causing the problem,right ? or maybe it is incompatible with the Debian 12 ? let me know please. I don't want to return to Devuan because libvirt is easier to configure on Debian. But take in consideration that those errors don't happen on Devuan.

hexdump0815, 2023-09-03:
sorry - i so far did not see those errors yet on snow, at least not during my bookworm image testing - does anything not work as a result? you are right that seems to come from the wireless device/driver.

Marietto2008, 2023-09-03:
Sometimes it disconnects and reconnects. But the real problem is that it seems that the error destroy the file system slowly. It forces the file system to check and fix errors produced by the frequent errors produced by the bad state of the driver.