velvet-os/imagebuilder #166: Trying to permanently install the FENNEL14 img with dd does not work at all.

Dae-Hang-Sook, 2023-09-20:
Upon reboot it fails to read the SDcard and the HardDrive.
I do know that's entering dev mode, pushing ctrl + alt + f2 (-->) and using mount -o remount,rw / ; /usr/share/vboot/bin/make_dev_ssd.sh --remove_rootfs_verification --partitions 2 works instantly. Any steps to "update the kernel" via this way? It would be incredibly helpful if I had extra info on this. Thanks!

hexdump0815, 2023-09-21:
can you please dexcribe what exactly you did and what exactly is the problem? the above explaination is not really clear to me.

Dae-Hang-Sook, 2023-09-21:
> can you please describe what exactly you did and what exactly is the problem? the above explaination is not really clear to me.

Okay, so using the sudo dd method to permanent install did not work. Using the above commands I found out my kernel was locked and now it is not. Would you have a method to replace the kernel with something you've created so that I may make this process possible?

Dae-Hang-Sook, 2023-09-28:
Being stopped at " zcat chromebook_kukui-aarch64-jammy.img.gz | dd of=/dev/$TGTDEV bs=1024k status=progress

gzip: chromebook_kukui-aarch64-jammy.img.gz: not in gzip format
0+0 records in
0+0 records out
0 bytes copied, 0.000210616 s, 0.0 kB/s"

closed 2023-09-30 by Dae-Hang-Sook

hexdump0815, 2023-09-30:
did you get your problem solved? to me it looks like you did not extend the rootfs, so the wget die not have enough space to write the image .gz file completely

Dae-Hang-Sook, 2023-09-30:
> did you get your problem solved? to me it looks like you did not extend the rootfs, so the wget die not have enough space to write the image .gz file completely

No, I never did. I had already extended the rootfs to the rest of the 32gb sd card I was using at the time.

hexdump0815, 2023-09-30:
i have no idea then why it did not work.

Dae-Hang-Sook, 2023-09-30:
> i have no idea then why it did not work.

I will look into this more. I'm running a few of the pre-downloaded scripts to see what's available to me at this time.

Dae-Hang-Sook, 2023-09-30:
@hexdump0815 I'd also like to ask about other desktop interfaces that may be available. For now it seems all you have is xfce which is really not that useful to me (mainly due to the things I do) So maybe some point in time you might be able to make desktops for KDE Plasma, Mate, etc.??

Dae-Hang-Sook, 2023-09-30:
@hexdump0815 urgent info for you.
I managed to finally write to my eMMC but there's one problem. "End kernel panic - not syncing: Attempted to kill init! Exitcode=0x00000200

Dae-Hang-Sook, 2023-09-30:
> @hexdump0815 urgent info for you.
> I managed to finally write to my eMMC but there's one problem. "End kernel panic - not syncing: Attempted to kill init! Exitcode=0x00000200

This is occurring in the SDcard as well

Dae-Hang-Sook, 2023-10-19:
Hey, I have a serious question. How are you managing all of this? Google kukui fennel14 mt8183

- Add missing SVS
- Check/add/test mali clocks, opp tables
- Add SCP definition in MDP node, check GCE correctness
- Add missing DPI0 
- mdp3_corr: add mdp3-id
- Add missing DIP (Digital Image Processor)
- Add missing FD (Face Detection)
- Add missing vcodec_dec and vcodec_enc (HW Video Decoding/Encoding)
- Add missing camisp (Camera Image Signal Processor)

- Add missing HDMI codec to mt8183-sound
- Add missing DPI0 binding for IT6505 (USB-C AltDP)
- Add mali power models
- Add definition for IT6505 to i2c3
- State driver type for mmc1
- Add GPIO (PINMUX) definitions for cameras and DPI
- State IT6505 GPIOs

- Add missing sensor + lens oscilator definitions to i2c2 for OV8856/DW9768 (main/back)
- Add missing sensor definition to i2c4 for OV02A10 (front)
- Add missing seninf (Sensor Interface - MIPI-CSI2)

Dae-Hang-Sook, 2023-10-28:
Hey, HexDump

A question

Are you using the newest linux kernel version?
If so there should be added support for 'Mediatek SoC'
In the gpu, drm, platform specific. If you've not added these, you should. The requirements of the mt8183 is that all of the hardware on the board relies on the software to work. I'm sure you'd know this

hexdump0815, 2023-10-29:
mt8183 is supported quite well in mainline linux which is used for the images with a few required patches on top - this is how the kernel gets built: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/lts/readme.mt8

Dae-Hang-Sook, 2023-11-02:
> mt8183 is supported quite well in mainline linux which is used for the images with a few required patches on top - this is how the kernel gets built: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/lts/readme.mt8

I think i should put you in contact with the PostMarketOS devs. They're building device trees and everything from scratch and aren't using mainline linux. They say that making mainline linux work is the entire point but they're using 6.1.39 only.

Dae-Hang-Sook, 2023-11-03:
@hexdump0815

mkfs -t ext4 -O ^has_journal -m 0 -L bootemmc /dev/${TGTDEV}p3

Returns this:

The file /dev/mmcblk0p3 does not exist and no size was specified

Using the dd method just tells me the file *direct link to it* is kot in gzip format

hexdump0815, 2023-11-24:
sorry, but i cannot really follow the problems you descibed when trying to install onto emmc - several people seem to have been able to install to emmc by following the instructions ... regarding the kernel you are correct: a special kernel is required as a regular mainline kernel does not offer all changes required to run properly on chromebooks for most of those

Dae-Hang-Sook, 2023-12-09:
> sorry, but i cannot really follow the problems you descibed when trying to install onto emmc - several people seem to have been able to install to emmc by following the instructions ... regarding [...]

About the "special" kernel needs. Fedora should work for any x86_64 (64-bit) devices do to it always having the latest drivers, firmware, etc for hardware needs.

hexdump0815, 2023-12-09:
fennel14 is not a x64_64 device but aarch64 and there the situation is sadly not that nice: many systems still require additional patches to work well with mainline linux and in the case of chromebooks also the boot procedure is non-standard

Dae-Hang-Sook, 2023-12-09:
> fennel14 is not a x64_64 device but aarch64 and there the situation is sadly not that nice: many systems still require additional patches to work well with mainline linux and in the case of chromebooks also the boot procedure is non-standard

No, no, no. It was simply a suggestion. About the booting. Contact chrultrabook devs for U-Boot which they use (they've made on for the mt8183 but it might not be stable yet)

hexdump0815, 2023-12-09:
ah - ok - thanks for the info about the chrultrabook work - i know that there is some mt8183 u-boot mainline port, but i think it is for a dev-board and not the chromeos devices ... for this project here i prefer to stay with using the default chromebook bootloader for now as it is working well for all the arm64 chromebooks