velvet-os/imagebuilder #167: touchscreen broken with cz1, while it works fine on postmarketos.

krrrak, 2023-09-29:


hexdump0815, 2023-09-29:
which image? which device exactly? any info in dmesg?

krrrak, 2023-09-30:
Device:asus chromebook detachable cz1 - katsu
Image:https://github.com/hexdump0815/imagebuilder/releases/tag/230917-01

what `dmesg | grep` shows on this image
`[    0.822113] i2c_dev: i2c /dev entries driver
[    0.981539] mediatek-dpi 14015000.dpi: Found bridge node: /soc/i2c@1100f000/it6505dptx@5c`
while on PostmarketOS,
`[    0.063085] platform 14015000.dpi: Fixed dependency cycle(s) with /soc/i2c@1100f000/it6505dptx@5c
[    0.861190] i2c_dev: i2c /dev entries driver
[    0.975509] mediatek-dpi 14015000.dpi: Found bridge node: /soc/i2c@1100f000/it6505dptx@5c
[    1.522509] i2c_hid_of_goodix 0-005d: supply vdd not found, using dummy regulator
[    1.901454] input: gt7375p 27C6:0E21 as /devices/platform/soc/11007000.i2c/i2c-0/0-005d/0018:27C6:0E21.0003/input/input7
[    1.902144] input: gt7375p 27C6:0E21 Stylus as /devices/platform/soc/11007000.i2c/i2c-0/0-005d/0018:27C6:0E21.0003/input/input8
[    1.902580] input: gt7375p 27C6:0E21 UNKNOWN as /devices/platform/soc/11007000.i2c/i2c-0/0-005d/0018:27C6:0E21.0003/input/input9
[    1.903198] input: gt7375p 27C6:0E21 UNKNOWN as /devices/platform/soc/11007000.i2c/i2c-0/0-005d/0018:27C6:0E21.0003/input/input10
[    8.625915] synth uevent: /devices/platform/soc/11012000.spi/spi_master/spi2/spi2.0/11012000.spi:cros-ec@0:i2c-tunnel/i2c-12/12-000b/power_supply/sbs-12-000b: failed to send uevent
`

is that useful?

hexdump0815, 2023-09-30:
yes - it looks like there is a kernel option missing which seems to be enabled on pmos i guess:
```
$ grep GOODIX config.mt8
CONFIG_TOUCHSCREEN_GOODIX=m
# CONFIG_I2C_HID_OF_GOODIX is not set
```
i'll add that to the kernel config soon

hexdump0815, 2023-10-01:
i have added the config option now ( https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/commit/2a75a38af57c7267e00931fa8d71810759e9fb41 ) and added a test kernel with it built in here: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.1.51-stb-mt8%2B ... here is some info about how to install a different kernel: https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md

krrrak, 2023-10-01:
seems still not work...

krrrak, 2023-10-02:
> seems still not work...

Just because it defaultly boot from the root fs on mmcblk but not usb......yes, touch screen works, but the system seems to get swapped x and y value from touch screen.

krrrak, 2023-10-02:
> > seems still not work...
> 
> Just because it defaultly boot from the root fs on mmcblk but not usb......yes, touch screen works, but the system seems to get swapped x and y value from touch screen.

refering to [https://wiki.postmarketos.org/wiki/Lenovo_IdeaPad_Duet_Chromebook_(google-krane)](url), I set an udev rule, but nothing changed, and after switch to gnome, the touchscreen works fine...