velvet-os/imagebuilder #171: HP 14-ed0xxx snapdragon 7c gen 2 working with snapdragon_7c_woa without touchpad and wifi (using acer aspire 1 version)

Bbertatum, 2023-10-07:
Thanks for the image builder and all scripts. Working really well following the instructions.

Would it be possible to include wifi and touchpad driver for HP 14-ed0xxx ? Seems to be this one :
https://www.qualcomm.com/products/technology/wi-fi/qca6174a
Also seems to be this driver too : https://linux-hardware.org/?id=pci:168c-003e-17aa-0827
On windows the touchpad shown as ELAN model.

Hopping this feedback helps and the OS could be fully fonctionnal on this model someday.

Thanks

hexdump0815, 2023-10-08:
it is not that easy to add support for more hardware as it is usually not that well documented in the case of windows on arm machines ... a good first step would be to dump the acpi tables and create a pull request like this one: https://github.com/aarch64-laptops/build/pull/83 in that repo for the dumps ... if they are available we will have to check if some useful information can be extracted from them ... as this device seems to be close to the refrence design (as it seems to work quite well already with the aspire 1 setup) maybe @TravMurav (who did most of the aspire 1 work) can help to get more things working ...

TravMurav, 2023-10-09:
Running the DT for a different device is generally a bad idea. You would need to create a new devicetree for your device and make necessary changes for the devices that don't work. Since you say that some hardware already works with the Aspire1 dt, we can start from that and check which parts work correctly and which are not.

For the wifi, I'd imagine you need to extract the firmware blobs from your windows install since they are usually paired to a specific device model or vendor with the digital signature checks. (I see the user has to manually copy those files per [doc/required-firmware.md](https://github.com/hexdump0815/imagebuilder/blob/main/systems/snapdragon_7c_woa/doc/required-firmware.md) for the imagebuilder builds)

For other stuff you would need to gather some intel on the device to better know how it's made

I'd recommend you to find the following things:
 - Extract the acpi tables and decompile them, make a pr as already suggested
 - Find FCC ID of the device and try to get "internal photos" document on fccid.io (alternatively can disassemble the device and make high res photos of the pcb - would be better quality)
 - If at all possible, try to find a schematic/repair manual for the laptop - it would be the best thing to aid in porting.
 
 Feel free to join `#aarch64-laptops` IRC room on OFTC, we are there if you have some quick questions
 
 FWIW there is also some notes in https://github.com/hexdump0815/imagebuilder/issues/136#issuecomment-1554131693 and around that might be useful