velvet-os/imagebuilder #174: Succssful ChromeBook Nyan Big Tegra K1 GPU Acceleration with NVIDIA Blobs

nvidialinuxuser, 2023-11-07:
Good morning,

I'm here to happily report that I've successfully installed Ubuntu 20.04 LTS on my Acer ChromeBook 13 (Nyan Big) with working GPU acceleration via the NVIDIA proprietary blobs (L4T/Linux for Tegra).  After some trial and error, I was able to set up Ubuntu to run with the proprietary NVIDIA driver stack by doing the following:

- [ ] Using and modifying your Debian Bullseye legacy image (as a base)
- [ ] Using your ChromeOS legacy kernel
- [ ] Ubuntu Xenial LTS packages (a few had to be repackaged in order to avoid dependency conflicts)
- [ ] [Using the NVIDIA L4T/Linux for Tegra script file](https://developer.nvidia.com/linux-tegra-r218)
- [ ] Changing the Tegra udev rule to enable total GPU access with the group "video" (just for my own personal convenience and sake of precaution)
- [ ] Readding the fstab file
- [ ] (Most likely optional) generating a new vmlinux.kpart using your excellent documentation but with total root filesystem read/write ('rw' command line switch)
- [ ] Editing `/etc/ld.so.conf.d/arm-linux-gnueabihf.conf` to manually include '/usr/lib/arm-linux-gnueabihf/tegra' and '/usr/lib/arm-linux-gnueabihf/tegra-egl'
- [ ] Launching XFCE from the command line with `startxfce4`

Note: I initially started with an Ubuntu Xenial chroot, pinned several Xorg packages, upgraded to Bionic (if I remember correctly), then to Focal.

Here is 'pin.pref' (located in /etc/apt/preferences.d'):

package: *xorg*
pin: version *
pin-priority: -1

package: libaudit*
pin: version *
pin-priority: -1

package: libfs6
pin: version *
pin-priority: -1

package: libglu1-mesa
pin: version *
pin-priority: -1

package: libmtdev1
pin: version *
pin-priority: -1

package: libpciaccess0
pin: version *
pin-priority: -1

package: libxkbfile1
pin: version *
pin-priority: -1

package: multiarch-support
pin: version *
pin-priority: -1

package: x11*
pin: version *
pin-priority: -1

package: xfonts*
pin: version *
pin-priority: -1

package: xinit
pin: version *
pin-priority: -1

package: xserver*
pin: version *
pin-priority: -1

package: *libx11-xcb1*
pin: version *
pin-priority: -1

This was used as a reference: [https://forums.macrumors.com/threads/making-g3s-and-radeon-graphics-great-again.2191877/](https://forums.macrumors.com/threads/making-g3s-and-radeon-graphics-great-again.2191877/)

I'm not sure if this is something of your interest to try and/or try implementing to your images, HexDump as it's a bit 'hacky' and I've noticed you (understandably) prefer upstream, non-blobby drivers when possible.  Hopefully this'll help someone who's interested in making use of their NVIDIA GK20/Tegra K1 GPU with on a newer Linux system.

I've tried Nouveau it's worse than I thought when I tried a newer Mesa, the patched one (with artifacting) turned out better.  At least I was able to reclock it without it freezing.  An old 2010 MacBook Pro of froze when I tried to reclock it with Nouveau.

'glxinfo' and 'glxgears' detect the NVIDIA drivers and the minute I got XFCE to recognize the NVIDIA drivers did it enable hardware compositing automatically.  It runs very well.  The biggest downside is the fact that it needs the ancient ChromeOS 3.10 kernel.  I'm considering seeing if I could potentially bring the open kernel modules forward, but compared to the Mali kernel modules, it looks kind of complicated.  Looks like no one has tried for the kind of hardware yet.  I'm nowhere near experienced, but I suppose I won't know until I try!

I'll send more info here later.  I'll see if I can get Jammy to run with the blobs.  Have a fantastic morning, you all!

nvidialinuxuser, 2023-11-07:
I stand corrected.  Lightdm also works now.

nvidialinuxuser, 2023-11-07:
It seems buggy though.  At least we have a way of launching XFCE though.

nvidialinuxuser, 2023-11-07:
Update: Update: I can confirm that Ubuntu 22.04 LTS also works with NVIDIA proprietary blobs using my method.

hexdump0815, 2023-11-07:
it is very nice to see this - this is one of the reasons i create all those images: so that it is easy for people to start playing around with linux on those devices without having to struggle a lot to get them even boot (and maybe even giving up along the way in the end) to find maybe different and new use-cases for them ... it does not always have to be the latest raspberry pi for tinkering around while there is so much still very useable hardware around :)

hexdump0815, 2023-11-07:
fyi: i linked this github issue in the nyan readme now, so that others interested in the same topic can easily find it: https://github.com/hexdump0815/imagebuilder/commit/c81e9c3783c03b65089b33349c67b171fa13410b

nvidialinuxuser, 2023-11-07:
Also, @hexdump0815, I noticed that you have done a custom Xorg build for at least one of your images.  Surrounding building older Xorg, do you think this could work?  I'm thinking about potentially making a package or just deploying Xorg in that kind of way.  I don't mind making Xenial chroots and upgrading them, but as you know of course, the most optimized method is the better method for these kinds of things.

nvidialinuxuser, 2023-11-11:
Going to post my progress here.  I should note that I absolutely no C knowledge here.  Even then still, I'm trying to port the 'nvgpu' driver for the GK20 GPU and interestingly enough, with some kind of limited semi-success.  It was more difficult on 4.14 LTs, so I figured to port this on a kernel that is still supported in some way that has many of the functions GPU driver asks for, kernel 4.4 LTS by Civil Infrastructure Platform.

I'm not an expert at this, but I've gotten it to build to an extent, right now, I face this:
```
drivers/gpu/nvgpu/gk20a/gk20a.c: In function 'gk20a_pm_initialise_domain':
drivers/gpu/nvgpu/gk20a/gk20a.c:1272:3: error: implicit declaration of function 'pm_genpd_set_poweroff_delay' [-Werror=implicit-function-declaration]
 1272 |   pm_genpd_set_poweroff_delay(domain, platform->railgate_delay);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/nvgpu/gk20a/gk20a.c: In function 'gk20a_secure_page_alloc':
drivers/gpu/nvgpu/gk20a/gk20a.c:1315:3: error: implicit declaration of function 'tegra_periph_reset_assert' [-Werror=implicit-function-declaration]
 1315 |   tegra_periph_reset_assert(platform->clk[0]);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/nvgpu/gk20a/gk20a.c:1318:3: error: implicit declaration of function 'tegra_periph_reset_deassert' [-Werror=implicit-function-declaration]
 1318 |   tegra_periph_reset_deassert(platform->clk[0]);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/nvgpu/gk20a/gk20a.c: In function 'gk20a_probe':
drivers/gpu/nvgpu/gk20a/gk20a.c:1466:6: error: passing argument 4 of 'debugfs_create_bool' from incompatible pointer type [-Werror=incompatible-pointer-types]
 1466 |      &gk20a->mm.ltc_enabled_debug);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |      |
      |      u32 * {aka unsigned int *}
In file included from drivers/gpu/nvgpu/gk20a/gk20a.c:41:
include/linux/debugfs.h:96:16: note: expected 'bool *' {aka '_Bool *'} but argument is of type 'u32 *' {aka 'unsigned int *'}
   96 | struct dentry *debugfs_create_bool(const char *name, umode_t mode,
      |                ^~~~~~~~~~~~~~~~~~~
drivers/gpu/nvgpu/gk20a/gk20a.c:1476:6: error: passing argument 4 of 'debugfs_create_bool' from incompatible pointer type [-Werror=incompatible-pointer-types]
 1476 |      &gk20a->timeouts_enabled);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
      |      |
      |      u32 * {aka unsigned int *}
In file included from drivers/gpu/nvgpu/gk20a/gk20a.c:41:
include/linux/debugfs.h:96:16: note: expected 'bool *' {aka '_Bool *'} but argument is of type 'u32 *' {aka 'unsigned int *'}
   96 | struct dentry *debugfs_create_bool(const char *name, umode_t mode,
      |                ^~~~~~~~~~~~~~~~~~~
  CC      drivers/clk/clk-divider.o
drivers/gpu/nvgpu/gk20a/gk20a.c: In function 'gk20a_remove':
drivers/gpu/nvgpu/gk20a/gk20a.c:1513:2: error: implicit declaration of function 'nvhost_module_disable_clk'; did you mean 'nvhost_module_idle_ext'? [-Werror=implicit-function-declaration]
 1513 |  nvhost_module_disable_clk(&dev->dev);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
      |  nvhost_module_idle_ext
At top level:
drivers/gpu/nvgpu/gk20a/gk20a.c:1240:12: error: 'gk20a_pm_resume' defined but not used [-Werror=unused-function]
 1240 | static int gk20a_pm_resume(struct device *dev)
      |            ^~~~~~~~~~~~~~~
drivers/gpu/nvgpu/gk20a/gk20a.c:1222:12: error: 'gk20a_pm_suspend' defined but not used [-Werror=unused-function]
 1222 | static int gk20a_pm_suspend(struct device *dev)
      |            ^~~~~~~~~~~~~~~~
```