velvet-os/imagebuilder #177: Kernel panic  on boot for kukui juniper-rev4

Endermen1094, 2023-11-11:
I get unknown file system of root fs , currently doing a rewrite without a dock

Endermen1094, 2023-11-11:
I should note I am using the chromebook_kukui-aarch64-bookworm.img

Endermen1094, 2023-11-11:
Also I beleave I got it from https://github.com/hexdump0815/imagebuilder/releases/tag/230917-01

Endermen1094, 2023-11-11:
It was a bad flash , made sure to sync before rebooting and it works

avoiceofreason, 2023-11-12:
Tip: Use Raspberry Pi imaging utility (rpi-imager) to burn sd-card images. It's got a solid erase option and also does a verify pass which may take a minute or so to run but saves a lot of "kernel panic" pain later.

Endermen1094, 2023-11-12:
Will do so in the future. Also, what is the best way to install to internal mmc storage ?

Endermen1094, 2023-11-12:
Also it has been acting strangely with not being as performant  on logging in , with some hang (I think is what it's called when seemingly  not doing anything but recovers with some time.

hexdump0815, 2023-11-12:
see https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md for installing to emmc