velvet-os/imagebuilder #182: chromebook_trogdor: status: wormdingler (lenovo chromeboot duet 3)

vulpes2, 2023-12-08:
Noticed there wasn't a devcie-specific issue for wormdingler, so I thought I'd open one to keep info in the correct place.

## Software setup

- kernel: 6.4.16 (based on config from [v6.1.51](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.1.51-stb-cbq))
- distro: Debian sid
- mesa: 23.2.1 (from the Debian sid repo)
- window protocol: wayland
- desktop environment: gnome

## Working

- Wi-Fi (including WPA3-SAE and 802.11w) and Bluetooth
- Built-in display (orientation, brightness controls)
- Energy: battery status, dpms, suspend/resume
- IIO sensors: gyroscope (auto-screen orientation, [udev rule](https://wiki.postmarketos.org/wiki/Lenovo_IdeaPad_Duet_Chromebook_(google-krane)) required). Link is for krane but the rule should work the same.
- Trackpad: works well, libinput quirk required to configure pressure correctly just like krane.
- USB: USB host, PD charging, display out (no audio)
- Touchscreen: works out of the box, orientation is correct and follows screen rotation on GNOME
- Graphics: OpenGL works out of the box with freedreno, performance is noticeably better than mali
- Audio: speakers and mics seem okay

## Needs work

Will investigate in these when I have more time.

- Hardware video decoding/encoding: venus firmware is loaded, tested decoding with `mpv --hwdec=v4l2m2m-copy`, it's extremely choppy and unusable
- Pen digitizer: cannot test, this device will only work with **USI2** pen which I don't have
- Audio: ucm config needs to be modified to remove the nonexistent headset and mic. dt might also need tweaks to fix the flood of `ASoC: error at __soc_pcm_hw_params on MultiMedia2: -22` messages.

## Broken
- Cameras: CSI cameras, no mainline support for the subsystem.

## Notes

1. IIO sensors require `iio-sensor-proxy` to be installed and running for them to work with GNOME.
2. Hardware compositing work out of the box with chromium and firefox once they are running natively on Wayland.

vulpes2, 2023-12-08:
Unfortunately starting from kernel 6.5 the display has stopped working, the device will only boot to a blank screen. Tested two config files, one was from the existing v6.1.51, one was [config.cbq-6.6.0-stb-cbq](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/config.cbq-6.6.0-stb-cbq%2B), both gave the same results. Power management seems to be working as the power button does turn the backlight on and off, but there is simply nothing displayed on the screen. From what I can tell a homebrew SuzyQ should give me access to the AP serial console, don't know when I'll get to it but that's probably the next step for troubleshooting.

hexdump0815, 2023-12-08:
@vulpes2 - thanks a lot for creating thise separate gh issue for wormdingler, i have now also linked it in the trogdor readme: https://github.com/hexdump0815/imagebuilder/commit/ee9e71bb13dac378f9292a22aa5b8beefbab30ba

for debugging blank screen problems this might be helpful: https://github.com/hexdump0815/imagebuilder/issues/61#issuecomment-1843374506 ... did you try v6.6 already? if not: maybe its fixed there and in general it is a very good target as it is a lts kernel, so it will get fixes for the next three years

in case you would prefer to buy something suzyqable like: there seems to be someone selling working devices now on ebay - see the link at the end of https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks ... i have no relation to this, just read on irc that it seems to work

vulpes2, 2023-12-09:
Both 6.5 and 6.6 boot to a blank screen, I'll try the script trick you mentioned and see if I can get some logs. It's really a shame because I was really hoping 6.6 would work out of the box, since it's a nice LTS release.

In terms of SuzyQ, I do actually have the breakout board, just haven't gotten around to putting it together yet. It's inevitable that I'll get frustrated enough and decide to put it together, it's just a matter of time basically.

vulpes2, 2023-12-09:
Added the below snippet to `rc.local` to get some logs, and I got a completely normal boot sequence with no additional error messages on v6.5.13. Since there were a lot of renames in the dts, I tried using the dt from 6.4 with kernel 6.5 too, and nothing seems to have changed. I'm out of ideas at this point, even getting a serial console to the soc won't help me, will stick to v6.4 for now. It would be great if anyone can provide some more hints on how to debug this further. 
```shell
if [ ! -f /marker ]; then
	dmesg > /dmesg.txt
	touch /marker
	shutdown -h now
fi
```

hexdump0815, 2023-12-09:
@vulpes2 - if you have a working and a non working version, then git bisect should lead to the breaking commit - it is a bit painful (due to having to compile and install multiple kernels) and requires a bit of discipline (to always install the proper kernel each time, as the kernel version will jump around a bit between last, new and various -rcx versions during bisecting) but it helped me to find the root cause of problems quite a few times already ... if you do that, then do it on a usb device, so that you can easily revert to a working kernel via dd on another machine - for the emmc case otherwise this would be of interest: https://github.com/hexdump0815/imagebuilder/blob/main/doc/test-booting-a-kernel-on-chromebooks.txt ... good luck

btw. when you mention suzyq breakout board above, do you mean the google servoe board or some self built variant?

vulpes2, 2023-12-09:
Yeah, that's what I have been doing, building the kernel and flashing the kpart image to a flashdrive so there will be a working copy on the emmc at all times. I've previously tried to bisect a kernel issue on a different board and it was extremely painful due to the amount of manual work involved, even with the image building & flashing process completely scripted, so I'm pretty hesitant to try that again. 

About the suzyq thing, I have one of these generic breakout boards to build a cable with, not a fancy servo board unfortunately.
[image]

vulpes2, 2023-12-25:
An update on the display situation, I tried bisecting the kernel source and found the root cause, the dsi bridge is powered on too late so panel init is failing. I'm not much of a developer so my ability to troubleshoot this kind of stuff is very limited, but I did come up with a hack that can reliably mitigate this issue for now. The patch and more details can be found in the upstream bug report: https://gitlab.freedesktop.org/drm/msm/-/issues/47

By the way, I did end up putting together a debug cable, having access to the AP serial console is super helpful for debugging, but I'm not sure how I can unlock ccd without reinstalling ChromeOS.

vulpes2, 2023-12-26:
A proper fix has been suggested and it has been tested. Will try to get it upstreamed and backported to linux-stable.

hexdump0815, 2023-12-27:
i usually do a ccd open and setting the gbb flags to 0x19 as a first thing before using a chromebook with linux ... it might be required to backup (maybe dd) your current system, restore chromeos, ccd open and then restore linux again ... maybe searching the web for "sh1mmer" might bring up an option to boot and ccd open your device from usb :)

vulpes2, 2023-12-28:
Unfortunately there's no leaked RMA shim for trogdor so that's a dead end. I'll leave it as is for now and reinstall ChromeOS at some point to fix it.

hexdump0815, 2023-12-28:
@vulpes2 - i think wormdingler is based on strongbad (see: https://www.chromium.org/chromium-os/developer-information-for-chrome-os-devices/ - and strongbad somehow seems to be some variant of trogdor - don't ask me about the logic behind all this naming :) )

FrostNovaHD, 2024-02-16:
@hexdump0815 can you make a updated kernel build for chromebook trogdor i cant do it my self

hexdump0815, 2024-02-17:
@FrostNovaHD - i have no time to build new kernels right now - but it is not that complicated to build one yourself and it can be done easily on the device itself - see: https://github.com/hexdump0815/imagebuilder/blob/main/doc/building-own-kernels.md

vulpes2, 2024-02-24:
You do not need a new bootable image, you can simply build the kernel and install it on top of the existing image. I may post an updated kernel image at some point, no promises though.

XxTHKxX, 2024-06-23:
Heya, just hopping on to the party here. The last bootable image for trogdor is kernel 6.1, uh. can I get some hand holding to get the device to a good stable new kernel/image? ((i just got the chromebook like 2 days ago so I've got not much experience with it's hardwares, I know a bit of linux and such but not much)). Dev mode is enabled, and a test boot of the last bootable sd images is successful, just wanted to be a bit more up to date

LukIsHere, 2024-06-23:
the latest avaliable kernel is [here](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.6.23-stb-cbq%2B) (6.6.23)
you just need to follow [this guide](https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md) it is really simple just do it step by step

XxTHKxX, 2024-06-26:
> the latest avaliable kernel is [here](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.6.23-stb-cbq%2B) (6.6.23) you just need to follow [this guide](https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md) it is really simple just do it step by step @XxTHKxX

Uh, if I may ask, is there a way for me to edit the cmdline for this kernel (without rebuilding it completely at least)? I'd like to reenable ipv6 support on the kernel, as part of my network requires the usage of ipv6 in order to connect to their service

XxTHKxX, 2024-06-26:
Okay so I tried to build and install the latest kernel based on instruction on building-your-own-kernel and the corresponding readme file for my device, but seems like even after installing the new kernel via 
`dd if=/boot/vmlinux.kpart-6.6.23-stb-cbq+ of=/dev/mmcblk0p1`
The booted image still use the old kernel, any suggestion?

LukIsHere, 2024-06-26:
> Uh, if I may ask, is there a way for me to edit the cmdline for this kernel 
afaik no without building it at least once.
it isn't very complicated process as long as u just copy comand one by one and modifying one file

if [depthcharge-tools are added to the images](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/issues/13#issuecomment-2143590367) it should be possible

> for my device, but seems like even after installing the new kernel via
> dd if=/boot/vmlinux.kpart-6.6.23-stb-cbq+ of=/dev/mmcblk0p1
> The booted image still use the old kernel, any suggestion?

try running ```lsblk``` and checking if ur emmc aint called mmcblk1 instead of mmcblk0
if that's the case the command shoud be 
```
dd if=/boot/vmlinux.kpart-6.6.23-stb-cbq+ of=/dev/mmcblk0p1
```
@XxTHKxX

LukIsHere, 2024-07-18:
about stylus
it does "work" but has same issue as [lenovo 10e](https://github.com/hexdump0815/imagebuilder/issues/202#issuecomment-2177195444) x and y axis are swapped
i still don't know solution for it but when i find one will make sure to share it

na-sa-do, 2024-07-19:
Maybe it's going to be a mess, but it's also half the price of the two Linux-native tablets I'm looking at, so I'm still considering getting one of these. [The postmarketOS wiki says](https://wiki.postmarketos.org/wiki/Google_Trogdor_Chromebook_(google-trogdor)) the modem doesn't work after suspending, but this issue doesn't note anything like that -- is that specific to their setup? Also, if hardware video decoding is unusably choppy atm, how bad is software decoding for perf & battery?

LukIsHere, 2024-07-19:
@na-sa-do if you're asking for recommendation. for daily drive i would go with either
## (my choice) [Lenovo IdeaPad Duet 5](https://wiki.postmarketos.org/wiki/Lenovo_IdeaPad_Duet_5_(google-homestar))

it has
 - great size (13,3") way bigger than duet 3
 - correcly working stylus (isn't the case with all trogdor's)
 - [almost out of the box audio](https://github.com/hexdump0815/imagebuilder/issues/68#issuecomment-2186001368)
## [Acer Chromebook Spin 513](https://wiki.postmarketos.org/wiki/Acer_Chromebook_Spin_513_(google-lazor))
it has
 - working camera (if u care for that)
 - sometimes funny screen rotation behaviour
 - working camera (if u care abt that)

in general i'd pick duet 5 given choice, while acer spin works well it doesn't feel that well made (just prefference)
if you want a **small tablet** i'd go with duet 3, if you want sth slightly bigger with good (attachable) keyboard, i'd say duet 5

for all, **video decoding** seams to currently be working fine (on the latest kelner) with ffplay and firefox, [encoding not so much](https://github.com/hexdump0815/imagebuilder/issues/223#issuecomment-2190003468)

on the topic of **modem** as far as i know neither acer spin, duet 3 nor duet 5 has mobile data capability so it doesn't really apply to them

also i'd recommend [updating the kelner](https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md) to [the latest](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.6.23-stb-cbq%2B) right way after installing the os (it fixed broken touchscreen on duet 5)

na-sa-do, 2024-07-19:
Thanks, @LukIsHere. I know it doesn't have cellular, but I asked about the modem anyway because someone told me that on most Qualcomm chips, the cellular, wifi, and bluetooth are all linked together somehow, so if the cellular was broken the others might be too. But I take it you haven't had that issue?

LukIsHere, 2024-07-19:
@na-sa-do nope
wifi has no issue with it (at least i don't know of any)
i've had duet 3 for just a few days and haven't noticed anything
same on others i had for longer

etihwjd, 2024-11-15:
I've just installed the latest Ubuntu build on this device, and I may be overlooking something glaringly obvious, but I'm not getting any Wi-Fi when I boot from USB.

thenameisluk, 2024-11-15:
@etihwjd trogdor https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md ?

etihwjd, 2024-11-15:
> @etihwjd trogdor https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md ?

That'd explain it! 😅 I didn't spot that as I was reading around

thenameisluk, 2024-11-15:
yeah unless you know where it is it's hard to spot

hopefully will he fix in future images

thenameisluk, 2024-11-23:
just a quick note regarding usi stylus on this device (which is rotated 90*)

putting file like this
```
ENV{DEVNAME}=="/dev/input/event3",ENV{LIBINPUT_CALIBRATION_MATRIX}="0 -1 1 1 0 0"
```
in ```/etc/udev/rules.d/99-wormdingler-stylus.rules```
and rebooting the device should fix the issue

**for other devices** you need to change /dev/input/eventX to correct one
you might be also interested in trying [diffrent rotation matrix'es](https://wayland.freedesktop.org/libinput/doc/latest/device-configuration-via-udev.html)
_Encourage. you to share correct config on other device issues if found one :3_

should probably put it in doc's at some point with other device calibratino stuff

vulpes2, 2025-03-06:
Anyone who's got a working system right now, could you please submit a hardware info dump at libwacom to get this device properly supported upstream? On my system the hid descriptors aren't getting dumped even if I run the script as root, and I don't have time to debug this yet. https://github.com/linuxwacom/wacom-hid-descriptors/tree/master?tab=readme-ov-file#contributing

vulpes2, 2025-03-06:
I'm currently working on a proper fix for the touchscreen and accelerometer orientation for trogdor that is suitable for inclusion in the upstream systemd hwdb. Hopefully we can get rid of the annoying downstream udev rules soon.

thenameisluk, 2025-03-06:
Homestar

thenameisluk, 2025-03-06:
sorry missclick

vulpes2, 2025-03-06:
Thanks, opened an issue upstream with the sysinfo attached. Yours seems to include the HID descriptors, not sure what's wrong with my setup yet.

thenameisluk, 2025-03-06:
Wormdingler

[sysinfo.FApjDxOqZN.tar.gz](https://github.com/user-attachments/files/19110005/sysinfo.FApjDxOqZN.tar.gz)

thenameisluk, 2025-03-06:
Coachz

[sysinfo.QpItYxvd9t.tar.gz](https://github.com/user-attachments/files/19110139/sysinfo.QpItYxvd9t.tar.gz)

vulpes2, 2025-03-06:
Thanks for sending these through, looks like it's not just me who's not getting the HID descriptors on wormdingler (the `.hid.bin` file is missing). Not sure why it can't be dumped, or what we can do about it.

thenameisluk, 2025-03-06:
lazor

[sysinfo.eivV8YFDBY.tar.gz](https://github.com/user-attachments/files/19110222/sysinfo.eivV8YFDBY.tar.gz)

thenameisluk, 2025-03-06:
yeah that's all for trogdors

do you already have krane (kukui)? cause i have one i could dump but it kinda has some framebuffer smart home interface instead of gui (makes it harder to dump)

vulpes2, 2025-03-06:
I do have a krane but I need to reimage it first which is kinda annoying to do, which is why I was asking if someone else has a working one.

thenameisluk, 2025-03-06:
Kodama (kukui)

[sysinfo.cM5ZoYn7OK.tar.gz](https://github.com/user-attachments/files/19110419/sysinfo.cM5ZoYn7OK.tar.gz)

ok imma get krane

thenameisluk, 2025-03-06:
Krane (kukui)

[sysinfo.DzuHdfGOij.tar.gz](https://github.com/user-attachments/files/19110518/sysinfo.DzuHdfGOij.tar.gz)

vulpes2, 2025-03-06:
Thanks for all the hardware info, I've opened issues at the upstream repo, hopefully these will get processed soon. Once they've been added upstream, plasma and gnome should be able to display correct metadata and configuration menus for the tablet.

thenameisluk, 2025-03-06:
Np, that's all for my chromebooks with stylus/touch support

thenameisluk, 2025-03-06:
not sure i usually disable the auto rotation right away since it's annoying to me

i know lenovo 10e is 90 degrees but don't recall the other

thenameisluk, 2025-03-06:
look
https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/systems/kukui/kodama.md

vulpes2, 2025-03-07:
That would be great, anything that has stylus support can be reported to libwacom.

vulpes2, 2025-03-08:
Stylus rotation is a bug with mutter, it works out of the box on plasma. There was an issue about this exact problem on mutter already: https://gitlab.gnome.org/GNOME/mutter/-/issues/2826#note_2372530

The fact that a mount matrix is required is also an oversight in mutter, the cr50 reports the device's physical orientation, not the panel's orientation, which is why it's always off on devices with rotated panels. That udev rule is not required on devices without it (like coachz). Filed a bug report with mutter: https://gitlab.gnome.org/GNOME/mutter/-/issues/3957

Aleksanaa, 2025-03-18:
Has anyone encountered suspend time out (returned by EC) issue on 6.12 kernel?  I'm using another unreleased engineering prototype of strongbad close to wormdingler, but I suspect recent changes to cros_ec may have caused this.

thenameisluk, 2025-03-18:
@Aleksanaa which chomebook are you reffering to?
if it's so new its device tree is probably not upsteam yet
so it won't work well

Aleksanaa, 2025-03-18:
If that's not the case here, I don't want to mess up with this issue. I've opened a new discussion: https://github.com/hexdump0815/imagebuilder/discussions/315

Aleksanaa, 2025-03-21:
Here's a patchset for (potential) sc7180 trogdor camera support:

https://patchwork.kernel.org/project/linux-arm-msm/cover/20240621-b4-sc7180-camss-v1-0-14937929f30e@gmail.com/

But I feel that the author is a bit confused about their work, and they did not test it on sc7180, but only on sm7125. It looks like it needs a little tweaking, but I haven't tested it yet so I'm not sure.

hexdump0815, 2025-03-21:
@Aleksanaa - thanks a lot for sharing this, i did not yet run across those patches ... sadly i have no time to play around with them right now (little time and new image release preparations), but maybe later during the year or someone else picks it up

thenameisluk, 2025-03-21:
i can try them but i'm no kernel dev to fix them saddly

thenameisluk, 2025-03-23:
i've had a look at these patches and they don't seam to fully apply

the rejections follow a pattern
```patch
--- drivers/media/platform/qcom/camss/camss-video.c
+++ drivers/media/platform/qcom/camss/camss-video.c
@@ -1027,6 +1027,7 @@ int msm_video_register(struct camss_video *video, struct v4l2_device *v4l2_dev,
                }
                break;
        case CAMSS_845:
+       case CAMSS_7180:
        case CAMSS_8250:
        case CAMSS_8280XP:
                video->formats = formats_rdi_845;
```

```patch
--- drivers/media/platform/qcom/camss/camss-csiphy.c
+++ drivers/media/platform/qcom/camss/camss-csiphy.c
@@ -577,6 +577,7 @@ int msm_csiphy_subdev_init(struct camss *camss,
                csiphy->nformats = ARRAY_SIZE(csiphy_formats_8x96);
                break;
        case CAMSS_845:
+       case CAMSS_7180:
        case CAMSS_8250:
        case CAMSS_8280XP:
                csiphy->formats = csiphy_formats_sdm845;
```

```patch
--- drivers/media/platform/qcom/camss/camss-vfe.c
+++ drivers/media/platform/qcom/camss/camss-vfe.c
@@ -1519,6 +1520,7 @@ int msm_vfe_subdev_init(struct camss *camss, struct vfe_device *vfe,
                        }
                        break;
                case CAMSS_845:
+               case CAMSS_7180:
                case CAMSS_8250:
                case CAMSS_8280XP:
                        l->formats = formats_rdi_845;
```
when i try to apply them manually i cannot find any switch case anywhere
also noticed kernel devs use goto for whatevere reason (terrible practice and it doesn't require rust to be fixed)

either way i'll still attempt to get it compiled and test

thenameisluk, 2025-03-23:
doesn't compile

thenameisluk, 2025-03-23:
i will go with this https://github.com/99degree/linux/commits/working-20241104/
c code isn't anything

dts is also important

thenameisluk, 2025-03-23:
imma use these 5 patches
```
 1404  wget https://github.com/99degree/linux/commit/b2f575b212e8bc30eff81a697143881f4e533e48.patch
 1405  wget https://github.com/99degree/linux/commit/a6ea6478c53092cd1660aa480e753e7a1ef36c25.patch
 1406  wget https://github.com/99degree/linux/commit/5818e4dbed8c913c965d9458410af18a21da000e.patch
 1407  wget https://github.com/99degree/linux/commit/9ee9e7813347ff40e418dcfd0ba31603fe71704c.patch
 1408  wget https://github.com/99degree/linux/commit/b626aedf30e0d440b83c95ca77758cb476dfbc98.patch
```

thenameisluk, 2025-03-23:
nothing
no camera detected
no new stuff v4l related in /sys

only interesting thing i found was
```
[    0.948681] reg-fixed-voltage pp2800-uf-cam-regulator: nonexclusive access to GPIO for (default)
[    0.949041] reg-fixed-voltage pp2800-wf-cam-regulator: nonexclusive access to GPIO for (default)
```
in dmesg 
don't think removing that fixed regulator would fix anything

other than that no clues what is missing

hexdump0815, 2025-03-24:
@thenameisluk - looks like this one is a bit more complicated ... sometimes it helps to put it aside for a while and sometimes then new ideas might come up ... i might have a look at this (and the kukui cam stuff) at some point too, but due to limited time not in the near future

[10 more replies]