velvet-os/imagebuilder #184: Working on Lenovo Chromebook C330 (Aarch64, board-name: hana)

johnwuelk, 2023-12-31:
I confirm that chromebook_oak-aarch64-focal.img as well as chromebook_oak-aarch64-bookworm.img is working and booting successfully, also locally, on aforementioned c330 arm-device.
Thank you for your work!
I dd'd the image into the biggest internal partition and then booting without USB using internal storage was successful.

Only **Sound** is **not** working, and I don't know if it's a driver or a broken hardware issue as ChromeOS always freezes within a matter of minutes after login and the only remedy is a force-reboot.
In Linux the system freezes **only if** sound is played! If sound is not touched, the system works flawlessly. Perhaps someone might have an idea what is causing this problem.  Either way, I'm going to have to keep Linux on this device as it is unusable with ChromeOS.

hexdump0815, 2024-01-06:
i'm not aware of any audio-freeze problems on oak - it might really be a hardware problem on your system.

GerardoHD, 2024-05-23:
I also got it working on the C330, i changed from Xfce to GNOME Shell and it feels great!

Currently updating it to the testing repos so i can get newer packages.

Everything working except HDMI ouput, but that seems like asking too much haha.

Thank you!

ericinc1, 2024-11-21:
where can i find the img file

hexdump0815, 2024-11-21:
@ericinc1 - https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_oak and please make sure to carefully read https://github.com/hexdump0815/imagebuilder/tree/main/doc/chromebooks ... good luck

ericinc1, 2024-11-21:
wow thanks for the quick response. And thank you i really need it lol